introlab/rtabmap_ros #1068: Improve Mapping Qaulity

360wcui, 2023-11-19:
We have a custom camera setup.   We are able to port in point cloud, hires, and vio for rtabmap.   Below is the resulting 3D map in RViz.  I wonder if there is anything we can do to further improve mapping quality.    Thank you. 

[
[image]
](url)

360wcui, 2023-11-19:
The hires, vio, and point cloud are published at 5 Hz

matlabbe, 2023-11-20:
Is it with a TOF camera? In term of quality, do you mean point cloud noise, color, misalignement? The frame rate is not that important on mapping side as long as the VIO pose and all image topics are already synchronized.

360wcui, 2023-11-20:
Yes, we use a ToF sensor.  Th max range is about 5m.

closed 2023-11-20 by 360wcui