introlab/rtabmap_ros #1070: ros2 humble realsense d435i in gazebo

yuan-0816, 2023-11-23:
I want to simulate the RealSense D435i in Gazebo and use RTAB-Map for localization. 
But RTAB-Map is unable to receive the camera topic.

Here are my xacro and launch files.

`<?xml version="1.0"?>

<robot xmlns:xacro="http://ros.org/wiki/xacro">
    <xacro:macro name="gazebo_d435i" params=" reference_link
                                            camera_name:=camera
                                            topics_ns:=camera

                                            depth_optical_frame
                                            color_optical_frame
                                            infrared1_optical_frame
                                            infrared2_optical_frame
                                            accel_optical_frame
                                            gyro_optical_frame

                                            visualize:=false
                                            align_depth:=false
                                            enable_pointCloud:=false

                                            unite_imu_method:=false
                                            accel_fps:=250
                                            gyro_fps:=400

                                            clip_distance:=-1.0
                                            depth_width:=1280
                                            depth_height:=720
                                            depth_fps:=30

                                            infra_width:=640
                                            infra_height:=480
                                            infra_fps:=30

                                            color_width:=1920
                                            color_height:=1080
                                            color_fps:=30
                                            ">
        
        <xacro:property name="deg_to_rad" value="0.01745329251994329577"/>
        <gazebo reference="${reference_link}">
            <self_collide>0</self_collide>
            <enable_wind>0</enable_wind>
            <kinematic>0</kinematic>
            
            
            <mu2>1</mu2>
            <fdir1>0 0 0</fdir1>
            
            <kp>1e+13</kp>
            <kd>1</kd>
            
            <sensor name="${camera_name}color" type="camera">
                <camera name="${camera_name}">
                    <horizontal_fov>${69.4*deg_to_rad}</horizontal_fov>
                    <image>
                        <width>${color_width}</width>
                        <height>${color_height}</height>
                        <format>RGB_INT8</format>
                    </image>
                    <clip>
                        <near>0.1</near>
                        <far>100</far>
                    </clip>
                    <noise>
                        <type>gaussian</type>
                        <mean>0.0</mean>
                        <stddev>0.007</stddev>
[... 462 lines cut ...]
[rviz2-3] [INFO] [1700752195.882848604] [rclcpp]: signal_handler(signum=2)
[imu_filter_madgwick_node-5] [INFO] [1700752195.884572765] [rclcpp]: signal_handler(signum=2)
[rgbd_odometry-1] [WARN] [1700752195.888917444] [rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
[rgbd_odometry-1] rgbd_odometry subscribed to (exact sync):
[rgbd_odometry-1]    /camera/infra1/image_rect_raw,
[rgbd_odometry-1]    /camera/depth/image_rect_raw,
[rgbd_odometry-1]    /camera/infra1/camera_info
[rtabmap-2] [INFO] [1700752195.932880938] [rtabmap]: Parameters are not saved (No configuration file provided...)
[INFO] [imu_filter_madgwick_node-5]: process has finished cleanly [pid 48029]
[INFO] [point_cloud_xyzrgb-4]: process has finished cleanly [pid 48027]
[INFO] [rviz2-3]: process has finished cleanly [pid 48025]
[ERROR] [rtabmap-2]: process[rtabmap-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[ERROR] [rgbd_odometry-1]: process[rgbd_odometry-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [rtabmap-2]: sending signal 'SIGTERM' to process[rtabmap-2]
[INFO] [rgbd_odometry-1]: sending signal 'SIGTERM' to process[rgbd_odometry-1]
[rtabmap-2] [INFO] [1700752200.887038270] [rclcpp]: signal_handler(signum=15)
[rgbd_odometry-1] [INFO] [1700752200.887479670] [rclcpp]: signal_handler(signum=15)
[rtabmap-2] rtabmap: Saving database/long-term memory... (located at /home/yuan/.ros/rtabmap.db)
[rtabmap-2] rtabmap: Saving database/long-term memory...done! (located at /home/yuan/.ros/rtabmap.db, 0 MB)
[INFO] [rtabmap-2]: process has finished cleanly [pid 48023]
[INFO] [rgbd_odometry-1]: process has finished cleanly [pid 48021]
`
[image]

"I'm using ROS2 Humble on Ubuntu 22.04."

yuan-0816, 2023-11-23:
[image]

matlabbe, 2023-11-24:
I suggest to do `rostopic hz` instead of `rostopic list` to acutally know that the topics are published.
```
rostopic hz /camera/infra1/image_rect_raw
rostopic hz /camera/depth/image_rect_raw
rostopic hz /camera/infra1/camera_info
```
I see you are also using `approx_sync` set to false (exact sync), are the image topics published with same stamp?

yuan-0816, 2023-11-24:
I used ```ros2 topic hz```, and all three topics are being published.
```
yuan@airlab-ubuntu:~$ ros2 topic hz /camera/infra1/image_rect_raw
average rate: 26.126
	min: 0.032s max: 0.047s std dev: 0.00314s window: 28
average rate: 26.148
	min: 0.032s max: 0.050s std dev: 0.00367s window: 55
average rate: 26.629
	min: 0.032s max: 0.050s std dev: 0.00329s window: 83
average rate: 26.471
	min: 0.032s max: 0.050s std dev: 0.00355s window: 109
average rate: 26.757
	min: 0.031s max: 0.050s std dev: 0.00340s window: 137
average rate: 26.751
	min: 0.031s max: 0.054s std dev: 0.00365s window: 164
average rate: 26.904
	min: 0.031s max: 0.054s std dev: 0.00349s window: 192
average rate: 26.845
	min: 0.031s max: 0.054s std dev: 0.00359s window: 219
^Cyuan@airlab-ubuntu:~$ ros2 topic hz /camera/depth/image_rect_raw
average rate: 20.165
	min: 0.033s max: 0.112s std dev: 0.02065s window: 22
average rate: 19.930
	min: 0.033s max: 0.144s std dev: 0.02396s window: 42
average rate: 19.556
	min: 0.033s max: 0.144s std dev: 0.02437s window: 61
average rate: 18.718
	min: 0.033s max: 0.181s std dev: 0.02850s window: 78
average rate: 18.535
	min: 0.033s max: 0.181s std dev: 0.02846s window: 96
average rate: 18.546
	min: 0.033s max: 0.181s std dev: 0.02738s window: 116
^Cyuan@airlab-ubuntu:~$ ros2 topic hz camera/infra1/camera_info
average rate: 28.262
	min: 0.031s max: 0.040s std dev: 0.00247s window: 30
average rate: 27.359
	min: 0.031s max: 0.050s std dev: 0.00369s window: 57
average rate: 27.674
	min: 0.031s max: 0.050s std dev: 0.00323s window: 86
average rate: 27.647
	min: 0.031s max: 0.053s std dev: 0.00361s window: 115
average rate: 27.510
	min: 0.031s max: 0.053s std dev: 0.00351s window: 142
average rate: 27.419
	min: 0.031s max: 0.053s std dev: 0.00356s window: 169

```
I set ```use sim time``` to true, so the camera topic should be published with the same timestamp as the simulation time.
And I also used ```ros2 run rqt_image_view rqt_image_view``` to subscribe to the camera topic, and there is a visual feed.

[image]
[image]

matlabbe, 2023-11-26:
The topics don't seem to be published at the same time, set `approx_sync` to true for all rtabmap nodes. The infra1 and depth images are not the same resolution, make sure they are using same focal length with same resolution from same link in your URDF, so that depth is perfectly aligned with infra1.

yuan-0816, 2023-11-29:
It works !!!
[image]

But I still want to know, If I set ```use sim time``` to ```true``` in the Gazebo launch file and also in RTAB-Map, why is the depth camera information still not synchronized?

matlabbe, 2023-12-10:
Is `use_sim_time` added to all rtabmap nodes? With how the new warning works, it won't use the simulated clock if `use_sim_time` is not set, then will complain that the topics are not received from the past 5 seconds. As the simulation time would start at epoch=0, all topics will be always in the past from the point of view of that check. Just tested with [turtebot3 example](https://github.com/introlab/rtabmap_ros/tree/ros2#example-with-turtlebot3) and it seems to do what I just described (no warning when `use_sim_time:=true`, warning when `use_sim_time:=false` but the nodes are working correctly).

yuan-0816, 2023-12-15:
Yes, I set the ```use_sim_time=True``` in all rtabmap nodes
This is my rtabmap launch file 

```
# Requirements:
#   A realsense D435i
#   Install realsense2 ros2 package (ros-$ROS_DISTRO-realsense2-camera)
#
#   RTAB-Map in mapping mode

import os

from launch import LaunchDescription, Substitution, LaunchContext
from launch.actions import DeclareLaunchArgument, SetEnvironmentVariable, LogInfo, OpaqueFunction
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir, PythonExpression
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from launch_ros.actions import SetParameter
from typing import Text
from ament_index_python.packages import get_package_share_directory

class ConditionalText(Substitution):
    def __init__(self, text_if, text_else, condition):
        self.text_if = text_if
        self.text_else = text_else
        self.condition = condition

    def perform(self, context: 'LaunchContext') -> Text:
        if self.condition == True or self.condition == 'true' or self.condition == 'True':
            return self.text_if
        else:
            return self.text_else

            
def launch_setup(context, *args, **kwargs):     

    use_sim_time = LaunchConfiguration('use_sim_time')
    qos = LaunchConfiguration('qos')
    localization = LaunchConfiguration('localization')

    parameters={
    # 'frame_id':'camera_link',
    'frame_id':'base_link',
    'use_sim_time':use_sim_time,
    'subscribe_depth':True,
    'subscribe_odom_info':True,
    'approx_sync':True,
    'qos_image':qos,
    'qos_imu':qos,
    'wait_imu_to_init':False
    }

    # use infra camera
    # remappings=[
    #       ('imu', '/imu/data'),
    #       ('rgb/image', '/camera/infra1/image_rect_raw'),
    #       ('rgb/camera_info', '/camera/infra1/camera_info'),
    #       ('depth/image', '/camera/depth/image_rect_raw')] 
    
    # use color camera 
    # remappings=[
    #     ('imu', '/imu/data'),
    #     ('rgb/image', '/camera/color/image_raw'),
    #     ('rgb/camera_info', '/camera/color/camera_info'),
    #     ('depth/image', '/camera/depth/image_rect_raw')] 

    # use color camera 
    remappings=[
        ('imu', '/imu/data'),
        ('rgb/image', '/camera/image_raw'),
[... 90 lines cut ...]
                              description='Launch RTAB-Map UI (optional).'),
        
        DeclareLaunchArgument('rviz', default_value='true', 
                              description='Launch RVIZ (optional).'),
        
        DeclareLaunchArgument('rviz_cfg', default_value=config_rviz, 
                              description='Configuration path of rviz2.'),
        
        DeclareLaunchArgument('initial_pose', default_value='', 
                              description='Set an initial pose (only in localization mode). Format: "x y z roll pitch yaw" or "x y z qx qy qz qw". Default: see "RGBD/StartAtOrigin" doc'),
        
        DeclareLaunchArgument('rtabmap_args',   default_value='',                   
                              description='Backward compatibility, use "args" instead.'),

        DeclareLaunchArgument('use_sim_time', default_value='true',
                              description='Use simulation (Gazebo) clock if true'),
        
        DeclareLaunchArgument('qos', default_value='2',
                              description='QoS used for input sensor topics'),

        OpaqueFunction(function=launch_setup)

        
    ])
```

matlabbe, 2023-12-16:
Can you doublecheck if the parameter is correctly set? 
```
ros2 param list
ros2 param get rtabmap use_sim_time
Boolean value is: True
```

Which rtabmap version are you using?

yuan-0816, 2023-12-20:
I run ``` ros2 param list```
```
yuan@airlab-ubuntu:~$ ros2 param list
/camera_controller:
  camera.depth.image_raw.format
  camera.depth.image_raw.jpeg_quality
  camera.depth.image_raw.png_level
  camera.depth.image_raw.tiff.res_unit
  camera.depth.image_raw.tiff.xdpi
  camera.depth.image_raw.tiff.ydpi
  camera.image_raw.format
  camera.image_raw.jpeg_quality
  camera.image_raw.png_level
  camera.image_raw.tiff.res_unit
  camera.image_raw.tiff.xdpi
  camera.image_raw.tiff.ydpi
  qos_overrides./clock.subscription.depth
  qos_overrides./clock.subscription.durability
  qos_overrides./clock.subscription.history
  qos_overrides./clock.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  update_rate
  use_sim_time
/diff_drive:
  qos_overrides./clock.subscription.depth
  qos_overrides./clock.subscription.durability
  qos_overrides./clock.subscription.history
  qos_overrides./clock.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  use_sim_time
/gazebo:
  enable_performance_metrics
  publish_rate
  qos_overrides./clock.subscription.depth
  qos_overrides./clock.subscription.durability
  qos_overrides./clock.subscription.history
  qos_overrides./clock.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time
/imu_filter_madgwick:
  constant_dt
  declination
  fixed_frame
  gain
  mag_bias_x
  mag_bias_y
  mag_bias_z
  orientation_stddev
  publish_debug_topics
  publish_tf
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
[... 797 lines cut ...]
  rgbd_cameras
  rtabmap
  subscribe_depth
  subscribe_odom
  subscribe_odom_info
  subscribe_rgb
  subscribe_rgbd
  subscribe_scan
  subscribe_scan_cloud
  subscribe_scan_descriptor
  subscribe_stereo
  subscribe_user_data
  use_sim_time
  wait_for_transform
/rviz:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  tf_buffer_cache_time_ms
  use_sim_time
```

and ```ros2 param get rtabmap use_sim_time```
Boolean value is: True

matlabbe, 2023-12-23:
Are you using ros2 rtabmap binaries? I see version 0.21.1. There were some refactoring done to change when this message is shown. You may try latest code.

yuan-0816, 2024-01-03:
How to update the latest code.

matlabbe, 2024-01-03:
See build from source here: https://github.com/introlab/rtabmap_ros/tree/ros2#from-source

yuan-0816, 2024-01-09:
It still the same, but I think the problem is from realsense gazebo plugin.
When I try to use gazebo depth camera plugin, it is working!
Thank you><

closed 2024-01-19 by matlabbe

asandula, 2024-09-12:
Hello @yuan-0816 Can you please share if there is any solution for simulating the realsense camera in gazebo, with topics set to sim time? It would be great help if you can help us.

yuan-0816, 2024-09-12:
You can see here 
https://github.com/AirLab-Drone/drone_vslam/issues/1#issuecomment-1999804903

hope this will help U