introlab/rtabmap_ros #1071: rtabmap cannot function properly with rosbag

AddyPete, 2023-11-24:
ROS2 Version: Humble
RTABMap Version: RTABMap ROS2 Latest

[Video Link](https://youtu.be/sdkeO5Dv1PU)

Mapping with RTABMap works properly and smoothly with Zed Wrapper when I do it live.

However when I try to do mapping with the rosbag recorded with zed wrapper, it can't function properly.

During the recording process I did this steps:
1. 
`ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i`
- to launch the zed wrapper
2. 
`ros2 bag record -a `
- to record all topics.

Next, I tried mapping with the rosbag with:
1.
```
ros2 launch rtabmap_launch rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start" \
    rgb_topic:=/zed/zed_node/rgb/image_rect_color \
    depth_topic:=/zed/zed_node/depth/depth_registered \
    camera_info_topic:=/zed/zed_node/rgb/camera_info \
    frame_id:=zed_camera_center \
    approx_sync:=false \
    wait_imu_to_init:=true \
    imu_topic:=/zed/zed_node/imu/data \
    qos:=1 \
    use_sim_time:=true \
    rviz:=true
```
    
Then play the rosbag
2.  
`ros2 bag play zed_office_test_1/zed_office_test_1_0.db3`

However, mapping won't properly function with errors given such as TF_OLD_DATA and imu_link error messages. I also tried not adding use_sim_time:=false, but still won't work properly. You can check the video for reference. Thank you!

AddyPete, 2023-11-24:
I also tried adding --clock flag in rosbag
at the same time adding use_sim_time:=true in rtabmap launch 

but it produces another kind of error

```
[rgbd_odometry-1] [ERROR] [1700787162.614928777] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_center", TF not available at time 1700553808.332129
[rgbd_odometry-1] [ERROR] [1700787163.368926398] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_center", TF not available at time 1700553809.046681
[rgbd_odometry-1] [INFO] [1700787163.608385196] [rtabmap.rgbd_odometry]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.050037s
[rtabmap-2] [INFO] [1700787163.609433973] [rtabmap.rtabmap]: rtabmap (1): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[rgbd_odometry-1] [ERROR] [1700787164.196360636] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_center", TF not available at time 1700553809.917001
[rgbd_odometry-1] [ERROR] [1700787164.621871336] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_center", TF not available at time 1700553810.317546
[rgbd_odometry-1] [ERROR] [1700787164.814506473] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_center", TF not available at time 1700553810.532654
[rgbd_odometry-1] [ERROR] [1700787165.645171164] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_center", TF not available at time 1700553811.341161

```

AddyPete, 2023-11-24:
[image]

I also tried 
**frame_id:=zed_camera_link** 
instead of zed_camera_center

it maps, but still laggy and odom will lost eventually. It has also errors:

```
[rgbd_odometry-1] [ERROR] [1700787376.131512555] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_link", TF not available at time 1700553820.247576
[rgbd_odometry-1] [ERROR] [1700787376.708945620] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_link", TF not available at time 1700553820.800856
[rgbd_odometry-1] [ERROR] [1700787377.906563152] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_link", TF not available at time 1700553821.986814
[rgbd_odometry-1] [ERROR] [1700787378.109684955] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_link", TF not available at time 1700553822.221701
[rgbd_odometry-1] [ERROR] [1700787378.688240381] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_link", TF not available at time 1700553822.797794
[rgbd_odometry-1] [WARN] [1700787378.688394028] [rtabmap.rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rgbd_odometry-1] rgbd_odometry subscribed to (exact sync):
[rgbd_odometry-1]    /zed/zed_node/rgb/image_rect_color,
[rgbd_odometry-1]    /zed/zed_node/depth/depth_registered,
[rgbd_odometry-1]    /zed/zed_node/rgb/camera_info
[rgbd_odometry-1] [ERROR] [1700787379.063363964] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_link", TF not available at time 1700553823.203284
[rgbd_odometry-1] [ERROR] [1700787379.438308625] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "zed_camera_link", TF not available at time 1700553823.579104
[rgbd_odometry-1] [INFO] [1700787379.519100601] [rtabmap.rgbd_odometry]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.075001s
[rtabmap-2] [WARN] [1700787379.632034209] [rtabmap.rtabmap]: rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap-2] rtabmap subscribed to (exact sync):
[rtabmap-2]    /rtabmap/odom \
[rtabmap-2]    /zed/zed_node/rgb/image_rect_color \
[rtabmap-2]    /zed/zed_node/depth/depth_registered \
[rtabmap-2]    /zed/zed_node/rgb/camera_info \
[rtabmap-2]    /rtabmap/odom_info
[rtabmap_viz-3] [WARN] [1700787379.776429792] [rtabmap.rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap_viz-3] rtabmap_viz subscribed to (exact sync):
```

matlabbe, 2023-11-24:
If you do `ros2 bag info`, are the number of images recorded make sense for the length of the bag? (If we expect 30 Hz frames, for 10 seconds bag you should see 300 msgs for each camera topic).

I would not recommend doing `ros2 bag record -a` when there are publishers using image_transport because you will be recording the same images in all different compression plugins. This could make the computer not able to write fast enough, thus dropping frames.

AddyPete, 2023-11-24:
`Files:             zed_office_test_1/zed_office_test_1_0.db3
Bag size:          18.0 GiB
Storage id:        sqlite3
Duration:          87.41s
Start:             Nov 21 2023 16:03:04.397 (1700553784.397)
End:               Nov 21 2023 16:04:31.439 (1700553871.439)
Messages:          106256
Topic information: Topic: /diagnostics | Type: diagnostic_msgs/msg/DiagnosticArray | Count: 86 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 13 | Serialization Format: cdr
                   Topic: /tf | Type: tf2_msgs/msg/TFMessage | Count: 5110 | Serialization Format: cdr
                   Topic: /tf_static | Type: tf2_msgs/msg/TFMessage | Count: 1 | Serialization Format: cdr
                   Topic: /zed/robot_description | Type: std_msgs/msg/String | Count: 1 | Serialization Format: cdr
                   Topic: /zed/zed_node/atm_press | Type: sensor_msgs/msg/FluidPressure | Count: 2108 | Serialization Format: cdr
                   Topic: /zed/zed_node/confidence/confidence_map | Type: sensor_msgs/msg/Image | Count: 1026 | Serialization Format: cdr
                   Topic: /zed/zed_node/depth/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 941 | Serialization Format: cdr
                   Topic: /zed/zed_node/depth/depth_info | Type: zed_interfaces/msg/DepthInfoStamped | Count: 1267 | Serialization Format: cdr
                   Topic: /zed/zed_node/depth/depth_registered | Type: sensor_msgs/msg/Image | Count: 1012 | Serialization Format: cdr
                   Topic: /zed/zed_node/disparity/disparity_image | Type: stereo_msgs/msg/DisparityImage | Count: 1267 | Serialization Format: cdr
                   Topic: /zed/zed_node/imu/data | Type: sensor_msgs/msg/Imu | Count: 16262 | Serialization Format: cdr
                   Topic: /zed/zed_node/imu/data_raw | Type: sensor_msgs/msg/Imu | Count: 16272 | Serialization Format: cdr
                   Topic: /zed/zed_node/imu/mag | Type: sensor_msgs/msg/MagneticField | Count: 4237 | Serialization Format: cdr
                   Topic: /zed/zed_node/left/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1255 | Serialization Format: cdr
                   Topic: /zed/zed_node/left/image_rect_color | Type: sensor_msgs/msg/Image | Count: 1275 | Serialization Format: cdr
                   Topic: /zed/zed_node/left_cam_imu_transform | Type: geometry_msgs/msg/TransformStamped | Count: 16580 | Serialization Format: cdr
                   Topic: /zed/zed_node/left_gray/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1248 | Serialization Format: cdr
                   Topic: /zed/zed_node/left_gray/image_rect_gray | Type: sensor_msgs/msg/Image | Count: 1249 | Serialization Format: cdr
                   Topic: /zed/zed_node/left_raw/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1255 | Serialization Format: cdr
                   Topic: /zed/zed_node/left_raw/image_raw_color | Type: sensor_msgs/msg/Image | Count: 1237 | Serialization Format: cdr
                   Topic: /zed/zed_node/left_raw_gray/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1248 | Serialization Format: cdr
                   Topic: /zed/zed_node/left_raw_gray/image_raw_gray | Type: sensor_msgs/msg/Image | Count: 1249 | Serialization Format: cdr
                   Topic: /zed/zed_node/odom | Type: nav_msgs/msg/Odometry | Count: 1268 | Serialization Format: cdr
                   Topic: /zed/zed_node/odom/status | Type: zed_interfaces/msg/PosTrackStatus | Count: 1267 | Serialization Format: cdr
                   Topic: /zed/zed_node/path_map | Type: nav_msgs/msg/Path | Count: 167 | Serialization Format: cdr
                   Topic: /zed/zed_node/path_odom | Type: nav_msgs/msg/Path | Count: 168 | Serialization Format: cdr
                   Topic: /zed/zed_node/point_cloud/cloud_registered | Type: sensor_msgs/msg/PointCloud2 | Count: 868 | Serialization Format: cdr
                   Topic: /zed/zed_node/pose | Type: geometry_msgs/msg/PoseStamped | Count: 1262 | Serialization Format: cdr
                   Topic: /zed/zed_node/pose/status | Type: zed_interfaces/msg/PosTrackStatus | Count: 1267 | Serialization Format: cdr
                   Topic: /zed/zed_node/pose_with_covariance | Type: geometry_msgs/msg/PoseWithCovarianceStamped | Count: 1264 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1215 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb/image_rect_color | Type: sensor_msgs/msg/Image | Count: 1250 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb_gray/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1250 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb_gray/image_rect_gray | Type: sensor_msgs/msg/Image | Count: 1249 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb_raw/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1249 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb_raw/image_raw_color | Type: sensor_msgs/msg/Image | Count: 1257 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb_raw_gray/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1241 | Serialization Format: cdr
                   Topic: /zed/zed_node/rgb_raw_gray/image_raw_gray | Type: sensor_msgs/msg/Image | Count: 1244 | Serialization Format: cdr
[... cut ...]
                   
                   
I recorded for about 1 minute +, so I think the number if images does not match with the length of the recording. 

Question:
1. should I use --use-sim-time flag in when doing ros2 bag record?
2. In rgbd mapping. should I only record the topic

```
    /zed/zed_node/rgb/image_rect_color
    /zed/zed_node/depth/depth_registered
    /zed/zed_node/rgb/camera_info
    /zed/zed_node/imu/data
```
? Only this four topics needed right when recording in a ros2 bag?

matlabbe, 2023-11-24:
Note that to avoid TF conflict with odom and map frames, disable them when launching zed wrapper. Just tried with:
```
ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2 publish_tf:=false publish_map_tf:=false

ros2 bag record /tf /tf_static \
  /zed/zed_node/rgb/image_rect_color \
  /zed/zed_node/depth/depth_registered \
  /zed/zed_node/rgb/camera_info \
  /zed/zed_node/imu/data
```

The mapping would work whether you set `use_sim_time` to true or false, however without `use_sim_time`, you may see warnings that rtabmap is not receiving data, but it does. 
```
ros2 launch rtabmap_launch rtabmap.launch.py  \
   rtabmap_args:="--delete_db_on_start"  \
   rgb_topic:=/zed/zed_node/rgb/image_rect_color  \
   depth_topic:=/zed/zed_node/depth/depth_registered  \
   camera_info_topic:=/zed/zed_node/rgb/camera_info  \
   frame_id:=zed_camera_link  
   approx_sync:=false  \
   wait_imu_to_init:=true  \
   imu_topic:=/zed/zed_node/imu/data  \
   qos:=1  \
   use_sim_time:=true
```

I got also this [issue](https://github.com/ros2/rosbag2/issues/1049) (my computer is on Foxy), but I solved it by remapping `/tf_static` to `/tf`:
```
ros2 bag play rosbag2_2023_11_23-18_42_35_0.db3 --remap /tf_static:=/tf
```

AddyPete, 2023-11-24:
thank you very much. it works now!

closed 2023-11-24 by AddyPete