introlab/rtabmap_ros #1080: Request for Assistance with RTAB-Map and Realsense L515 on ROS Noetic

win895564, 2023-12-06:
I have installed Ubuntu version 20.04 with ROS Noetic, and I'm trying to implement RTAB-Map SLAM using the Realsense L515. I've already installed RTAB-Map using `sudo apt install ros-noetic-rtabmap` and I can run RTAB-Map, but there is no image display. However, I keep encountering errors when entering the following commands. I'm earnestly seeking help from someone kind. Thank you.

```
rosrun nodelet nodelet standalone rtabmap_ros/point_cloud_xyz \
    _approx_sync:=false  \
    /depth/image:=/camera/depth/image_rect_raw \
    /depth/camera_info:=/camera/depth/camera_info \
    _decimation:=4
type is rtabmap_ros/point_cloud_xyz
[ERROR] [1701852626.770658304]: Failed to load nodelet [/rtabmap_ros_point_cloud_xyz] of type [rtabmap_ros/point_cloud_xyz] even after refreshing the cache: According to the loaded plugin descriptions the class rtabmap_ros/point_cloud_xyz with base class type nodelet::Nodelet does not exist. Declared types are RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet apriltag_ros/ContinuousDetector imu_complementary_filter/ComplementaryFilterNodelet imu_filter_madgwick/ImuFilterNodelet pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid realsense2_camera/RealSenseNodeFactory rtabmap_legacy/data_odom_sync rtabmap_legacy/data_throttle rtabmap_legacy/obstacles_detection_old rtabmap_legacy/stereo_throttle rtabmap_legacy/undistort_depth rtabmap_odom/icp_odometry rtabmap_odom/rgbd_odometry rtabmap_odom/rgbdicp_odometry rtabmap_odom/stereo_odometry rtabmap_slam/rtabmap rtabmap_sync/rgb_sync rtabmap_sync/rgbd_sync rtabmap_sync/rgbdx_sync rtabmap_sync/rtabmap rtabmap_sync/stereo_sync rtabmap_util/disparity_to_depth rtabmap_util/imu_to_tf rtabmap_util/lidar_deskewing rtabmap_util/obstacles_detection rtabmap_util/point_cloud_aggregator rtabmap_util/point_cloud_assembler rtabmap_util/point_cloud_xyz rtabmap_util/point_cloud_xyzrgb rtabmap_util/pointcloud_to_depthimage rtabmap_util/rgb_sync rtabmap_util/rgbd_relay rtabmap_util/rgbd_split rtabmap_util/stereo_sync
```
Please help me add this passage into the document and convert it into a Word file. Thank you.

matlabbe, 2023-12-10:
```
rosrun nodelet nodelet standalone rtabmap_ros/point_cloud_xyz 
```
doesn't exist anymore. It is now:
```
rosrun nodelet nodelet standalone rtabmap_util/point_cloud_xyz 
```
See [migration ](http://wiki.ros.org/rtabmap_ros/noetic_and_newer#Migration_Guide_New_Interface_Noetic.2FROS2)guide in case you have other similar errors.

win895564, 2023-12-11:
Original issue resolved, thank you! But now when I try to run RTAB-Map ROS, I encounter a new problem that prevents RTAB-Map ROS from functioning properly and there's no display. My launch command includes parameters like --Icp/VoxelSize 0.05, --Icp/PointToPlaneRadius 0, etc. However, I keep receiving warnings about topics like /scan and /cloud not receiving data for 5 seconds. I saw a solution in this ROS Answers post suggesting to change approx_sync to true. Even after following this advice, the problem persists. Please help!

> roslaunch rtabmap_ros rtabmap.launch\
>     rtabmap_args:="\
>       --delete_db_on_start \
>       --Icp/VoxelSize 0.05 \
>       --Icp/PointToPlaneRadius 0 \
>       --Icp/PointToPlaneK 20 \
>       --Icp/CorrespondenceRatio 0.2 \
>       --Icp/PMOutlierRatio 0.65 \
>       --Icp/Epsilon 0.005 \
>       --Icp/PointToPlaneMinComplexity 0 \
>       --Odom/ScanKeyFrameThr 0.7 \
>       --OdomF2M/ScanMaxSize 15000 \
>       --Optimizer/GravitySigma 0.3 \
>       --RGBD/ProximityPathMaxNeighbors 1 \
>       --Reg/Strategy 1" \
>     icp_odometry:=true \
>     scan_cloud_topic:=/cloud \
>     subscribe_scan_cloud:=true \
>     depth_topic:=/camera/aligned_depth_to_color/image_raw \
>     rgb_topic:=/camera/color/image_raw \
>     camera_info_topic:=/camera/color/camera_info \
>     approx_sync:=true \
>     wait_imu_to_init:=true \
>     imu_topic:=/rtabmap/imu

> [ WARN] [1702233598.826564204]: /rtabmap/icp_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
> /rtabmap/icp_odometry subscribed to /scan and /cloud (make sure only one of this topic is published, otherwise remap one to a dummy topic name).
> [ WARN] [1702233599.225060963]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
> /rtabmap/rtabmap subscribed to (exact sync):
>    /rtabmap/odom    /camera/color/image_raw    /camera/aligned_depth_to_color/image_raw    /camera/color/camera_info    /cloud    /rtabmap/odom_info
> [ WARN] [1702233599.834856268]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
> /rtabmap/rtabmap_viz subscribed to (exact sync):
>    /rtabmap/odom    /camera/color/image_raw    /camera/aligned_depth_to_color/image_raw    /camera/color/camera_info    /rtabmap/odom_filtered_input_scan    /rtabmap/odom_info

matlabbe, 2023-12-16:
Is `/cloud` published?
```
rostopic hz  /cloud 
```