introlab/rtabmap_ros #1081: Build failing on ROS1/noetic when generating msgs: AttributeError: module 'em' has no attribute 'RAW_OPT'

matlabbe, 2023-12-10:
Current build is failing on noetic:
https://github.com/introlab/rtabmap_ros/actions/runs/7154252311/job/19481531490

Seems that someone has the same issue on melodic yesterday: https://www.reddit.com/r/ROS/comments/18dubcj/traceback_error_when_i_do_catkin_make/

Maybe python update happened and broke `genmsg`?

Full log:
```
Traceback (most recent call last):
  File "/opt/ros/noetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 49, in <module>

                                                                                
    genmsg.template_tools.generate_from_command_line_options(
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options

                                                                                
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file

                                                                                
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 93, in _generate_msg_from_file

                                                                                
    _generate_from_spec(input_file,
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec

                                                                                
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: module 'em' has no attribute 'RAW_OPT'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 72, in apport_excepthook

                                                                                
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>

                                                                                
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 32, in <module>

                                                                                
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 12, in <module>

                                                                                
    import os, glob, subprocess, os.path, time, pwd, sys, requests_unixsocket

                                                                                
  File "/usr/lib/python3/dist-packages/requests_unixsocket/__init__.py", line 1, in <module>

                                                                                
    import requests

                                                                                
  File "/usr/lib/python3/dist-packages/requests/__init__.py", line 95, in <module>

                                                                                
    from urllib3.contrib import pyopenssl

                                                                                
  File "/usr/lib/python3/dist-packages/urllib3/contrib/pyopenssl.py", line 46, in <module>

                                                                                
    import OpenSSL.SSL

                                                                                
  File "/usr/lib/python3/dist-packages/OpenSSL/__init__.py", line 8, in <module>

                                                                                
    from OpenSSL import crypto, SSL

                                                                                
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1553, in <module>

                                                                                
    class X509StoreFlags(object):

                                                                                
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1573, in X509StoreFlags

                                                                                
    CB_ISSUER_CHECK = _lib.X509_V_FLAG_CB_ISSUER_CHECK

                                                                                
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_CB_ISSUER_CHECK'

                                                                                

                                                                                
Original exception was:

                                                                                
Traceback (most recent call last):

                                                                                
  File "/opt/ros/noetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 49, in <module>

                                                                                
    genmsg.template_tools.generate_from_command_line_options(

                                                                                
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options

                                                                                
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)

                                                                                
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file

                                                                                
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)

                                                                                
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 93, in _generate_msg_from_file

                                                                                
    _generate_from_spec(input_file,

                                                                                
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
[build 10:02.5 s] [3/17 complete] [4/4 jobs] [1 queued] [rtabmap_msgs:make ... 
```

THOM-AwS, 2023-12-13:
https://github.com/PX4/PX4-Autopilot/issues/22515

a rollback of empy to 3.3.4 got me past this. 

``` bash
pip3 uninstall empy
pip3 install empy==3.3.4
```

matlabbe, 2023-12-19:
Linking other issues:
https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1519
https://stackoverflow.com/a/77656642

Added the fix in commit above

matlabbe, 2023-12-19:
fixed by https://github.com/introlab/rtabmap_ros/commit/377c9e1a2219b6f7c1aa1c9c602042d12b506df3

closed 2023-12-19 by matlabbe

w12l3-c, 2024-04-08:
Hi I still have this error of module em has not attribute RAW_OUTPUT after rolling back to empy 3.3.4, any idea?

  File "/opt/ros/noetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 49, in <module>
    genmsg.template_tools.generate_from_command_line_options(
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 93, in _generate_msg_from_file
    _generate_from_spec(input_file,
  File "/opt/ros/noetic/lib/python3/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: module 'em' has no attribute 'RAW_OPT'

matlabbe, 2024-04-09:
Maybe there is another empy version on your python path. I think `RAW_OPT` should exist with 3.3.4.

w12l3-c, 2024-04-09:
I have accidentally downloaded 4.0 before but I made sure I uninstall it first before installing the 3.3.4. I am using a venv, should I just rebuild the venv?

matlabbe, 2024-04-11:
Note sure how your environment is setup, but you would need to make sure ros finds the 3.3.4 version in the current python environment.