introlab/rtabmap_ros #1085: Frame id unexpected change

kosmonauta144, 2023-12-14:
Hi :smile: 

I'm using Ubuntu 22.04 with ROS Humble.

I am using Rtabmap with laser scanner and Realsense D435i camera. Until now, I launched everything with tmux script or from commandline and it worked alright. But in the longrun I would like to leave tmux behind and use only launch file and then interesting thing happen.

I changed default parameters in rtabmap.launch.py to fit with my topic names. Then I've included this in my launch file  
```
    rtabmap_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
        get_package_share_directory('robot_autonomy'), 'launch'),
        '/rtabmap_mapping_launch.py']))

```
Frame_id is set to base_link, but when started from the launch file, in terminal I got a message that frame_id is set to laser_link
`[icp_odometry-14] [INFO] [1702565066.418044611] [rtabmap.icp_odometry]: Odometry: frame_id = laser_link`, 
`[rtabmap-15] [INFO] [1702565066.477872890] [rtabmap.rtabmap]: rtabmap: frame_id  = laser_link
`
 and 
`[rtabmap.rtabmap]: Could not convert rgb/depth msgs! Aborting rtabmap update...`

I have changed only default values, however it seams that when including rtabmap launch file in my own launch file, something goes wrong. I would be grateful for hints what did I miss.

Below my main launch file which includes the same rtabmap launch which I call from commandline or tmux script.

```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import RegisterEventHandler, DeclareLaunchArgument, SetEnvironmentVariable, IncludeLaunchDescription, TimerAction
from launch.substitutions import LaunchConfiguration
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.event_handlers import OnExecutionComplete

def generate_launch_description():

    basic_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
        get_package_share_directory('robot_autonomy'), 'launch'),
        '/basic_launch.py']))
    
    sensors_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
         get_package_share_directory('robot_autonomy'), 'launch'),
         '/sensors_launch.py']))
    
    control_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
        get_package_share_directory('robot_control'), 'launch'),
        '/diffbot.launch.py']))
    
    twist_mux_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
        get_package_share_directory('robot_autonomy'), 'launch'),
        '/tmux_launch.py']))
    
    rtabmap_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
        get_package_share_directory('robot_autonomy'), 'launch'),
        '/rtabmap_mapping_launch.py']))

    nav2_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
         get_package_share_directory('robot_autonomy'), 'launch'),
         '/navigation_launch.py']))

    vizanti_include = IncludeLaunchDescription(PythonLaunchDescriptionSource([os.path.join(
        get_package_share_directory('vizanti_server'), 'launch'),
        '/vizanti_server.launch.py']))
    

    return LaunchDescription([

        #include nav2 launch
        basic_include,
        sensors_include,
        twist_mux_include,
        vizanti_include,
        Node(
            package='vesc_service',
            executable='service',
            name='vesc_service'
        ),
        Node(
            package= 'micro_ros_agent',
            executable= 'micro_ros_agent',
            name='micro_ros_agent',
            parameters=[{
                "serial": "--dev /dev/serial/by-path/pci-0000:00:14.0-usb-0:3.3:1.0-port0"
            }]
        ),
        TimerAction(
            period= 5.0,
            actions= [(control_include)]
        ),
        TimerAction(
            period= 10.0,
            actions= [(rtabmap_include)]
        ),
        TimerAction(
            period = 15.0,
            actions= [(nav2_include)]
        ),
    ])

```

matlabbe, 2023-12-16:
What does `rtabmap_mapping_launch.py` look like ? I tried doing the same by including `rtabmap.launch.py` from `rtabmap_launch` package and changing the `frame_id` inside that file and it worked as expected. The frame `laser_link` is not something hard-coded somewhere in the code, it would mean that parameter (`frame_id`) was set before launching the launch file. You may do `ros2 param list` to see if that parameter is already set.

kosmonauta144, 2023-12-18:
Hi @matlabbe,

thank you for your quick reply. I pasted `rtabmap_mapping_launch.py` below

```
#
# To avoid log buffering:
# "stdbuf -o L ros2 launch rtabmap_launch rtabmap.launch.py ..."
#

import os

from launch import LaunchDescription, Substitution, LaunchContext
from launch.actions import DeclareLaunchArgument, SetEnvironmentVariable, LogInfo, OpaqueFunction
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir, PythonExpression
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from launch_ros.actions import SetParameter
from typing import Text
from ament_index_python.packages import get_package_share_directory

#Based on https://answers.ros.org/question/363763/ros2-how-best-to-conditionally-include-a-prefix-in-a-launchpy-file/
class ConditionalText(Substitution):
    def __init__(self, text_if, text_else, condition):
        self.text_if = text_if
        self.text_else = text_else
        self.condition = condition

    def perform(self, context: 'LaunchContext') -> Text:
        if self.condition == True or self.condition == 'true' or self.condition == 'True':
            return self.text_if
        else:
            return self.text_else
            
class ConditionalBool(Substitution):
    def __init__(self, text_if, text_else, condition):
        self.text_if = text_if
        self.text_else = text_else
        self.condition = condition

    def perform(self, context: 'LaunchContext') -> bool:
        if self.condition:
            return self.text_if
        else:
            return self.text_else
            
def launch_setup(context, *args, **kwargs):      
               
    return [
        DeclareLaunchArgument('depth', default_value=ConditionalText('false', 'true', IfCondition(PythonExpression(["'", LaunchConfiguration('stereo'), "' == 'true'"]))._predicate_func(context)), description=''),
        DeclareLaunchArgument('subscribe_rgb', default_value=LaunchConfiguration('depth'), description=''),
        DeclareLaunchArgument('args',  default_value=LaunchConfiguration('rtabmap_args'), description='Can be used to pass RTAB-Map\'s parameters or other flags like --udebug and --delete_db_on_start/-d'),
        DeclareLaunchArgument('qos_image',       default_value=LaunchConfiguration('qos'), description='Specific QoS used for image input data: 0=system default, 1=Reliable, 2=Best Effort.'),
        DeclareLaunchArgument('qos_camera_info', default_value=LaunchConfiguration('qos'), description='Specific QoS used for camera info input data: 0=system default, 1=Reliable, 2=Best Effort.'),
        DeclareLaunchArgument('qos_scan',        default_value=LaunchConfiguration('qos'), description='Specific QoS used for scan input data: 0=system default, 1=Reliable, 2=Best Effort.'),
        DeclareLaunchArgument('qos_odom',        default_value=LaunchConfiguration('qos'), description='Specific QoS used for odometry input data: 0=system default, 1=Reliable, 2=Best Effort.'),
        DeclareLaunchArgument('qos_user_data',   default_value=LaunchConfiguration('qos'), description='Specific QoS used for user input data: 0=system default, 1=Reliable, 2=Best Effort.'),
        DeclareLaunchArgument('qos_imu',         default_value=LaunchConfiguration('qos'), description='Specific QoS used for imu input data: 0=system default, 1=Reliable, 2=Best Effort.'),
        DeclareLaunchArgument('qos_gps',         default_value=LaunchConfiguration('qos'), description='Specific QoS used for gps input data: 0=system default, 1=Reliable, 2=Best Effort.'),
        
        #These arguments should not be modified directly, see referred topics without "_relay" suffix above
        DeclareLaunchArgument('rgb_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('rgb_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('rgb_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('depth_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('depth_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('depth_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('left_image_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('left_image_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('left_image_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('right_image_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('right_image_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('right_image_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('rgbd_topic_relay',      default_value=ConditionalText(''.join(LaunchConfiguration('rgbd_topic').perform(context)), ''.join([LaunchConfiguration('rgbd_topic').perform(context), "_relay"]), LaunchConfiguration('rgbd_sync').perform(context)), description='Should not be modified manually!'),
    
        SetParameter(name='use_sim_time', value=LaunchConfiguration('use_sim_time')),
        # 'use_sim_time' will be set on all nodes following the line above
    
[... cut ...]
Only place where I set up frame_id is separate launch file for sensors, where I turn on the rplidar with this command

```
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([str(rplidar_launch_src), '/rplidar_a2m8_launch.py']),
                launch_arguments={'frame_id': 'laser_link', 'serial_port': '/dev/laser_usb', 'scan_mode':'Standard'}.items(),
        ),
```
Is it possible that the parameter is somehow passed to rtabmap? Sensors are started before `rtabmap` is launched. I would like to add that I am a fan of the algorithm and detailed documentation

closed 2023-12-23 by matlabbe