introlab/rtabmap_ros #1086: build error with save_objects_example on ros2

takahiko-hasegawa-hdjp, 2023-12-15:
Im trying to use svae_objects_example and then I have an error when i try to do a build. 

  ament_target_dependencies() the passed package name 'rtabmap_common' was
  not found before
Call Stack (most recent call first):
  CMakeLists.txt:481 (ament_target_dependencies)

what is the problem?

takahiko-hasegawa-hdjp, 2023-12-19:
Thanks for that! now the apriltags msg error is solved but another problem.

the version of save_object_tags.py that I have seems to use ROS1 interface as it  says ROSPY.

As im using ROS2 interface, is there a ros2 version?

matlabbe, 2023-12-23:
Where is that python script? Converting from rospy to rclpy is not too hard, look at this page https://docs.ros.org/en/rolling/How-To-Guides/Migrating-from-ROS1/Migrating-Python-Packages.html

For the c++, that node had not been ported to ros2: https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_demos/src/SaveObjectsExample.cpp, here is the cpp migration guide: https://docs.ros.org/en/foxy/The-ROS2-Project/Contributing/Migration-Guide.html