introlab/rtabmap_ros #1087: Offset of the map made using rtabmap_costmap_plugins::StaticLayer

JoachimDenantes, 2023-12-15:
Hi, I am using ROS noetic, producing gridmap with RTAB-Map and trying to use this map to make a global map in move_base. So I am using the plugin rtabmap_costmap_plugins::StaticLayer as you can see in costmap_global_laser.yaml.
You can see the result in the image of rviz I sent. The global map is build correctly using the RTAB-Map gridmap, but it is offset, not aligned with the rest. Any idea on how to fix this ?
[image]
[image]
[image]

matlabbe, 2023-12-16:
I think `static_map` should be `true` for the global costmap. The `width` and `height` don't need to be explicitly set. `rolling_window` should be false.

JoachimDenantes, 2023-12-18:
It work! Thank you very much

closed 2023-12-19 by matlabbe