introlab/rtabmap_ros #1088: fatal error: rtabmap/core/LocalGrid.h: no such file or directory，But I'm on rtabmap_ The header file path was added to the Cmakelist.txt folder in the util folder, but it still reported an error. Why

2467ly, 2023-12-19:
n file included from /home/ly/a_ws/xju-robot/src/thirdparty/rtabmap_ros/rtabmap_util/src/MapsManager.cpp:27:
/home/ly/a_ws/xju-robot/src/thirdparty/rtabmap_ros/rtabmap_util/include/rtabmap_util/MapsManager.h:34:10: fatal error: rtabmap/core/LocalGrid.h: 没有那个文件或目录
   34 | #include <rtabmap/core/LocalGrid.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

matlabbe, 2023-12-19:
Make sure to use latest master from rtabmap too.

WenyuLWY, 2023-12-28:
I met the same question. 

For rtabmap_ros and rtabmap, I use the default master branch. But when I turned rtabmap_ros to noetic-devel branch (i use noetic), the error was resolved.

I am not sure if there are potential risks but it did work

matlabbe, 2024-01-02:
Did `rtabmap` has been re-installed? `rtabmap/core/LocalGrid.h` has been added in latest version. The master branch of rtabmap and rtabmap_ros should be noetic compatible (as shown in this [build](https://github.com/introlab/rtabmap_ros/actions/runs/7265100510/job/19794068784)).

LightProhvet, 2024-01-04:
I am also getting this error. Similar to @WenyuLWY I managed to build using humble-devel branch, which isn't up-to-date - hence successful build. I tried re-installing many times, and so for me atleast the rtabmap was latest version. The latest build of rtabmap_ros (more specifically the latest commit: 2f365ab7 on 18.12.23 at 08:46) seems to be the problem. 

I checked, the LocalGrid.h file exists and is built along with the other .h files included from rtabmap/core in MapsManager.h (in rtabmap_util/include) so this error is very illogical as this is not the first include from rtabmap/core (and therefore shouldn't be a dependency issue)

@matlabbe i have little experience and little idea what else could cause this error if 1) the file exists 2) other similar #include-s work. So i can provide more information only based on questions.

ros2 doctor gives warnings that 3 packages have been updated (and i haven't): robot_localization, rqt_joint_trajectory_controller, behaviortree_cpp_v3

matlabbe, 2024-01-04:
rtabmap_util (though rtabmap_conversions) may have find another rtabmap installed somewhere not having the header file. The humble-devel branch has been updated yesterday to use latest code. I also updated the minimum rtabmap version to 0.21.3 so that cmake/colcon should reject any other rtabmap found with lower version.

LightProhvet, 2024-01-04:
Yes, that seemed to fix it. util built succesfully. I think this is now solved?

closed 2024-01-06 by matlabbe