introlab/rtabmap_ros #1089: Could not estimate motion of velodyne accordingly to fixed frame velodyne_stabilized between stamps 1703147510.130089 and 1703147510.180440! (Lookup would require extrapolation 0.022364622s into the future.  Requested time 1703147510.130089283 but the latest data is at time 1703147510.107724667, when looking up transform from frame [velodyne_stabilized] to frame [velodyne]. canTransform returned after 0.0100917 timeout was 0.01.

khuechuong, 2023-12-21:
Hi! So I am following this code here:

https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_examples/launch/test_velodyne_d435i_deskewing.launch

instead of D camera, I use zed mini since I don't have a D cam, since the zed depth wasn't giving me what I wanted. 

Here is my modification:
```
<arg name="rtabmap_viz"   default="true"/>
  <arg name="scan_20_hz"    default="false"/> <!-- If we launch the velodyne with "rpm:=1200" argument -->
  <arg name="deskewing"     default="true"/>
  <arg name="slerp"         default="false"/> <!-- If true, a slerp between the first and last time will be used to deskew each point, which is faster than using tf for every point but less accurate -->
  <arg name="scan_topic"    default="/velodyne_points"/>
  <arg name="use_sim_time"  default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>

  <arg name="imu_topic"       default="/zed/zed_nodelet/imu/data"/>
  <arg name="frame_id"        default="velodyne"/>       <!-- base frame of the robot: for this example, we use velodyne as base frame -->
  <arg name="queue_size"      default="10"/>
  <arg name="queue_size_odom" default="1"/>
  <arg name="loop_ratio"      default="0.2"/>

  <arg name="resolution"         default="0.05"/> <!-- set 0.05-0.3 for indoor, set 0.3-0.5 for outdoor -->
  <arg name="iterations"         default="10"/>

  <!-- Grid parameters -->
  <arg name="ground_is_obstacle" default="true"/>
  <arg name="grid_max_range"     default="20"/>

  <!-- For F2M Odometry -->
  <arg name="ground_normals_up" default="false"/> <!-- set to true when velodyne is always horizontal to ground (ground robot, car) -->
  <arg name="local_map_size"    default="15000"/>
  <arg name="key_frame_thr"     default="0.4"/>

  <!-- For FLOAM Odometry -->
  <arg name="floam"    default="false"/> <!-- RTAB-Map should be built with FLOAM http://official-rtab-map-forum.206.s1.nabble.com/icp-odometry-with-LOAM-crash-tp8261p8563.html -->
  <arg name="floam_sensor" default="0"/> <!-- 0=16 rings (VLP16), 1=32 rings, 2=64 rings -->

  <!-- Static transform between velodyne and D435i: TODO: Adjust with real position/orientation!!! -->
  <node unless="$(arg use_sim_time)" pkg="tf" type="static_transform_publisher" name="velodyne_to_base_link_tf" args="0.105, -0.070, -0.504 0, -0.07, 0 velodyne base_link 100"/>
  <!-- <node pkg="tf" type="static_transform_publisher" name="zed_to_velodyne"
        args="-0.2 0.07 0.12 0 0.07 0 zed_base_link velodyne 100" /> -->
  <!-- Velodyne sensor VLP16 -->
  <include unless="$(arg use_sim_time)" file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
     <arg     if="$(arg scan_20_hz)" name="rpm" value="1200"/>
     <arg unless="$(arg scan_20_hz)" name="rpm" value="600"/>
     <arg name="organize_cloud" value="true"/> <!-- should be organized for deskewing -->
  </include>

  <include file="$(find zed_nodelet_example)/launch/zed_laserscan_nodelet.launch" />
  <!-- IMU orientation estimation and publish tf accordingly to os_sensor frame -->
  <node pkg="nodelet" type="nodelet" name="imu_nodelet_manager" args="manager">
    <remap from="imu/data_raw" to="$(arg imu_topic)"/>
    <remap from="imu/data" to="$(arg imu_topic)/filtered"/>
  </node>
  <node pkg="nodelet" type="nodelet" name="imu_filter" args="load imu_filter_madgwick/ImuFilterNodelet imu_nodelet_manager">
    <param name="use_mag" value="false"/>
    <param name="world_frame" value="enu"/>
    <param name="publish_tf" value="false"/>
  </node>
  <node pkg="nodelet" type="nodelet" name="imu_to_tf" args="load rtabmap_util/imu_to_tf imu_nodelet_manager">
    <remap from="imu/data" to="$(arg imu_topic)/filtered"/>
    <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
    <param name="base_frame_id" value="$(arg frame_id)"/>
  </node>
  
  <node if="$(arg deskewing)" pkg="nodelet" type="nodelet" name="lidar_deskewing" args="standalone rtabmap_util/lidar_deskewing" output="screen">
    <param name="wait_for_transform" value="0.01"/>
    <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
    <param name="slerp" value="$(arg slerp)"/>
[... 97 lines cut ...]
    mc_bag:                     true                            # Enable/disable the detection of bags for 'MULTI_CLASS_BOX_X' models
    mc_animal:                  true                            # Enable/disable the detection of animals for 'MULTI_CLASS_BOX_X' models
    mc_electronics:             true                            # Enable/disable the detection of electronic devices for 'MULTI_CLASS_BOX_X' models
    mc_fruit_vegetable:         true                            # Enable/disable the detection of fruits and vegetables for 'MULTI_CLASS_BOX_X' models
    mc_sport:                   true                            # Enable/disable the detection of sport-related objects for 'MULTI_CLASS_BOX_X' models
```

Here's zedm.yaml:
```
general:
    camera_model:               'zedm'
    resolution:                 1           # '0': HD2K, '1': HD1080, '3': HD720, '5': VGA, '6': AUTO
    grab_frame_rate:            15          # Frequency of frame grabbing for internal SDK operations

depth:
    min_depth:                  0.2             # Min: 0.1, Max: 3.0 - Default 0.3 - Note: reducing this value wil require more computational power and GPU memory
    max_depth:                  20.0            # Max: 20.0  

```

Here's my TF tree as well:

[image]

Any help is appreciated! Thank you!

matlabbe, 2023-12-23:
I think zed IMU data has orientation already computed, you may remove the madgwick filter and connect directly the imu data to `imu_to_tf` node.

How much did you try to increase `wait_for_transform` (0.05s, 0.1s, 1s)? How bad was the extrapolation delay? Very large delay could be casued by poor synchronization between the clocks of zed camera and the velodyne. Is the velodyne clock synchronized with host clock, with a GPS clock?

khuechuong, 2023-12-24:
so basically don't have the velodyne_stabliizer? like:
```
<node pkg="nodelet" type="nodelet" name="imu_to_tf" args="load rtabmap_util/imu_to_tf imu_nodelet_manager">
    <remap from="imu/data" to="$(arg imu_topic)"/>
    <param name="fixed_frame_id" value="$(arg frame_id)"/>
    <param name="base_frame_id" value="$(arg frame_id)"/>
  </node>
  
  <node if="$(arg deskewing)" pkg="nodelet" type="nodelet" name="lidar_deskewing" args="standalone rtabmap_util/lidar_deskewing" output="screen">
    <param name="wait_for_transform" value="0.01"/>
    <param name="fixed_frame_id" value="$(arg frame_id)"/>
    <param name="slerp" value="$(arg slerp)"/>
    <remap from="input_cloud" to="$(arg scan_topic)"/>
  </node>
```

Also I have tried 0.05 and 0.01 and 0.5 but still the same issue. It seems to be like 0.001 sec in the future. I am not using GPS, just the velodyne.

matlabbe, 2023-12-24:
Keep the stabilized frame:
```xml
  <node pkg="nodelet" type="nodelet" name="imu_to_tf" args="load rtabmap_util/imu_to_tf imu_nodelet_manager">
    <remap from="imu/data" to="$(arg imu_topic)"/>
    <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
    <param name="base_frame_id" value="$(arg frame_id)"/>
  </node>
  
  <node if="$(arg deskewing)" pkg="nodelet" type="nodelet" name="lidar_deskewing" args="standalone rtabmap_util/lidar_deskewing" output="screen">
    <param name="wait_for_transform" value="0.01"/>
    <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
    <param name="slerp" value="$(arg slerp)"/>
    <remap from="input_cloud" to="$(arg scan_topic)"/>
  </node>
```

What is the warning using `wait_for_transform=1` ? It is strange if the extrapolation in the future is below 1 sec, even after waiting 1 sec to get.

khuechuong, 2024-01-05:
@matlabbe  
yes so I tried the code you put and ``` wait_for_transform=1``` but still same error:
[image]

matlabbe, 2024-01-06:
The error message is slightly different, it seems that TF velodyne_stabilized -> velodyne is just not published.

khuechuong, 2024-01-09:
@matlabbe I'm not sure why, I just did what you told me: 
remove the
```
<node pkg="nodelet" type="nodelet" name="imu_filter" args="load imu_filter_madgwick/ImuFilterNodelet imu_nodelet_manager">
    <param name="use_mag" value="false"/>
    <param name="world_frame" value="enu"/>
    <param name="publish_tf" value="false"/>
  </node>
  <node pkg="nodelet" type="nodelet" name="imu_to_tf" args="load rtabmap_util/imu_to_tf imu_nodelet_manager">
    <remap from="imu/data" to="$(arg imu_topic)/filtered"/>
    <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
    <param name="base_frame_id" value="$(arg frame_id)"/>
  </node>
```

part and add the

```
<node pkg="nodelet" type="nodelet" name="imu_to_tf" args="load rtabmap_util/imu_to_tf imu_nodelet_manager">
    <remap from="imu/data" to="$(arg imu_topic)"/>
    <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
    <param name="base_frame_id" value="$(arg frame_id)"/>
  </node>
  
  <node if="$(arg deskewing)" pkg="nodelet" type="nodelet" name="lidar_deskewing" args="standalone rtabmap_util/lidar_deskewing" output="screen">
    <param name="wait_for_transform" value="0.01"/>
    <param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
    <param name="slerp" value="$(arg slerp)"/>
    <remap from="input_cloud" to="$(arg scan_topic)"/>
  </node>
```

matlabbe, 2024-01-13:
Is the `imu_to_tf` still published? otherwise the timestamps may be very off than the ones used by the velodyne.