introlab/rtabmap_ros #1090: Building from source with CUDA support

coen132, 2023-12-22:
Hey, I'm trying to build the packages from source with cuda support enabled for g2o and pcl and opencv4. 

I have installed these:

https://pcl.readthedocs.io/projects/tutorials/en/master/gpu_install.html
https://github.com/opencv/opencv.git
https://github.com/RainerKuemmerle/g2o

With all of them I used "ccmake .." to configure CUDA and was able to compile. (after setting the cuda toolkit CUDACXX=path/etc)

Now I'm trying to build RTabmap and rtabmap_ros from source to link them against those libraries. Everything builds except one package: rtabmap_rviz_plugins "

The error I receive is:

--- stderr: rtabmap_rviz_plugins                                                                                                    
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt6Core/Qt6CoreVersionlessTargets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: Qt::Core

  Targets not yet defined: Qt::CorePrivate

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Qt6Core/Qt6CoreConfig.cmake:63 (include)
  /usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/Qt6/QtPublicDependencyHelpers.cmake:14 (find_dependency)
  /usr/lib/x86_64-linux-gnu/cmake/Qt6OpenGL/Qt6OpenGLDependencies.cmake:96 (_qt_internal_find_dependencies)
  /usr/lib/x86_64-linux-gnu/cmake/Qt6OpenGL/Qt6OpenGLConfig.cmake:50 (include)
  /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake:219 (find_package)
  /usr/local/lib/cmake/vtk-9.3/VTK-vtk-module-find-packages.cmake:585 (find_package)
  /usr/local/lib/cmake/vtk-9.3/vtk-config.cmake:159 (include)
  /usr/local/share/pcl-1.13/PCLConfig.cmake:272 (find_package)
  /usr/local/share/pcl-1.13/PCLConfig.cmake:321 (find_VTK)
  /usr/local/share/pcl-1.13/PCLConfig.cmake:564 (find_external_library)
  /usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /home/riwo/ros2_ws/install/rtabmap/lib/rtabmap-0.21/RTABMapConfig.cmake:7 (find_dependency)
  /home/riwo/ros2_ws/install/rtabmap_conversions/share/rtabmap_conversions/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/riwo/ros2_ws/install/rtabmap_conversions/share/rtabmap_conversions/cmake/rtabmap_conversionsConfig.cmake:41 (include)
  CMakeLists.txt:18 (find_package)

---
Failed   <<< rtabmap_rviz_plugins [8.04s, exited with code 1]
                         
Iknow rtabmap uses qt5 or 6 to visualize some stuff? Any help is appreciated.

matlabbe, 2023-12-24:
Which linux version are you using? Rtabmap would use the default Qt version (4 or 5 or 6) installed on the computer.

coen132, 2023-12-24:
Thanks for your response, I'm using 22.04 with ROS2 humble. Qt5 was installed by default and QT6 I installed extra but still received the error

coen132, 2023-12-25:
I think I solved it by removing the qt 6 and manually setting the version to qt5 for all dependencies. Rebuilding finished without issues! I'll close this issue

closed 2023-12-25 by coen132