introlab/rtabmap_ros #1091: Is using visual feature-based approaches?

kento77, 2023-12-24:
Sorry for my poor English.

I'm trying to understand how RTABMap works, but could you please help me with the following questions?

1.I am using depth camera realsense455. In this case, is the default setting of RTABMap using a method that uses reprojection error of visual feature-based approaches?
2. If so, is only the sensor orientation to be optimized? Or is the 3D point cloud of landmarks optimized in addition to the sensor orientation?

matlabbe, 2023-12-24:
1. Yes, default parameters should be fine, unless resolution is higher than 720p (parameters are tuned for 480p).
2. In visual odometry, both pose and 3D features are optimized.

kento77, 2023-12-24:
Thanks for the reply.
Is there a paper or something that has the formulas for question 2.

matlabbe, 2023-12-24:
This will give you the overview of the approach https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf