introlab/rtabmap_ros #1094: Reading Node poses from rtabmap.db

360wcui, 2024-01-02:
Hello,  I am trying to read the node poses from rtabmap.db directly as it is in sqlite format.   With https://inloop.github.io/sqlite-viewer/,   I notice that the size of the pose is 48 bytes.   Shouldn't it be geometry_msgs/Pose, which is 3 x 8 + 4 x 8 = 56 bytes, i.e., 3 transnational float64 values x, y, z,  4 float64 quaternions, qx, qy, qz, qw?   Not sure how to deserialize the 48 bytes to get the pose?  Any advice would be greatly appreciated.

360wcui, 2024-01-02:
In Node table,   for example,  the pose bytes in the rtabmap.db I get is 163,23,48,57,12,255,197,184,0,0,128,191,200,162,138,55,104,198,122,186,248,255,127,191,199,168,197,56,180,211,201,58,248,255,127,191,88,197,122,58,221,47,48,185,232,42,29,60. 

which should be xyz=(5.36652e-05,0.0015834,0.00954197),  rpy=(1.78297,1.56982,-1.35865),  but I am unable to replicate the doubles with the pose bytes.   [163,23,48,57,12,255,197,184] corresponds to -3.3096324082674704e-35.   Here is the JavaScript:
let pixels = new Uint8Array([163, 23, 48, 57, 12, 255, 197, 184]); // 
let floatArr = new Float64Array(pixels.buffer) // I 
console.log(floatArr) 

[image]
[image]

matlabbe, 2024-01-03:
We are saving the poses in 3x4 transformation matrix (R3x3 T3x1). They are in float (4 bytes per float), so 3x4 x4 = 48 bytes.

closed 2024-01-03 by 360wcui