introlab/rtabmap_ros #1098: Unable to locate package ros-humble-rtabmap-ros

kjgada, 2024-01-10:
Details - I have arm64 architecture device running Ubuntu 22.04 and ROS2 Humble.

Up until few days ago, I was able to download and install ros-humble-rtabmap-ros but as of today I get the error that it is unable to locate the package.

matlabbe, 2024-01-10:
We updated humble binaries recently. The problem is that `rtabmap_demos` is [now](https://github.com/introlab/rtabmap_ros/commit/0e4a16d1ca6f1d8aeeb52b4c53374f6941977bf5) depending on `turtlebot3_gazebo`, which is not available on arm64: https://build.ros2.org/job/Hbin_ujv8_uJv8__turtlebot3_gazebo__ubuntu_jammy_arm64__binary/

`rtabmap_ros` depends on `rtabmap_demos`, so it is why it is not there. However, as rtabmap_ros is a dummy package to install all rtabmap stack at once, you can still install individual packages. See http://repo.ros2.org/status_page/humble_default.html?q=ros-humble-rtabmap to see which ones are available on arm64 (last column).

I'll check how gracefully we could fix `rtabmap_demos` for arm64.

kjgada, 2024-01-12:
This worked, thank you :)

closed 2024-01-12 by kjgada