introlab/rtabmap_ros #1100: Undefined Symbol Issue : librtabmap_util_plugins.so: undefined symbol

BytePaul, 2024-01-11:
I am new to using RTABMAP. I would appreciate your input on this issue.

I keep getting the following issue : librtabmap_util_plugins.so: undefined symbol

Following is the output of the roslaunch command

ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [5285]
process[rtabmap/rgbd_sync-2]: started with pid [5286]
process[rtabmap/rtabmap-3]: started with pid [5287]
process[rtabmap/rtabmap_viz-4]: started with pid [5288]
process[move_base-5]: started with pid [5290]
process[rviz-6]: started with pid [5295]
[ INFO] [1704995482.612311611]: Initializing nodelet with 12 worker threads.
[ INFO] [1704995482.754017912]: Starting node...
[ INFO] [1704995482.783862640]: Initializing nodelet with 12 worker threads.
[ WARN] [1704995482.828191708, 65.247000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist. canTransform: source_frame base_footprint does not exist.. canTransform returned after 65.247 timeout was 0.1.
[ INFO] [1704995482.875662184, 65.297000000]: /rtabmap/rgbd_sync: approx_sync = true
[ INFO] [1704995482.875900640, 65.297000000]: /rtabmap/rgbd_sync: approx_sync_max_interval = 0.000000
[ INFO] [1704995482.876018101, 65.297000000]: /rtabmap/rgbd_sync: queue_size  = 10
[ INFO] [1704995482.876138858, 65.298000000]: /rtabmap/rgbd_sync: depth_scale = 1.000000
[ INFO] [1704995482.876221641, 65.298000000]: /rtabmap/rgbd_sync: decimation = 1
[ INFO] [1704995482.876302157, 65.298000000]: /rtabmap/rgbd_sync: compressed_rate = 0.000000
[ INFO] [1704995482.886863459, 65.307000000]: 
/rtabmap/rgbd_sync subscribed to (approx **sync):**
   /camera/rgb/image_raw \
   /camera/depth/image_raw \
   /camera/rgb/camera_info
**/home/paul/catkin_ws/devel/lib/rtabmap_slam/rtabmap: symbol lookup error: /home/paul/catkin_ws/devel/lib/librtabmap_util_plugins.so: undefined symbol: _ZN7rtabmap13OccupancyGridC1EPKNS_14LocalGridCacheERKSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESA_St4lessISA_ESaISt4pairIKSA_SA_EEE**
[ INFO] [1704995482.903820214]: Starting node...
[rtabmap/rtabmap-3] process has died [pid 5287, exit code 127, cmd /home/paul/catkin_ws/devel/lib/rtabmap_slam/rtabmap -d move_base:=/move_base scan:=/scan odom:=/odom rgbd_image:=rgbd_image grid_map:=/map __name:=rtabmap __log:=/home/paul/.ros/log/bf52919e-b0a9-11ee-82f8-91264e8526c5/rtabmap-rtabmap-3.log].
log file: /home/paul/.ros/log/bf52919e-b0a9-11ee-82f8-91264e8526c5/rtabmap-rtabmap-3*.log
[ INFO] [1704995483.028380273]: rtabmap_viz: Using configuration from "/home/paul/catkin_ws/src/rtabmap_ros/rtabmap_demos/launch/config/rgbd_gui.ini"
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1704995484.118719021, 66.527000000]: rtabmap_viz: Reading parameters from the ROS server...
[ INFO] [1704995484.120159332, 66.528000000]: rtabmap_viz: Cannot get rtabmap's parameters, waiting max 5 seconds in case the node has just been launched.

matlabbe, 2024-01-13:
You may have multiple rtabmap_util installed. Make sure to remove the binaries if you built it form source:
```
sudo apt remove ros-noetic-rtabmap*
```

BytePaul, 2024-01-13:
> You may have multiple rtabmap_util installed. Make sure to remove the binaries if you built it form source:
> 
> ```
> sudo apt remove ros-noetic-rtabmap*
> ```

I tried this but it did not work. It seems that there are no duplicate files. Now I get this error 

[ INFO] [1705140027.140963156]: rtabmap_viz: Using configuration from "/home/paul/catkin_ws/src/rtabmap_ros/rtabmap_demos/launch/config/rgbd_gui.ini"
[rtabmap/rtabmap-2] process has died [pid 3946, exit code -11, cmd /home/paul/catkin_ws/devel/lib/rtabmap_slam/rtabmap --delete_db_on_start scan:=/jn0/base_scan rgb/image:=/data_throttled_image depth/image:=/data_throttled_image_depth rgb/camera_info:=/data_throttled_camera_info __name:=rtabmap __log:=/home/paul/.ros/log/92ee08a2-b1fa-11ee-ac9d-975dd882d1d3/rtabmap-rtabmap-2.log].
log file: /home/paul/.ros/log/92ee08a2-b1fa-11ee-ac9d-975dd882d1d3/rtabmap-rtabmap-2*.log
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile

closed 2024-01-13 by BytePaul