introlab/rtabmap_ros #1104: Set param no taking effect on Humble.

brayanpa, 2024-01-18:
Hi,

I am trying to modify the RGBD/OptimizeMaxError parameter at runtime from for example 3.0 to 4.0. Using `ros2 param get /rtabmap RGBD/OptimizeMaxError` I can see that the change is taking effect. But from the rtabmap logs I still see the previous value, for example in the log:

`Rejecting localization ....... graph error ratio of 6.163492 (edge 72->74, type=0, abs error=23.730078 deg, stddev=0.067197). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.`

I have been trying to modify other parameters and they change but internally rtabmap continues with the same value. For example, `RGBD/MarkerDetection`, changing it from the launchfile file takes effect, but changing it at runtime is ignored.

This test worked in the latest Galactic version.

I'm building rtabmap on the commit e619c577595ad221890d550ea3dbe6cbdebe9502 and rtabmap_ros 2f365ab7e10025c4d13b7855dfb4b7c8bb01f2d3 .

matlabbe, 2024-01-20:
There was a bug that if the node was not started in a namespace, parameters events are ignored by the node. I fixed it in commit above. There was also an issue when setting parameters from command line, the type was not handled correctly, you would see 
```
[INFO] [1705785354.377991218] [rtabmap]: Parameters event received!
[INFO] [1705785354.378043818] [rtabmap]: Setting RTAB-Map parameter "RGBD/OptimizeMaxError"=""
[INFO] [1705785354.378052685] [rtabmap]: rtabmap: Updating parameters
[INFO] [1705785354.378063078] [rtabmap]: RTAB-Map rate detection = 1.000000 Hz
```
with
```
ros2 param set rtabmap RGBD/OptimizeMaxError 4
```
thus parameters would also be ignored. This has also been fixed in commit above. For previous versions, the workaround is to start a `rtabmap_viz` node in same namespace than `rtabmap` node and change parameters in Preferences.

brayanpa, 2024-01-25:
Thank you very much Mathieu, applying these changes fixed the issue.

closed 2024-01-25 by brayanpa