introlab/rtabmap_ros #1105: Problem with getting overlapped map

danyal2075, 2024-01-19:
I scan a corridor to create a map of the environment using rtabmap, The issue i am  facing is that when i go over the corridor and then come back, it creates an overlapping map. Anyone knows the solution please write to me!!!

another task i am struggling with is i would to create a path for my robot to define no-go areas in the map considering my robot size. it would be nice to help me with this as well

matlabbe, 2024-01-20:
For the first issue, it has been described in this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/8/87/LabbeAURO2017.pdf), figure 4.
[image]

If you don't have a lidar, either you need 2 cameras (forward and backward) to detect loop closures in both directions, or make sure to turn 180 deg again when you reach the beginning of the corridor so that a loop closure can be detected.

> no-go areas in the map considering my robot size

do you mean that the robot should stay far enough from the obstacles to not collide with them? I'll suggest to look at the [navigation stack](http://wiki.ros.org/navigation), with which a costmap with inflation layer can be created so that the planner make sure the robot doesn't collide with any obstacles.

danyal2075, 2024-01-20:
I am using realsense d455 camera and only using one camera i tried to move 180 degree after reaching back to starting point but still it give the overlap map. 

The second problem is that when i get the 2d map using rtabmap i want to provide another map in which it is define where my robot cannot go e.g if a entrance width is 10 cm it will show in map that its open area but if my robot size is bigger than 10 cm it will go there and hit the walls. I am new in this field so any further guidance would be appreciated. Thanks

matlabbe, 2024-01-21:
> 180 degree after reaching back to starting point but still it give the overlap map.
Is it even corrected? If a loop closure cannot be found, it won't correct the drift.