introlab/rtabmap_ros #1107: undefined reference to `vtable for vtkSubdivisionFilter'

lorenzo-eng, 2024-01-23:
Hello,

I am trying to build rtabmap `0.21.3` with Open3D support on `ubuntu 20.04 - ros noetic`.
Unfortunately, I get the following link error:

```
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/StereoEval/CMakeFiles/stereoEval.dir/build.make:202: bin/rtabmap-stereoEval] Error 1
make[1]: *** [CMakeFiles/Makefile2:1047: tools/StereoEval/CMakeFiles/stereoEval.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/KittiDataset/CMakeFiles/kitti_dataset.dir/build.make:202: bin/rtabmap-kitti_dataset] Error 1
make[1]: *** [CMakeFiles/Makefile2:1074: tools/KittiDataset/CMakeFiles/kitti_dataset.dir/all] Error 2
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/Camera/CMakeFiles/camera.dir/build.make:202: bin/rtabmap-camera] Error 1
make[1]: *** [CMakeFiles/Makefile2:1020: tools/Camera/CMakeFiles/camera.dir/all] Error 2
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/ConsoleApp/CMakeFiles/consoleApp.dir/build.make:202: bin/rtabmap-console] Error 1
make[1]: *** [CMakeFiles/Makefile2:939: tools/ConsoleApp/CMakeFiles/consoleApp.dir/all] Error 2
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/build.make:202: bin/rtabmap-extractObject] Error 1
make[1]: *** [CMakeFiles/Makefile2:993: tools/ExtractObject/CMakeFiles/extractObject.dir/all] Error 2
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/EurocDataset/CMakeFiles/euroc_dataset.dir/build.make:203: bin/rtabmap-euroc_dataset] Error 1
make[1]: *** [CMakeFiles/Makefile2:1128: tools/EurocDataset/CMakeFiles/euroc_dataset.dir/all] Error 2
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/build.make:202: bin/rtabmap-rgbd_dataset] Error 1
make[1]: *** [CMakeFiles/Makefile2:1101: tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/all] Error 2
/usr/bin/ld: ../../bin/librtabmap_core.so.0.21.3: undefined reference to `vtable for vtkSubdivisionFilter'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/Recovery/CMakeFiles/recovery.dir/build.make:202: bin/rtabmap-recovery] Error 1
make[1]: *** [CMakeFiles/Makefile2:1155: tools/Recovery/CMakeFiles/recovery.dir/all] Error 2
```

build steps:
```
cmake -DWITH_OPEN3D=ON  ..
make
```

cmake output:
```
-- The C compiler identification is GNU 9.4.0                                                                                                                                                              
-- The CXX compiler identification is GNU 9.4.0                                                                                                                                                            
-- Detecting C compiler ABI info                                                                                                                                                                           
-- Detecting C compiler ABI info - done                                                                                                                                                                    
-- Check for working C compiler: /usr/bin/cc - skipped                                                                                                                                                     
-- Detecting C compile features                                                                                                                                                                            
-- Detecting C compile features - done                                                                                                                                                                     
-- Detecting CXX compiler ABI info                                                                                                                                                                         
-- Detecting CXX compiler ABI info - done                                                                                                                                                                  
-- Check for working CXX compiler: /usr/bin/c++ - skipped                                                                                                                                                  
-- Detecting CXX compile features                                                                                                                                                                          
-- Detecting CXX compile features - done 
....
-- Found PCL_VISUALIZATION: /usr/lib/x86_64-linux-gnu/libpcl_visualization.so                                                                                                                              
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).  
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so                                                                                                                                     
-- Found OpenMP_C: -fopenmp (found version "4.5")                                                                                                                                                          
-- Found OpenMP_CXX: -fopenmp (found version "4.5")                                                                                                                                                        
-- Found OpenMP: TRUE (found version "4.5") found components: C CXX                                                                                                                                        
-- Found OpenMP: /usr/lib/gcc/x86_64-linux-gnu/9/libgomp.so;/usr/lib/x86_64-linux-gnu/libpthread.so                                                                                                        
-- Add PCL_OMP to definitions                                                                                                                                                                              
[... 124 lines cut ...]
$ ldd bin/librtabmap_core.so.0.21.3 | grep vtk
        libvtkCommonCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonCore-7.1.so.7.1p (0x00007fa902c2a000)
        libvtkFiltersModeling-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkFiltersModeling-7.1.so.7.1p (0x00007fa8fffce000)
        libvtkFiltersCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkFiltersCore-7.1.so.7.1p (0x00007fa8ffa94000)
        libvtkCommonExecutionModel-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonExecutionModel-7.1.so.7.1p (0x00007fa8ff9c9000)
        libvtkCommonDataModel-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonDataModel-7.1.so.7.1p (0x00007fa8ff5f1000)
        libvtkIOGeometry-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkIOGeometry-7.1.so.7.1p (0x00007fa8fd3f0000)
        libvtkIOLegacy-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkIOLegacy-7.1.so.7.1p (0x00007fa8fd326000)
        libvtkIOPLY-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkIOPLY-7.1.so.7.1p (0x00007fa8fd307000)
        libvtkIOCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkIOCore-7.1.so.7.1p (0x00007fa8fd1bb000)
        libvtkImagingCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkImagingCore-7.1.so.7.1p (0x00007fa8fcfb6000)
        libvtkIOImage-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkIOImage-7.1.so.7.1p (0x00007fa8fcdeb000)
        libvtksys-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtksys-7.1.so.7.1p (0x00007fa8fcd9d000)
[... cut ...]
        libvtkCommonSystem-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonSystem-7.1.so.7.1p (0x00007fa8f7a56000)
        libvtkDICOMParser-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkDICOMParser-7.1.so.7.1p (0x00007fa8f4c3e000)
        libvtkmetaio-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkmetaio-7.1.so.7.1p (0x00007fa8f4b9f000)
        libvtkCommonComputationalGeometry-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonComputationalGeometry-7.1.so.7.1p (0x00007fa8f3924000)
```

Thank you for any suggestion

matlabbe, 2024-01-27:
Can you cleanup the build directory and retry? I cannot reproduce the issue.

lorenzo-eng, 2024-02-05:
sorry there was a mess with my build container I haven't fully understood, yet.
Simply cleaning up the build directory was not enough, but starting a new container and installing again dependencies has worked.
Thank you for your support

closed 2024-02-05 by lorenzo-eng