introlab/rtabmap_ros #1108: 3D map can not be generated and visualized on RVIZ

Jenanaputra, 2024-01-25:
Hi

I am working with ROS2 Humble to generate 3D map using rtabmap_ros. It is run in the turtlebot3 simulation.  I used the depth camera plugin on the robot urdf, so it can publish depthcamera, camerainfo and image_rect_color.

The way I run the simulation :

1.  ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
2.  run the bridge.py program (this is created by myself to remap the '/intel_realsense_r200_depth/depth/image_raw', '/intel_realsense_r200_depth/camera_info', '/intel_realsense_r200_depth/image_raw' to be '/camera/depth_registered/image_raw', '/camera/rgb/camera_info','/camera/rgb/image_rect_color')   
3. ros2 launch rtabmap_launch rtabmap.launch.py visual_odometry:=false frame_id:=base_footprint subscribe_scan:=true depth:=true approx_sync:=true odom_topic:=/odom scan_topic:=/scan qos:=2 args:="-d --RGBD/NeighborLinkRefining true --Reg/Strategy 1" use_sim_time:=true rviz:=true

When all those steps are executed, the 3d map did not come up. 
[image]

But I got the warning notification on my command line terminal:
[image]

I also attach the rqt_graph of the active nodes and msgs:
[image]

Does anyone know what happen here and how to solve it ? I am on the dead end. Thanks!!

matlabbe, 2024-01-27:
To make the turtlebot3 simulation works, look at the instructions in the header of that launch file:
https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_demos/launch/turtlebot3_rgbd_sync.launch.py

Jenanaputra, 2024-01-28:
Hii @matlabbe .. Thanks!!! it works

closed 2024-01-28 by Jenanaputra

basheeraltawil, 2025-07-25:
Can you share all model.sdf script 
Because I added this part of modification and couldn't get th 3D details.
I got the results as in this issue .

matlabbe, 2025-08-03:
The latest version of that launch file is here: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_demos/launch/turtlebot3/turtlebot3_sim_rgbd_demo.launch.py, though it looks pretty similar to instructions above.