introlab/rtabmap_ros #1109: create_publisher() failed setting topic QoS

kosmonauta144, 2024-01-26:
Hi, 

I was using `rtabmap_ros` for a while now with ROS2 Humble on Ubuntu 22.04. It was great so far, but today morning I was experimenting with DDS and also made an upgrade of some ros packages. After this, I am not able to use `rtabmap` because it crashes short after launching, no matter if I launch it from default launch file or from my own files. Parameters are properly read and then I got an error shown below

```
[rtabmap-2] 
[rtabmap-2] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[rtabmap-2] This error state is being overwritten:
[rtabmap-2] 
[rtabmap-2]   'Unknown QoS durability policy, at ./src/qos.cpp:75'
[rtabmap-2] 
[rtabmap-2] with this new error message:
[rtabmap-2] 
[rtabmap-2]   'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'
[rtabmap-2] 
[rtabmap-2] rcutils_reset_error() should be called after error handling to avoid this.
[rtabmap-2] <<<
[rtabmap-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rtabmap-2]   what():  could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116
```

I was trying to change QoS for 0, 1 and 2 but it does not help. I've deleted exporting DDS from my `.bashrc` file because I was hoping that this is the problem (also when changing DDS the computing consumption increases dramatically but it's different subject). 

I am not sure if the `update` and `upgrade` change something and now it's not working, but `rtabmap` crashes. `icp_odometry` is still working as a package but I am not able to get a map. Does anybody can tell me how can I fix this?

matlabbe, 2024-01-28:
I cannot reproduce the problem with [latest docker image using humble binaries](https://github.com/introlab/rtabmap_ros/blob/ros2/docker/humble/Dockerfile):
```
docker run -it  introlab3it/rtabmap_ros:humble ros2 launch rtabmap_launch rtabmap.launch.py rtabmap_viz:=false

[INFO] [launch]: All log files can be found below /root/.ros/log/2024-01-28-00-30-40-380480-d4d551cca63d-111
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgbd_odometry-1]: process started with pid [112]
[INFO] [rtabmap-2]: process started with pid [114]
[rgbd_odometry-1] [INFO] [1706401840.709966349] [rtabmap.rgbd_odometry]: Odometry: frame_id               = base_link
[rgbd_odometry-1] [INFO] [1706401840.710449868] [rtabmap.rgbd_odometry]: Odometry: odom_frame_id          = odom
[rgbd_odometry-1] [INFO] [1706401840.710653992] [rtabmap.rgbd_odometry]: Odometry: publish_tf             = true
[rgbd_odometry-1] [INFO] [1706401840.710662198] [rtabmap.rgbd_odometry]: Odometry: wait_for_transform     = 0.200000
[rgbd_odometry-1] [INFO] [1706401840.710677322] [rtabmap.rgbd_odometry]: Odometry: log_to_rosout_level    = 4
[rgbd_odometry-1] [INFO] [1706401840.710701404] [rtabmap.rgbd_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[rgbd_odometry-1] [INFO] [1706401840.710711019] [rtabmap.rgbd_odometry]: Odometry: ground_truth_frame_id  = 
[rgbd_odometry-1] [INFO] [1706401840.710716637] [rtabmap.rgbd_odometry]: Odometry: ground_truth_base_frame_id = 
[rgbd_odometry-1] [INFO] [1706401840.710721823] [rtabmap.rgbd_odometry]: Odometry: config_path            = 
[rgbd_odometry-1] [INFO] [1706401840.710726890] [rtabmap.rgbd_odometry]: Odometry: publish_null_when_lost = true
[rgbd_odometry-1] [INFO] [1706401840.710731576] [rtabmap.rgbd_odometry]: Odometry: publish_compressed_sensor_data = false
[rgbd_odometry-1] [INFO] [1706401840.710736904] [rtabmap.rgbd_odometry]: Odometry: guess_frame_id         = 
[rgbd_odometry-1] [INFO] [1706401840.710741741] [rtabmap.rgbd_odometry]: Odometry: guess_min_translation  = 0.000000
[rgbd_odometry-1] [INFO] [1706401840.710747612] [rtabmap.rgbd_odometry]: Odometry: guess_min_rotation     = 0.000000
[rgbd_odometry-1] [INFO] [1706401840.710753096] [rtabmap.rgbd_odometry]: Odometry: guess_min_time         = 0.000000
[rgbd_odometry-1] [INFO] [1706401840.710758765] [rtabmap.rgbd_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[rgbd_odometry-1] [INFO] [1706401840.710764078] [rtabmap.rgbd_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[rgbd_odometry-1] [INFO] [1706401840.710769580] [rtabmap.rgbd_odometry]: Odometry: min_update_rate        = 0.000000 Hz
[rgbd_odometry-1] [INFO] [1706401840.710775143] [rtabmap.rgbd_odometry]: Odometry: wait_imu_to_init       = false
[rgbd_odometry-1] [INFO] [1706401840.710780177] [rtabmap.rgbd_odometry]: Odometry: sensor_data_compression_format = .jpg
[rgbd_odometry-1] [INFO] [1706401840.710785550] [rtabmap.rgbd_odometry]: Odometry: sensor_data_parallel_compression = true
[rgbd_odometry-1] [INFO] [1706401840.710793136] [rtabmap.rgbd_odometry]: Odometry: stereoParams_=0 visParams_=1 icpParams_=0
[rgbd_odometry-1] [INFO] [1706401840.715158630] [rtabmap.rgbd_odometry]: RGBDOdometry: approx_sync    = false
[rgbd_odometry-1] [INFO] [1706401840.715173751] [rtabmap.rgbd_odometry]: RGBDOdometry: queue_size     = 10
[rgbd_odometry-1] [INFO] [1706401840.715180019] [rtabmap.rgbd_odometry]: RGBDOdometry: qos            = 1
[rgbd_odometry-1] [INFO] [1706401840.715184878] [rtabmap.rgbd_odometry]: RGBDOdometry: qos_camera_info = 1
[rgbd_odometry-1] [INFO] [1706401840.715189528] [rtabmap.rgbd_odometry]: RGBDOdometry: subscribe_rgbd = false
[rgbd_odometry-1] [INFO] [1706401840.715194446] [rtabmap.rgbd_odometry]: RGBDOdometry: rgbd_cameras   = 1
[rgbd_odometry-1] [INFO] [1706401840.715199187] [rtabmap.rgbd_odometry]: RGBDOdometry: keep_color     = false
[rgbd_odometry-1] [INFO] [1706401840.717027647] [rtabmap.rgbd_odometry]: 
[rgbd_odometry-1] rgbd_odometry subscribed to (exact sync):
[rgbd_odometry-1]    /camera/rgb/image_rect_color,
[rgbd_odometry-1]    /camera/depth_registered/image_raw,
[rgbd_odometry-1]    /camera/rgb/camera_info
[rtabmap-2] [INFO] [1706401840.751255127] [rtabmap.rtabmap]: rtabmap(maps): latch                      = true
[rtabmap-2] [INFO] [1706401840.751376195] [rtabmap.rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-2] [INFO] [1706401840.751390088] [rtabmap.rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-2] [INFO] [1706401840.751394493] [rtabmap.rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-2] [INFO] [1706401840.751397652] [rtabmap.rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-2] [INFO] [1706401840.751401993] [rtabmap.rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-2] [INFO] [1706401840.751405082] [rtabmap.rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-2] [INFO] [1706401840.751407964] [rtabmap.rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-2] [INFO] [1706401840.751410900] [rtabmap.rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-2] [INFO] [1706401840.751428314] [rtabmap.rtabmap]: rtabmap(maps): octomap_tree_depth         = 16
[rtabmap-2] [INFO] [1706401840.760135555] [rtabmap.rtabmap]: rtabmap: frame_id      = base_link
[... cut ...] clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap-2] rtabmap subscribed to (exact sync):
[rtabmap-2]    /rtabmap/odom \
[rtabmap-2]    /camera/rgb/image_rect_color \
[rtabmap-2]    /camera/depth_registered/image_raw \
[rtabmap-2]    /camera/rgb/camera_info \
[rtabmap-2]    /rtabmap/odom_info
```
Can you make sure that all ROS packages have been upgraded?

kosmonauta144, 2024-01-29:
Thanks for quick reply. I checked and it appears that two packages has been held back `ros-humble-rtabmap` and `ros-humble-rtabmap-utils`. After forcing update with `sudo apt-get --with-new-pkgs upgrade ros-humble-rtabmap ros-humble-rtabmap-utils` everything is fine again. Thank you very much for trying to reproduce error and ensuring me that I messed up on my machine.

closed 2024-01-30 by matlabbe