introlab/rtabmap_ros #1110: Reading mapGraph poses from rtabmap.db directly

360wcui, 2024-01-26:
Would it be possible to read mapGraph poses from rtabmap.db?    I tried to read poses from Node table in rtabmap.db and realized that they are odometry poses not mapGraph poses.  Any insight would be greatly appreciated. 

[image]
[image]

matlabbe, 2024-01-28:
Look under `Admin` table, there is `opt_poses` field.
https://github.com/introlab/rtabmap/blob/c27507bc772cfda34f6624d386ab2cd62d49e90d/corelib/src/resources/DatabaseSchema.sql.in#L123-L124

360wcui, 2024-01-28:
Thank you, but the field is null sometimes.   Would it be possible to enable it in real-time like /rtabmap/mapGraph which is published at 1 hz.   
[image]

matlabbe, 2024-01-30:
Currently it is saved only when the database is closed. In ROS, you would have already the same info with `/rtabmap/mapGraph`.

closed 2024-01-31 by 360wcui