introlab/rtabmap_ros #1111: rtabmap_viz vtkOpenGLPolyDataMapper warnings filling shell output

unicornuniform, 2024-01-27:
When I run rtabmap_ros, I get a nonstop flood of:
`[rtabmap_viz-4] 2024-01-27 12:43:14.845 (  13.573s) [        A2CDB3C0]vtkOpenGLPolyDataMapper:336   WARN| vtkOpenGLPolyDataMapper::GetGeometryShaderCode was deprecated for VTK 9.0 and will be removed in a future version.  Use vtkOpenGLShaderProperty::GetGeometryShaderCode instead.`

matlabbe, 2024-01-28:
It seems this warning is coming from VTK library itself: https://discourse.vtk.org/t/vtk-9-0-rc1/2916/3

Other related comment in PointCloud library:
https://github.com/PointCloudLibrary/pcl/issues/4830#issuecomment-873326759

We don't call these functions explicitly in rtabmap code. The fix would be to add `VTK_LEGACY_REMOVE` in cmake.

unicornuniform, 2024-02-01:
Will this make it's way into the binaries eventually?

matlabbe, 2024-02-04:
The commit above should fix the VTK warnings issue in next release.

closed 2024-02-10 by matlabbe