introlab/rtabmap_ros #1113: Build Error: The plugin for class 'rtabmap_ros/MapCloud' failed to load.  Error: According to the loaded plugin descriptions the class rtabmap_ros/MapCloud with base class type rviz::Display does not exist.

julianraheema, 2024-01-30:
I used to have a old rtabmap_ros package, and it worked fine with ROS Noetic. I was doing SLAM, and I was getting 2D and 3D depth image maps. Everything was working fine. But I did apt update and upgrade, and the rtabmap stopped working. I thought it is time to update to a new rtabmap_ros since it is broken, so I followed the instructions at https://github.com/introlab/rtabmap_ros#build-from-source "build from source" to build inside my workspace, and switched to noetic-dev branch and succeeded build. Now when I launch the rtabmap, I get these errors, and I can't see the topics for MapGraph, MapCloud, and Info.

OS:20.04
ROS: Noetic

[ERROR] [1706580783.874992568]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapCloud' failed to load.  Error: According to the loaded plugin descriptions the class rtabmap_ros/MapCloud with base class type rviz::Display does not exist. Declared types are  grid_map_rviz_plugin/GridMap octomap_rviz_plugin/ColorOccupancyGrid octomap_rviz_plugin/OccupancyGrid octomap_rviz_plugin/OccupancyGridStamped octomap_rviz_plugin/OccupancyMap octomap_rviz_plugin/OccupancyMapStamped rtabmap_rviz_plugins/Info rtabmap_rviz_plugins/MapCloud rtabmap_rviz_plugins/MapGraph rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped rviz_imu_plugin/Imu rviz_map_plugin/ClusterLabel rviz_map_plugin/Map3D rviz_map_plugin/Mesh rviz_plugin_tutorials/Imu
[ERROR] [1706580783.875324607]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapGraph' failed to load.  Error: According to the loaded plugin descriptions the class rtabmap_ros/MapGraph with base class type rviz::Display does not exist. Declared types are  grid_map_rviz_plugin/GridMap octomap_rviz_plugin/ColorOccupancyGrid octomap_rviz_plugin/OccupancyGrid octomap_rviz_plugin/OccupancyGridStamped octomap_rviz_plugin/OccupancyMap octomap_rviz_plugin/OccupancyMapStamped rtabmap_rviz_plugins/Info rtabmap_rviz_plugins/MapCloud rtabmap_rviz_plugins/MapGraph rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped rviz_imu_plugin/Imu rviz_map_plugin/ClusterLabel rviz_map_plugin/Map3D rviz_map_plugin/Mesh rviz_plugin_tutorials/Imu
[ERROR] [1706580783.875424699]: PluginlibFactory: The plugin for class 'rtabmap_ros/Info' failed to load.  Error: According to the loaded plugin descriptions the class rtabmap_ros/Info with base class type rviz::Display does not exist. Declared types are  grid_map_rviz_plugin/GridMap octomap_rviz_plugin/ColorOccupancyGrid octomap_rviz_plugin/OccupancyGrid octomap_rviz_plugin/OccupancyGridStamped octomap_rviz_plugin/OccupancyMap octomap_rviz_plugin/OccupancyMapStamped rtabmap_rviz_plugins/Info rtabmap_rviz_plugins/MapCloud rtabmap_rviz_plugins/MapGraph rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped rviz_imu_plugin/Imu rviz_map_plugin/ClusterLabel rviz_map_plugin/Map3D rviz_map_plugin/Mesh rviz_plugin_tutorials/Imu

matlabbe, 2024-01-30:
You would have to update the rviz config by re-adding the rtabmap's related displays, or edit in text the rviz config and change all rtabmap_ros by rtabmap_rviz_plugins.

matlabbe, 2024-01-30:
See also https://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2

The latest binaries and source code all use the new interface.

julianraheema, 2024-01-30:
Thank you so much. It seems working now. I will test it some more and then close this ticket if everything looks fine. Much appreciated.

julianraheema, 2024-01-30:
Tested and worked as expected. The proposed solved the problem. Thank you so much

closed 2024-01-30 by julianraheema