introlab/rtabmap_ros #1114: Segmentation fault in ROS Humble with GTSAM 4.2.0

brayanpa, 2024-02-01:
Hi Mathieu,

During the mapping process, rtabmap suddenly dies without any error message. I am mapping using a lidar and rgbd camera with the parameters:
```
'Reg/Strategy': '1'
'RGBD/OptimizeMaxError': '0'
```

It happens randomly over time. I'm not sure if it's related, but it hasn't happened to me so far with `RGBD/OptimizeMaxError` set to `3.0`.

I built and am launching with debug flags using GDB and I get this error from the debug console:

```
Thread 1 "rtabmap" received signal SIGSEGV, Segmentation fault.
0x0000ffffebe7876c in void std::vector<boost::shared_ptr<gtsam::GaussianBayesTreeClique>, tbb::tbb_allocator<boost::shared_ptr<gtsam::GaussianBayesTreeClique> > >::_M_realloc_insert<boost::shared_ptr<gtsam::GaussianBayesTreeClique> const&>(__gnu_cxx::__normal_iterator<boost::shared_ptr<gtsam::GaussianBayesTreeClique>*, std::vector<boost::shared_ptr<gtsam::GaussianBayesTreeClique>, tbb::tbb_allocator<boost::shared_ptr<gtsam::GaussianBayesTreeClique> > > >, boost::shared_ptr<gtsam::GaussianBayesTreeClique> const&) () from target:/usr/local/lib/libgtsam.so.4
```

The commits I am currently using are:
```
rtabmap commit e619c577595ad221890d550ea3dbe6cbdebe9502
rtabmap_ros commit 2f365ab7e10025c4d13b7855dfb4b7c8bb01f2d3
gtsam 4.2.0  commit 4f66a491ffc83cf092d0d818b11dc35135521612
```
Let me know what other information might be useful or if you have any ideas how we could replicate it, thanks.

unicornuniform, 2024-02-01:
One thing you can do is try compiling GTSAM with march=native / GTSAM_BUILD_WITH_MARCH_NATIVE disabled/turned off ?

matlabbe, 2024-02-03:
Having the debug log leading to this error would be useful. Add `--udebug` argument to rtabmap node.

brayanpa, 2024-05-27:
Compiling GTSAM with march=native solved it. Thank you so much.

closed 2024-05-27 by brayanpa