introlab/rtabmap_ros #1115: How do I publish an amcl_pose topic?

yemerge, 2024-02-02:
I am trying to estimate the location of go1 by modifying deco_unitree_quadruped_robot.launch. Added map server, pointCloud to LaserScan and amcl.
An error occurs when running rostopic echo /rtabmap/amcl_pose.
But /clock and use_sim_time are normal.
rostopic echo /rtabmap/scan data is displayed.
Publisher and subscriber associations have been verified with topic information.
How do I do 2D pose estimation in rviz?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

matlabbe, 2024-02-03:
You should use either `map_server`+`amcl` or `rtabmap` node, not both approaches at the same time. If you want to use AMCL, remove rtabmap node. you may also remove `rgbd_odometry` as your odometry seems coming from gazebo already (based on your tf tree).

So, if you remove all rtabmap parts, you may follow this AMCL tutorial directly to set the 2D pose estimate. In your case, I think it didn't work becasue the amcl node was launched in `/rtabmap` namespace, so its input `initialpose` topic is `rtabmap/initialpose` instead. Either move amcl outside rtabmap namespace, or change the topic of 2d pose estimate button in rviz tool's properties.

yemerge, 2024-02-05:
I use a 2D map converted from the rtabmap database.
Can I do 2d-pose-estimate on rviz without using amcl node?
I don't understand why I have to choose either map_server+amcl or rtabmap node.

I tried running it after commenting out <group ns="rtabmap"> and </group> in demo_unitree_quadruped_robot_navigation.launch, but it does not work properly.

yemerge, 2024-02-06:
How can I do rtabmap localization?(without amcl)

matlabbe, 2024-02-11:
You can relaunch rtabmap in [localization mode](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping#Localization_mode) by adding `Mem/IncrementalMemory` parameter set to `false`. Make sure you removed `--delete_db-on_start` argument if you used it to not delete the database. You don't need to start another map_server or amcl if you do so.

In your original workflow, if you exported the 2d occupancy grid and used map_server to publish it in localization session, you would need to add amcl node to localize in it (this would only work if the robot has a 2D lidar). The 2D pose estimate topic from RVIZ is sending on `/initialpose` topic by default, which is the one amcl is using.

yemerge, 2024-02-14:
<param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>
<remap from="grid_map" to="/map"/>
<include file="$(find unitree_move_base)/launch/pointCloud2LaserScan.launch"/>
<include file="$(find unitree_move_base)/launch/move_base.launch"/>
By adding , the 2D Nav goal was successful in rviz.
I tried changing the initial position of the robot using rostopic pub /rtabmap/initialpose geometry_msgs/PoseWithCovarianceStamped, but the robot changes to that position and then returns to the previous position.
Does localization mode automatically estimate the location?
Is there a way to adjust the robot's position if the robot's position on the rviz and the robot's position on the gazebo do not match?

matlabbe, 2024-02-17:
Localization mode still continue to do global localization even if you set an initial pose. For example, we can roughly set the location of the robot with 2D Pose estimate button, but the localization approach may refine that location and correct it if needed. If the robot is jumping back to a wrong location after setting the initial pose, there is something wrong the the localization. Is it the whole launch file that you are using? If AMCL+map_server is used, I cannot really say why the robot would jump back.