introlab/rtabmap_ros #1119: genScan using depth images

takahiko-hasegawa-hdjp, 2024-02-16:
Im trying to use RGBD images. Depth as laser scan data and hoping that this would improve localization.

First question: how does laser scan generated from depth images help with map-matching and also mapping. 
Second question: what is the difference between using dpeth images directly as input and converting into 2d laser scan.
Third: I tried it in my system and somehow the black dots(occupied in occupancy grid) looks like its in the front of the camera but looks very different when I just use depth images as it is. why?

using gen scan: 

[image]

not converting to scan and use as depth only:

[image]

its just a simple road so not converting to scan looks more correct.... why so?

matlabbe, 2024-02-17:
Is the camera perfectly looking forward (parallel to ground)? Based on the [doc](http://wiki.ros.org/rtabmap_slam), you should not use `gen_scan` if the camera is tilted.

For your first question, it won't really help for mapping unless the depth is relatively accurate (TOF cameras), but can be convenient if you have more than one cameras to generate a fake 360 laser scan, which can be useful for proximity detection.

Second question: if you convert to laser scan, you can enable ICP refinement (`Reg/Strategy=1`). In some cases when there is a lot of geometry, it can help to get more accurate loop closure transform. However, if the depth is not accurate (e.g., stereo camera), I don't recommend to convert to scan.

Third question: is the camera tilted? If so, it may be the floor detected as obstacle with `gen_scan`. If the camera is tilted, you would need to use a node like [pointcloud_to_laserscan](http://wiki.ros.org/pointcloud_to_laserscan) to generate the scan externally.

takahiko-hasegawa-hdjp, 2024-02-18:
Ok I see, that clarifies things now. My camera is a stereo camera and tilted down, so I guess you are right about the floor bing detected as object. Thanks!