introlab/rtabmap_ros #1125: build problem

Boris-Cui, 2024-02-21:
hallo, i was using ros2 humble in ubuntu22 to build rtabmap but failed. i follow the instructions to build.
cd ~/ros2_ws
git clone https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
rosdep update && rosdep install --from-paths src --ignore-src -r -y
export MAKEFLAGS="-j6" # Can be ignored if you have a lot of RAM (>16GB)
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
the below is the process:

Starting >>> rtabmap 
Starting >>> rtabmap_msgs
Starting >>> rtabmap_python
Finished <<< rtabmap_python [0.89s]                                                                          
[Processing: rtabmap, rtabmap_msgs]                                                            
--- stderr: rtabmap_msgs                                                                         
CMake Warning:
  Manually-specified variables were not used by the project:

    CATKIN_INSTALL_INTO_PREFIX_ROOT

---
Finished <<< rtabmap_msgs [51.5s]
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
[Processing: rtabmap]                                        
--- stderr: rtabmap                                          
In file included from /home/shiqing/ros2_ws/src/rtabmap/guilib/src/MainWindow.cpp:30:
/home/shiqing/ros2_ws/build/rtabmap/guilib/src/rtabmap_gui_autogen/include/ui_mainWindow.h: In member function ‘void Ui_mainWindow::setupUi(QMainWindow*)’:
/home/shiqing/ros2_ws/build/rtabmap/guilib/src/rtabmap_gui_autogen/include/ui_mainWindow.h:641:56: warning: ‘QDockWidget::AllDockWidgetFeatures’ is deprecated [-Wdeprecated-declarations]
  641 |         dockWidget_posterior->setFeatures(QDockWidget::AllDockWidgetFeatures);
      |                                                        ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QDockWidget:1,
                 from /home/shiqing/ros2_ws/build/rtabmap/guilib/src/rtabmap_gui_autogen/include/ui_mainWindow.h:16,
                 from /home/shiqing/ros2_ws/src/rtabmap/guilib/src/MainWindow.cpp:30:
/usr/include/x86_64-linux-gnu/qt5/QtWidgets/qdockwidget.h:82:9: note: declared here
   82 |         AllDockWidgetFeatures Q_DECL_ENUMERATOR_DEPRECATED =
      |         ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/shiqing/ros2_ws/src/rtabmap/guilib/src/MainWindow.cpp:30:
/home/shiqing/ros2_ws/build/rtabmap/guilib/src/rtabmap_gui_autogen/include/ui_mainWindow.h:641:56: warning: ‘QDockWidget::AllDockWidgetFeatures’ is deprecated [-Wdeprecated-declarations]
  641 |         dockWidget_posterior->setFeatures(QDockWidget::AllDockWidgetFeatures);
      |                                                        ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QDockWidget:1,
                 from /home/shiqing/ros2_ws/build/rtabmap/guilib/src/rtabmap_gui_autogen/include/ui_mainWindow.h:16,
                 from /home/shiqing/ros2_ws/src/rtabmap/guilib/src/MainWindow.cpp:30:
/usr/include/x86_64-linux-gnu/qt5/QtWidgets/qdockwidget.h:82:9: note: declared here
   82 |         AllDockWidgetFeatures Q_DECL_ENUMERATOR_DEPRECATED =
      |         ^~~~~~~~~~~~~~~~~~~~~
/home/shiqing/ros2_ws/src/rtabmap/guilib/src/MainWindow.cpp: In member function ‘void rtabmap::MainWindow::drawKeypoints(const std::multimap<int, cv::KeyPoint>&, const std::multimap<int, cv::KeyPoint>&)’:
/home/shiqing/ros2_ws/src/rtabmap/guilib/src/MainWindow.cpp:5051:39: warning: ‘static QSet<T> QSet<T>::fromList(const QList<T>&) [with T = int]’ is deprecated: Use QSet<T>(list.begin(), list.end()) instead. [-Wdeprecated-declarations]
 5051 |                 _lastIds = QSet<int>::fromList(QList<int>::fromStdList(kpts));
      |                                       ^~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/QSet:1,
                 from /home/shiqing/ros2_ws/src/rtabmap/guilib/src/../include/rtabmap/gui/MainWindow.h:35,
                 from /home/shiqing/ros2_ws/src/rtabmap/guilib/src/MainWindow.cpp:28:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qset.h:416:9: note: declared here
  416 | QSet<T> QSet<T>::fromList(const QList<T> &list)
      |         ^~~~~~~
[... 173 lines cut ...]
  Could not find a package configuration file provided by "apriltag_msgs"
  with any of the following names:

    apriltag_msgsConfig.cmake
    apriltag_msgs-config.cmake

  Add the installation prefix of "apriltag_msgs" to CMAKE_PREFIX_PATH or set
  "apriltag_msgs_DIR" to a directory containing one of the above files.  If
  "apriltag_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Finished <<< rtabmap_slam [1min 42s]
Finished <<< rtabmap_odom [1min 44s]                               
Starting >>> rtabmap_demos
Starting >>> rtabmap_examples                       
Starting >>> rtabmap_launch
Finished <<< rtabmap_demos [0.87s]                                                                                                        
Finished <<< rtabmap_examples [0.92s]                                                                      
Finished <<< rtabmap_launch [0.93s]
Starting >>> rtabmap_ros
Finished <<< rtabmap_ros [0.76s]                          

Summary: 14 packages finished [14min 9s]
  6 packages had stderr output: rtabmap rtabmap_conversions rtabmap_msgs rtabmap_rviz_plugins rtabmap_slam rtabmap_util

matlabbe, 2024-02-24:
The build completed, those are warnings.

Boris-Cui, 2024-02-26:
thanks for your kindly reply!

closed 2024-02-26 by Boris-Cui