introlab/rtabmap_ros #1127: rtabmap rgbd image is not attaching

Murali25111993, 2024-02-22:
Hello,

Ubuntu 20.04 
ROS Noetic

  I have a ToF camera which publishes colour/image_rect_raw at 2.2 Hz and depth/ image_raw at 6.8 Hz . Rtabmap node publishes rgbd_image topic at 3.6 Hz and rtabmap/odom at 4.5 Hz .
  
  These camera topics are synchronized.

  When i navigate my rover , as you can infer from below image, the frames are not stitching properly. I am moving my rover at  Linear_velocity.x = 0.1 m/s and angular_velocity.z = 0.5 rad/sec. I have made changes in ROS wrapper for camera which is Publishing in Z -axis to X-axis. But, the rtabmap/rgbd_image topic is not stitching.
  
[image]

Kindly, guide me where the problem lies. I will provide additional info ,if required.

matlabbe, 2024-02-24:
What is base frame `frame_id` used for odometry and which `frame_id` is set in the header of the color image? Between base frame and the frame set in image topic (you can use `rosrun tf tf_echo base_link color_frame` to check), there should be an optical rotation of around `-pi/2 0 pi/2`

Murali25111993, 2024-02-26:
Thanks for the response , @matlabbe .

tf link is 

map --> odom --> base_link -->camera_link.

I have resolved the rotation of frame from z-axis to x axis. Modified the image publisher in  ros wrapper to this:
               
               else {
					*iter_x = (u - cx) * Z / fx;
					*iter_y = ((v - cy) * Z / fy);
					*iter_z = Z;
				}
		//initially it was this :		
		else {
					*iter_x = (u - cx) * Z / fx;
					*iter_y = -((v - cy) * Z / fy);
					*iter_z = -Z;
					++iter_x; ++iter_y; ++iter_z;
				}

**Note:** I am not using IMU, wheel odometry for odometry. Just simply relying on Visual Odometry
Now, The problem i observe is image binding not happening.
my launch file.
`
  <launch>

    <arg name="localization" default="false" />
    <arg name="rviz" default="false" />
    <arg name="delete-on-start" default="true" />
    
    
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">        
        <arg name="subscribe_rgbd" value="true" />
        <arg name="rgbd_topic" value="/rgbd_image" />
        <arg name="frame_id" value="base_link" />
        <arg name="approx_sync" value="false" />
        <arg name="queue_size" value="10"/>
        <arg name="localization" value="$(arg localization)" />
        <arg name="compressed" value="true" />
        <arg unless="$(arg rviz)" name="rtabmapviz" value="true" />
        <arg if="$(arg rviz)" name="rviz" value="$(arg rviz)" />
        <arg if="$(arg delete-on-start)" name="args" value="true" />
        <param name="Reg/Force3DoF"    value="true" />
        <param name="Grid/MappingStrategy" value="true"/>
        <param name="Optimizer/Slam2D" value="true" />
        <param name="cloud_noise_filtering_radius" value="0.05"/>
		<param name="cloud_noise_filtering_min_neighbors" value="2"/>
    </include>

    
    
    

</launch>
`

matlabbe, 2024-02-27:
> Now, The problem i observe is image binding not happening.

What do you mean by image binding not happening? The node is not correctly subscribed? topics are not published?

Murali25111993, 2024-02-27:
Sorry for my English,

The topics are publishing well. Take a look at the image below; there is a gap between each frame. When I move the rover forward, the rgbd_image does not stitch properly (image IDs 0 and 1 ... etc should have a continuous flow for good visual odometry ). There is a small gap between each frame, and this issue worsens when I attempt to rotate the rover.

[image]

matlabbe, 2024-03-02:
There is maybe something wrong in TF tree. Can you try setting rtabmap with `frame_id` set to same `frame_id` in the input image topic? The resulting 3D map may appear upside down because it will be in optical frame, but at least the the point cloud should be correctly assembled. If they still appear like a plane, then the depth image format is strange. A rosbag could be nice to look at.

Murali25111993, 2024-03-04:
Sure, Thanks for the input. I am on it. Will update the progress here.

Murali25111993, 2024-03-19:
Hello,

I had my camera stable, but within few seconds there is lot of shakiness in the 3D map. I have attached the video for your reference. 

https://github.com/introlab/rtabmap_ros/assets/134040857/407494ad-3f77-465d-8145-2ed9f4486a6a

Can you please tell me the params which has to be tuned. 

Note: rosbag file is more than 50 Mb, so didn't attach here. 

Thanks

matlabbe, 2024-03-24:
I would like to see the bag, can you post a public link to afile in a dropbox or google drive.

Murali25111993, 2024-03-25:
Ok,

I am adding the link for [dropbox](https://www.dropbox.com/scl/fi/jsdnefhhe5f6xj85eca1g/vslam.orig.bag?rlkey=5zn4c3xik5n8rk8an1k78x82a&dl=0).

Tf tree is below

[image]

matlabbe, 2024-03-31:
The depth image looks pretty strange to me. With DepthCloud plugin in rviz I see what is the issue. Invalid values should be zero, not ~4.3 meters. If you fix that, then for this scene you would keep only the small patch on the ground, which is not really usable for estimating visual odometry. Consider using a stereo camera instead.

[image]

Murali25111993, 2024-04-03:
Ok, Thanks much for this insight. I am closing this issue.

closed 2024-04-03 by Murali25111993