introlab/rtabmap_ros #1132: RTAB-MAP+ORBSLAM2  input RGBD data, detect  stereo camera, then the rtab-map just finish initialization

zhuoyan28, 2024-03-08:
roslaunch rtabmap_ros euroc_datasets.launch args:="Odom/Strategy 5 OdomORBSLAM/VocPath /home/shen/rtab_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt" MH_seq:=true raw_images_for_odom:=true
... logging to /home/shen/.ros/log/4a97e338-dcf3-11ee-81bc-000c29b76e8c/roslaunch-shen-97501.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://shen:37605/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmapviz/
 * /rtabmap/stereo_odometry/

PARAMETERS
 * /imu_filter/publish_tf: False
 * /imu_filter/use_mag: False
 * /imu_filter/world_frame: enu
 * /point_to_tf/fixed_frame_id: world
 * /point_to_tf/frame_id: leica
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: base_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: False
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: True
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.1
 * /rtabmap/rtabmapviz/approx_sync: False
 * /rtabmap/rtabmapviz/frame_id: base_link
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: False
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgb: False
 * /rtabmap/rtabmapviz/subscribe_rgbd: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
[... 180 lines cut ...]
[ INFO] [1709864803.108182238]: rtabmapviz: Parameters successfully read.
[ INFO] [1709864803.508980210]: /rtabmap/rtabmapviz: subscribe_depth = false
[ INFO] [1709864803.509129260]: /rtabmap/rtabmapviz: subscribe_rgb = false
[ INFO] [1709864803.509203377]: /rtabmap/rtabmapviz: subscribe_stereo = true
[ INFO] [1709864803.509283496]: /rtabmap/rtabmapviz: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1709864803.509312067]: /rtabmap/rtabmapviz: subscribe_odom_info = true
[ INFO] [1709864803.509372345]: /rtabmap/rtabmapviz: subscribe_user_data = false
[ INFO] [1709864803.509401987]: /rtabmap/rtabmapviz: subscribe_scan = false
[ INFO] [1709864803.509423238]: /rtabmap/rtabmapviz: subscribe_scan_cloud = false
[ INFO] [1709864803.509443391]: /rtabmap/rtabmapviz: subscribe_scan_descriptor = false
[ INFO] [1709864803.509464960]: /rtabmap/rtabmapviz: queue_size    = 10
[ INFO] [1709864803.509510330]: /rtabmap/rtabmapviz: approx_sync   = false
[ INFO] [1709864803.509631794]: Setup stereo callback
[ INFO] [1709864803.601678297]: 
/rtabmap/rtabmapviz subscribed to (exact sync):
   /rtabmap/odom \
   /cam0/image_raw \
   /cam1/image_raw \
   /stereo_camera/left/camera_info \
   /stereo_camera/right/camera_info \
   /rtabmap/odom_info
[ INFO] [1709864803.602007859]: rtabmapviz started.
[ WARN] [1709864886.273990127, 1305031098.382347911]: Detected jump back in time of 3.78216s. Clearing TF buffer.
[ WARN] [1709864886.274197827, 1305031098.382347911]: Detected jump back in time of 3.78216s. Clearing TF buffer.
[ WARN] [1709864886.274191416, 1305031098.382347911]: Detected jump back in time of 3.78216s. Clearing TF buffer.

matlabbe, 2024-03-09:
Did you play the bag with `--clock`? or what is the command used to play the euroc bag?

zhuoyan28, 2024-03-10:
the command is : rosbag play --clock rgbd_dataset_freiburg1_xyz.bag

zhuoyan28, 2024-03-10:
I find the reason:
euroc_datasets is a stereo data set, but the test data set given by experience is rgbd, resulting in no image in the output.
the commands are: 
roslaunch rtabmap_ros euroc_datasets.launch args:="Odom/Strategy 5 OdomORBSLAM/VocPath /home/rosadmin/rtab_ws/src/ORB_SLAM3/Vocabulary/ORBvoc.txt" MH_seq:=true raw_images_for_odom:=true
rosbag play --clock V1_01_easy.bag

the instruction is 
【ROS】RTABMAP+ORBSLAM3环境搭建步骤
https://zhuanlan.zhihu.com/p/528557998

closed 2024-03-10 by zhuoyan28