introlab/rtabmap_ros #1133: RTABMAP republish doesn't accept camera input topic

lempiy, 2024-03-09:
Hi. 
I'm trying to make RTabmap + ORBSLAM3 working together. I managed to compile everything on ros noetic, including ORBSLAM3 lib, rtabmap with `-DWITH_ORB_SLAM=ON` and rtabmap_ros.
```
rtabmap --version
RTAB-Map:               0.21.4
PCL:                    1.10.0
With VTK:                7.1.1
OpenCV:                  4.2.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                false
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:           false
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:          true
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With Open3D:             false
With OctoMap:             true
With GridMap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM3:           true  <--- it's linked
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```
I'm starting my ros launch config:
`roslaunch realsense2_camera rs_rtabmap_d455.launch`
```
<launch>
    <!-- 
          NOTICE: Installation of rtabmap is required for using this launch file:
                  For installation type:
                        apt-get install ros-kinetic-rtabmap-ros
    -->
    <arg name="device_type_camera1"       default="d4.5"/>	<!-- Note: using regular expression. match D435, D435i, D415... -->
    <arg name="serial_no_camera1"         default=""/>
    <arg name="camera1"                   default="d455"/>
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="true"/>
    
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
              <arg name="enable_infra"        value="true"/>
              <arg name="enable_gyro"         value="true"/>
              <arg name="enable_accel"        value="true"/>
              <arg name="enable_pointcloud"   value="true"/>
              <arg name="align_depth"         value="true"/>
              <arg name="unite_imu_method"    value="copy"/>
    </include>

    <!-- Start IMU filter -->
    <node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter" output="screen">
        <param name="use_mag" value="false" />
        <param name="publish_tf" value="false" />
        <param name="world_frame" value="enu" />
        <remap from="/imu/data_raw" to="/camera/imu" />
        <remap from="/imu/data" to="/rtabmap/imu" />
    </node>
    
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="rgbd_topic"         value="/camera/aligned_depth_to_color/image_raw"/>
            <arg name="subscribe_rgbd"     value="true"/>
            <arg name="frame_id"           value="camera_link"/>
            <arg name="rgb_topic"          value="/camera/color/image_raw"/>
            <arg name="camera_info_topic"  value="/camera/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="wait_imu_to_init"   value="true" />
            <arg name="imu_topic"          value="/rtabmap/imu"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
            <arg name="cfg"                value="$(find realsense2_camera)/launch/rtabmap.ini" />
   </include>
</launch>
```
Getting error
```
Client [/rtabmap/republish_rgbd_image] wants topic /camera/aligned_depth_to_color/image_raw to have datatype/md5sum [rtabmap_msgs/RGBDImage/affef6cc8804ffba98ce6ed6f1ca8942], but our version has [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743]. Dropping connection.
```

Why `/rtabmap/republish_rgbd` expects `rtabmap_msgs/RGBDImage` ?
Would be thankful for help.

matlabbe, 2024-03-09:
I think you meant:
```xml
<arg name="depth_topic"       value="/camera/aligned_depth_to_color/image_raw"/>
<arg name="subscribe_rgbd"    value="false"/>
```
or
```xml
<arg name="depth_topic"       value="/camera/aligned_depth_to_color/image_raw"/>
<arg name="rgbd_sync"         value="true"/>
```

I think with D400 series, you could also set
```xml
<arg name="approx_sync"         value="false"/> <!-- if rgbd_sync is not set -->
<arg name="approx_rgbd_sync"    value="false"/> <!-- if rgbd_sync is set -->
```

lempiy, 2024-03-10:
Thank you for your rapid answer.
Your fixing topic from rgbd -> depth resolved problem with datatypes.

But know I've got rgbd_odometry crashing after ORB Voc loaded
```
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
[rtabmap/rgbd_odometry-6] process has died [pid 16803, exit code -11, cmd /home/anton/Projects/rtabmap_ros/devel/lib/rtabmap_odom/rgbd_odometry --delete_db_on_start rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image odom:=odom imu:=/rtabmap/imu __name:=rgbd_odometry __log:=/home/anton/.ros/log/a48aaf0e-deee-11ee-9aed-4338ecc5a3fa/rtabmap-rgbd_odometry-6.log].
log file: /home/anton/.ros/log/a48aaf0e-deee-11ee-9aed-4338ecc5a3fa/rtabmap-rgbd_odometry-6*.log
```

This happens in both cases (`<arg name="rgbd_sync" value="true"/>` and `<arg name="rgbd_sync" value="false"/>`)

Maybe you have some suggestion how it's possible to get more debug information about a reason of the crash?
`/home/anton/.ros/log/a48aaf0e-deee-11ee-9aed-4338ecc5a3fa/rtabmap-rgbd_odometry-6.log`, appearently, doesn't exist after error.

Thanks.

lempiy, 2024-03-10:
Full log:
```
... logging to /home/anton/.ros/log/b31ff276-def4-11ee-9aed-4338ecc5a3fa/roslaunch-anton-45029.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/opt/ros/noetic/share/realsense2_camera/launch/rs_rtabmap_d455.launch
[1mstarted roslaunch server http://anton:44159/[0m

SUMMARY
========

CLEAR PARAMETERS
 * /points_xyzrgb/
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rgbd_sync/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmap_viz/

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 306 lines cut ...]
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info[0m
[0m[ INFO] [1710085311.798630807]: rtabmap 0.21.4 started...[0m
[31m[ERROR] (2024-03-10 17:41:52.578) Rtabmap.cpp:1348::p 10/03 17:41:52,620 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/03 17:41:52,777 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/03 17:41:52,827 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/03 17:41:52,981 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/03 17:41:53,162 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/03 17:41:53,220 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/03 17:41:53,783 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/03 17:41:53,834 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
[rtabmap/rgbd_odometry-6] process has died [pid 45072, exit code -11, cmd /home/anton/Projects/rtabmap_ros/devel/lib/rtabmap_odom/rgbd_odometry --delete_db_on_start rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image odom:=odom imu:=/rtabmap/imu __name:=rgbd_odometry __log:=/home/anton/.ros/log/b31ff276-def4-11ee-9aed-4338ecc5a3fa/rtabmap-rgbd_odometry-6.log].
log file: /home/anton/.ros/log/b31ff276-def4-11ee-9aed-4338ecc5a3fa/rtabmap-rgbd_odometry-6*.log

 10/03 17:42:12,932 WARNING [140463160260352] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

matlabbe, 2024-03-13:
Which version of ORB_SLAM3 are you using? Make sure you built ORB_SLAM3 without `-march=native` (removed from these [lines](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/CMakeLists.txt#L12-L13))

You can also add
```
<arg name="gdb"         value="true"/>
```
under `rtabmap.launch` include in your launch file to launch rtabmap nodes with gdb. When odometry node crashes, do `bt` (backtrace) to see where it crashed.

lempiy, 2024-03-13:
> Which version of ORB_SLAM3

1.0 release built with 4.4 opencv. Same version used in my rtabmap and rtabmap_ros.

I didn't remove the `-march=native` param. Now I finally managed to make it work. Thank you. 
I checked this patch at the bottom
https://gist.githubusercontent.com/matlabbe/f5cb281304a1305b2824a6ce19792e13/raw/f8bbc796edc29b9f815cbf3c99a0c3e13e23663d/orbslam3_v4_rtabmap_fix.patch
realizing that the main code troubes were fixed in ORB_SLAMs v1, but didn't pay attention to `march` gcc param.

closed 2024-03-16 by matlabbe