introlab/rtabmap_ros #1137: How to choose between Scan-to-Scan or Scan-to-Map approach for 3D LiDAR odometry?

astef3, 2024-03-21:
Hi,
firstly I want to say that you did an outstanding job with RTABMap, it really works amazing. Through out the process of working with it, as far as I have read in the paper for RTABMap in the Lidar odometry explanation it says that there are two differnet approaches for estimation of the odometry: Scan-to-Scan or Scan-to-Map. 

Currently I am using Velodyne VLP16 sensor for estiamtion of the odometry and things work quite good (I am using the /rtabmap_examples/launch/test_velodyne.launch with small adaptations). I am running it on ROS Noetic. What I am interested is if someone can help me with instructions on what should I change/set in order to change the approach of estimation (S2S or S2M). 

I will be beyond grateful for every help, in case I need to provide more details I would gladly do so. Thank you in advance!

matlabbe, 2024-03-24:
Hi, thanks for the cheering!

To switch between S2S and S2M, you can set `Odom/Strategy` to 1 or 0 (default) respectively. A tip from past experiences, the lower `Odom/ScanKeyFrameThr` is the more accurate the odometry will be, but the more likely it can lose tracking, so there is a trade-off there.

cheers,
Mathieu

astef3, 2024-03-25:
Thank you for the fast response @matlabbe ! Thank you for the advice, you have saved me a lot of time!

Best regards, 
Aleksandar