introlab/rtabmap_ros #1141: exit code -6 when running rtabmap.launch on RPI4 ROS noetic

Ghada09, 2024-04-04:
pi@clover-1910:~/catkin_ws $ roslaunch rtabmap_launch rtabmap.launch \
>     args:="--delete_db_on_start" \
>     rgb_topic:=/stereo_inertial_publisher/color/image \
>     depth_topic:=/stereo_inertial_publisher/stereo/depth \
>     camera_info_topic:=/stereo_inertial_publisher/color/camera_info \
>     imu_topic:=/stereo_inertial_publisher/imu/data \
>     frame_id:=oak-d_frame \
>     approx_sync:=true \
>     wait_imu_to_init:=false
... logging to /home/pi/.ros/log/9360d9d4-f2de-11ee-acbd-d83add3555ba/roslaunch-clover-1910-28763.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://clover-1910.local:40099/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmap_viz/

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/approx_sync_max_interval: 0.0
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: oak-d_frame
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 5
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: oak-d_frame
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/loc_thr: 0.0
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 5
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap_viz/approx_sync: True
 * /rtabmap/rtabmap_viz/frame_id: oak-d_frame
 * /rtabmap/rtabmap_viz/odom_frame_id: 
 * /rtabmap/rtabmap_viz/queue_size: 5
 * /rtabmap/rtabmap_viz/subscribe_depth: True
 * /rtabmap/rtabmap_viz/subscribe_odom_info: True
 * /rtabmap/rtabmap_viz/subscribe_rgb: True
 * /rtabmap/rtabmap_viz/subscribe_rgbd: False
 * /rtabmap/rtabmap_viz/subscribe_scan: False
 * /rtabmap/rtabmap_viz/subscribe_scan_cloud: False
 * /rtabmap/rtabmap_viz/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap_viz/subscribe_stereo: False
 * /rtabmap/rtabmap_viz/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_odom/rgbd_odometry)
    rtabmap (rtabmap_slam/rtabmap)
    rtabmap_viz (rtabmap_viz/rtabmap_viz)

ROS_MASTER_URI=http://localhost:11311

process[rtabmap/rgbd_odometry-1]: started with pid [28787]
process[rtabmap/rtabmap-2]: started with pid [28788]
process[rtabmap/rtabmap_viz-3]: started with pid [28789]
rtabmap_viz: malloc.c:2385: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
[rtabmap/rtabmap_viz-3] process has died [pid 28789, exit code -6, cmd /home/pi/catkin_ws/devel/lib/rtabmap_viz/rtabmap_viz -d ~/.ros/rtabmap_gui.ini rgb/image:=/stereo_inertial_publisher/color/image depth/image:=/stereo_inertial_publisher/stereo/depth rgb/camera_info:=/stereo_inertial_publisher/color/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud scan_descriptor:=/scan_descriptor odom:=odom __name:=rtabmap_viz __log:=/home/pi/.ros/log/9360d9d4-f2de-11ee-acbd-d83add3555ba/rtabmap-rtabmap_viz-3.log].
log file: /home/pi/.ros/log/9360d9d4-f2de-11ee-acbd-d83add3555ba/rtabmap-rtabmap_viz-3*.log
rtabmap: malloc.c:2385: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
[rtabmap/rtabmap-2] process has died [pid 28788, exit code -6, cmd /home/pi/catkin_ws/devel/lib/rtabmap_slam/rtabmap --delete_db_on_start rgb/image:=/stereo_inertial_publisher/color/image depth/image:=/stereo_inertial_publisher/stereo/depth rgb/camera_info:=/stereo_inertial_publisher/color/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud scan_descriptor:=/scan_descriptor user_data:=/user_data user_data_async:=/user_data_async gps/fix:=/gps/fix tag_detections:=/tag_detections fiducial_transforms:=/fiducial_transforms odom:=odom imu:=/stereo_inertial_publisher/imu/data __name:=rtabmap __log:=/home/pi/.ros/log/9360d9d4-f2de-11ee-acbd-d83add3555ba/rtabmap-rtabmap-2.log].
log file: /home/pi/.ros/log/9360d9d4-f2de-11ee-acbd-d83add3555ba/rtabmap-rtabmap-2*.log
rgbd_odometry: malloc.c:2385: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
[rtabmap/rgbd_odometry-1] process has died [pid 28787, exit code -6, cmd /home/pi/catkin_ws/devel/lib/rtabmap_odom/rgbd_odometry --delete_db_on_start rgb/image:=/stereo_inertial_publisher/color/image depth/image:=/stereo_inertial_publisher/stereo/depth rgb/camera_info:=/stereo_inertial_publisher/color/camera_info rgbd_image:=rgbd_image_relay odom:=odom imu:=/stereo_inertial_publisher/imu/data __name:=rgbd_odometry __log:=/home/pi/.ros/log/9360d9d4-f2de-11ee-acbd-d83add3555ba/rtabmap-rgbd_odometry-1.log].
log file: /home/pi/.ros/log/9360d9d4-f2de-11ee-acbd-d83add3555ba/rtabmap-rgbd_odometry-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

matlabbe, 2024-04-06:
Are you using rtabmap noetic binaries or it has been rebuilt from source?

May be a duplicate of https://github.com/introlab/rtabmap/issues/872

Ghada09, 2024-04-07:
I built it from source, I tried the solutions in the link provided but nothing changed

matlabbe, 2024-04-07:
I'll give a try on my RPI4, but in the mean time can you show output of cmake for rtabmap library?
```
cd rtabmap/build
cmake ..
```

Also can you show output of:
```
ldd ~/catkin_ws/devel/lib/rtabmap_slam/rtabmap
```

Can you run `rtabmap` alone? (typing `rtabmap` in terminal)

matlabbe, 2024-04-08:
If you want to avoid rebuilding from source to use latest dependencies (which is pretty long on RPI4), you can use latest docker image directly: https://github.com/introlab/rtabmap_ros/tree/master/docker 

Your launch would look like this instead:
```
docker run -it --rm \
 --user $UID \
 -e ROS_HOME=/tmp/.ros \
 --network host \
 -v ~/.ros:/tmp/.ros \
 introlab3it/rtabmap_ros:noetic-latest \
 roslaunch rtabmap_launch rtabmap.launch rtabmap_viz:=false database_path:=/tmp/.ros/rtabmap.db rtabmap_args:="--delete_db_on_start" \
 rgb_topic:=/stereo_inertial_publisher/color/image \
 depth_topic:=/stereo_inertial_publisher/stereo/depth \
 camera_info_topic:=/stereo_inertial_publisher/color/camera_info \
 imu_topic:=/stereo_inertial_publisher/imu/data \
 frame_id:=oak-d_frame \
 approx_sync:=true \
 wait_imu_to_init:=false

Ghada09, 2024-04-08:
**This is the cmake output:**
pi@clover-9222:~/rtabmap/build $ cmake ..
-- MOBILE_BUILD=OFF
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/arm-linux-gnueabihf/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- FLANN found (include: , lib: lz4;flann;flann_cpp)
-- FLANN found (include: , lib: lz4;flann;flann_cpp)
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/arm-linux-gnueabihf/libqhull.so;debug;/usr/lib/arm-linux-gnueabihf/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found SQLite3: /usr/include /usr/lib/arm-linux-gnueabihf/libsqlite3.so
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE  found components: C CXX 
-- Found OpenMP: /usr/lib/gcc/arm-linux-gnueabihf/8/libgomp.so;/usr/lib/arm-linux-gnueabihf/libpthread.so
-- Add PCL_OMP to definitions
-- Found OpenCV: /usr/include/opencv4
-- Found PCL: /usr/include/pcl-1.9;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-7.1;/usr/include/freetype2;/usr/include/arm-linux-gnueabihf;/usr/lib/arm-linux-gnueabihf/openmpi/include/openmpi;/usr/lib/arm-linux-gnueabihf/openmpi/include;/usr/include/python3.7m;/usr/include/hdf5/openmpi;/usr/include/jsoncpp;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL2
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Could NOT find g2o (missing: g2o_DIR)
-- RTAB-Map's cmake g2o find module used for convenience (for older g2o versions)
-- Could NOT find CSPARSE (missing: CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY) 
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Architecture: armv7l
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   RTAB-Map Version =     0.21.4
--   CMAKE_VERSION =        3.16.3
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
[... 631 lines cut ...]
	libnss3.so => /lib/arm-linux-gnueabihf/libnss3.so (0x964fe000)
	libsmime3.so => /lib/arm-linux-gnueabihf/libsmime3.so (0x964cd000)
	libnspr4.so => /lib/arm-linux-gnueabihf/libnspr4.so (0x9648d000)
	liblcms2.so.2 => /lib/arm-linux-gnueabihf/liblcms2.so.2 (0x96436000)
	libgeos-3.7.1.so => /lib/arm-linux-gnueabihf/libgeos-3.7.1.so (0x962dc000)
	libpopt.so.0 => /lib/arm-linux-gnueabihf/libpopt.so.0 (0x962c1000)
	libminizip.so.1 => /lib/arm-linux-gnueabihf/libminizip.so.1 (0x962a7000)
	liburiparser.so.1 => /lib/arm-linux-gnueabihf/liburiparser.so.1 (0x9627c000)
	libkmlconvenience.so.1 => /lib/arm-linux-gnueabihf/libkmlconvenience.so.1 (0x96250000)
	libnsl.so.1 => /lib/arm-linux-gnueabihf/libnsl.so.1 (0x9622c000)
	libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0x961db000)
	libatspi.so.0 => /lib/arm-linux-gnueabihf/libatspi.so.0 (0x961a4000)
	libdatrie.so.1 => /lib/arm-linux-gnueabihf/libdatrie.so.1 (0x9618e000)
	libblkid.so.1 => /lib/arm-linux-gnueabihf/libblkid.so.1 (0x96136000)
	liborc-0.4.so.0 => /lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0x960bc000)
	libgfortran.so.5 => /lib/arm-linux-gnueabihf/libgfortran.so.5 (0x95fe1000)
	libnssutil3.so => /lib/arm-linux-gnueabihf/libnssutil3.so (0x95fae000)
	libplc4.so => /lib/arm-linux-gnueabihf/libplc4.so (0x95f9a000)
	libplds4.so => /lib/arm-linux-gnueabihf/libplds4.so (0x95f87000)
	libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0x95eea000)

**and yes I can run rtabmap alone, this is the output:**
[image]

**Finally, I am using Raspbian GNU/Linux 10 (buster), while you read through this, I will try the docker thing, thanks a lot.**

Ghada09, 2024-04-08:
I tried the docker, it gives me this error: 
pi@clover-9222:~ $ sudo docker pull introlab3it/rtabmap_ros:noetic-latest
noetic-latest: Pulling from introlab3it/rtabmap_ros
no matching manifest for linux/arm/v7 in the manifest list entries

matlabbe, 2024-04-08:
The [noetic armhf build for rtabmap](https://build.ros.org/job/Nbin_ufhf_uFhf__rtabmap__ubuntu_focal_armhf__binary/) is currently disabled because `ros-noetic-libpointmatcher` doesn't exist on 32 bits systems.

I am not sure why you get a malloc issue with the ros package, sometimes it is related to different versions of same libraries used by rtabmap library and ros libraries (like different version of opencv). 

The best fix would be to add rtabmap docker build on armhf too on our side. I'll check how feasible is that because it was disabled in this [commit](https://github.com/introlab/rtabmap_ros/commit/74d48b402b3698169ccd686bd456ac72e8113d40). Otherwise in the mean time, if you don't care about reflashing your RPI4, you can install a 64 bits version or arm on it to use the docker above. Recompiling from source on Ubuntu Mate 20.04 also works, though I prefer using docker on RPI (a lot faster to install!).

matlabbe, 2024-04-11:
[noetic-latest](https://hub.docker.com/layers/introlab3it/rtabmap_ros/noetic-latest/images/sha256-708ad84731642fc2b04af31d7269f90abd84f3f96165638d2e781684c583a693?context=explore) image supports armhf now, you may give a try:
```
docker pull introlab3it/rtabmap_ros:noetic-latest
```

AyaMoustafa5, 2024-04-14:
When I run the docker container as follows:
sudo docker run -it --rm \
 --user $UID \
 -e ROS_HOME=/tmp/.ros \
 --network host \
 -v ~/.ros:/tmp/.ros \
 introlab3it/rtabmap_ros:noetic-latest \
 roslaunch rtabmap_launch rtabmap.launch rtabmap_viz:=false database_path:=/tmp/.ros/rtabmap.db rtabmap_args:="--delete_db_on_start" \
 rgb_topic:=/stereo_inertial_publisher/color/image \
 depth_topic:=/stereo_inertial_publisher/stereo/depth \
 camera_info_topic:=/stereo_inertial_publisher/color/camera_info \
 imu_topic:=/stereo_inertial_publisher/imu/data \
 frame_id:=oak-d_frame \
 approx_sync:=true \
 wait_imu_to_init:=false  

I get this error:
[image]

Do you have any idea how can I solve this error?

matlabbe, 2024-04-14:
If you "sudo" docker, try removing `--user $UID`

AyaMoustafa5, 2024-04-14:
When I do so, I get this error:

Fatal Python error: pyinit_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted

Current thread 0xb6f1b5b0 (most recent call first):
<no Python frame>

matlabbe, 2024-04-15:
I tested 32 bits version from a [fresh raspberry pi image](https://www.raspberrypi.com/documentation/computers/getting-started.html#install-an-operating-system) and it should work. Here the steps I did: 

 * Setup docker:
```bash
curl -sSL https://get.docker.com | sh
sudo usermod -aG docker $USER
```

* Reboot, or login/logout, make sure this works without sudo:
```bash
docker run hello-world
```

* Make sure "~/.ros" directory exists on host
```bash
mkdir ~/.ros
```
* Follow directives from https://github.com/introlab/rtabmap_ros/tree/master/docker#docker with `noetic-latest` image:
```bash
docker run -it --rm \
 --user $UID \
 -e ROS_HOME=/tmp/.ros \
 --network host \
 -v ~/.ros:/tmp/.ros \
 introlab3it/rtabmap_ros:noetic-latest \
 roslaunch rtabmap_launch rtabmap.launch rtabmap_viz:=false database_path:=/tmp/.ros/rtabmap.db rtabmap_args:="--delete_db_on_start"
```
* Launch ui (without nvidia runtime)
```bash
XAUTH=/tmp/.docker.xauth
touch $XAUTH
xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -

docker run -it --rm \
  --privileged \
  -e DISPLAY=$DISPLAY \
  -e QT_X11_NO_MITSHM=1 \
  -e NVIDIA_VISIBLE_DEVICES=all \
  -e NVIDIA_DRIVER_CAPABILITIES=all \
  -e XAUTHORITY=$XAUTH \
  --network host \
  -v $XAUTH:$XAUTH \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  introlab3it/rtabmap_ros:noetic-latest \
  /bin/bash -c "export ROS_NAMESPACE=rtabmap && rosrun rtabmap_viz rtabmap_viz"
```

Ghada09, 2024-04-17:
its working now but i am getting a warning related to the time difference:

[ WARN] [1713360148.659939544]: The time difference between rgb and depth frames is high (diff=0.066622s, rgb=1713360147.231447s, depth=1713360147.164826s). You may want to set approx_sync_max_interval lower than 0.02s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.

when I try the approx_sync=false it doesn't work.

matlabbe, 2024-04-17:
Are both camera streams at the same frame rate? Should they be 30 Hz or 15Hz?

Probably your camera driver is not publishing all camera streams with exactly the same stamp, so you cannot use `approx_sync:=false`. 

You have to fix your camera driver to provide a more stable stream.