introlab/rtabmap_ros #1146: mapping object detection data

FlorianMehnert, 2024-04-15:
Hi there,

is it possible to use rtabmap with object detection data or are there better alternatives for storing detected objects?  I found a similar project that does basically what I want using rtabmap where they drive around detecting objects and storing them in rtabmap for later use: [https://github.com/ROBOTIS-JAPAN-GIT/turtlebot3_slam_3d](https://github.com/ROBOTIS-JAPAN-GIT/turtlebot3_slam_3d). 
[image] 
[image]
Since I ran into issues getting the turtlebot3_slam_3d to run (not maintained anymore) I was wondering whether this is supported out of the box/there are any attempts to add a similar feature to rtabmap?

matlabbe, 2024-04-16:
It seems they collected the objects in another file, with name and pose in the current map frame. As they are not saving against the graph, I am not sure if the pose of the objects is corrected on a loop closure. We have an example [here](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Robot_mapping_with_Find-Object) where objects were added to the graph, then shown back in rviz (follow `save_objects` argument). The main code is in this [example](https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_demos/src/SaveObjectsExample.cpp). There are two callbacks: one from some object detection topic that encodes detection in UserData msg to be recorded in rtabmap node, and one that subscribe to published MapData from rtabmap node an decode the UserData to create markers to be shown in rviz. If the map's graph changes, the markers are updated.

FlorianMehnert, 2024-04-17:
for the find object example I ran into a similar issue as #509 but the same fix with a small python script works

matlabbe, 2024-04-17:
Thanks to point out, I'll reupload a fixed rosbag for that demo.

matlabbe, 2024-04-27:
[Bag](https://docs.google.com/uc?export=download&confirm=XJb5&id=0B46akLGdg-uaUEVEX1VmQTdOQUE) has been updated.

closed 2024-04-27 by matlabbe