introlab/rtabmap_ros #1147: Problems with path planning using d435i camera navigation

smile6023326, 2024-04-16:
Hello, when I use rtabmap to navigate, why does the robot walk on the left side of the path plan? If the navigation is successful, I will share the program code with those who need it.
Thank you for taking the time to answer.

my tf tree:
[image]

mapping:
roslaunch realsense2_camera rs_rgbd.launch

roslaunch my_package rtabmap402.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/depth/image_rect_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info

rviz

navigation : 
roslaunch realsense2_camera rs_rgbd.launch

roslaunch my_package rtabmap402.launch database_path:="~/.ros/rtabmap.db" depth_topic:=/camera/depth/image_rect_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info localization:=true

roslaunch my_package robot_rtabmap_nav.launch

.launch:
[rs_rgbd.txt](https://github.com/introlab/rtabmap_ros/files/14997562/rs_rgbd.txt)
[rtabmap402.txt](https://github.com/introlab/rtabmap_ros/files/14997564/rtabmap402.txt)
[robot_rtabmap_nav.txt](https://github.com/introlab/rtabmap_ros/files/14997613/robot_rtabmap_nav.txt)

In actual situations, the green line is the planned path, and my robot does not follow the green line.
[image]

matlabbe, 2024-04-17:
If the green line is the global plan, it is normal that the robot doesn't follow it exactly. You may vheck the output of the local planner, I guess this one is planning slightly off the global one to be farther from obstacles.

smile6023326, 2024-04-18:
But when I use this method to navigate, there are no obstacles in front of me.
my robot Length 90cm,Width 70cm,Height 95cm.
D435i is located on a flat surface with a height of 90cm.
This is my video, the speed is much faster due to the upload size limit.
https://github.com/introlab/rtabmap_ros/assets/48814064/0579eb54-acd4-4676-9b2d-7ab086695e1a

matlabbe, 2024-04-28:
Try to visualize the local costmap at the same time, to see if the camera is seeing some ghost obstacles (noise).