introlab/rtabmap_ros #1148: Resolution parameter explanation

astef3, 2024-04-23:
Hi, 

I would have a simple question about the parameter for the resolution. I am currently using test_velodyne.launch file and I am playing a little with the resolution, as it says that there are different recommended values for the parameter based on whether its an outdoors or indoors environment. So, I would be really grateful if you can explain to me a little more about the parameter (the meaning and the units of the parameter).

Thank you in advance!

matlabbe, 2024-04-28:
Units are in meters. For example, if it is `0.1`, it will filter the point cloud to keep only 1 point for each 10 cm voxel. See https://pcl.readthedocs.io/projects/tutorials/en/latest/voxel_grid.html#voxelgrid for a more in depth explanation.

We voxel filter the point cloud to have a more uniform distribution of the points in the environment. This will help ICP to converge globally, and not only points close to the sensor (with lidar sensors, the surfaces are covered by more points the closer they are from the sensor). The larger the voxel, the faster ICP will be too (less points).

astef3, 2024-05-12:
Thank you for the explanation! Its perfect, now I understand!

closed 2024-05-12 by astef3