introlab/rtabmap_ros #1150: ROS2 Humble runs rtabmap slam, there is data in /mapData but no 3D map appears in rviz2

miku54, 2024-04-24:
Hi everyone, I'm using ros2-humble to test rtabmap slam and navigation, using [sudo apt install ros-humble-rtabmap-*] to install the algorithm with version number 0.21.4-2 jammy.20240416. My steps are as follows:
1. Start lidar
2. Start the binocular camera
3. Start rtabmap slam
During the test, it was found that everything ran normally, with 0 errors and 0 warnings. I cannot view the 3D map in rviz2. Is there any problem with the operation?
[image]

matlabbe, 2024-04-29:
You have "Cloud from scan" checked in MapCloud plugin, to see the camera cloud, uncheck that.

miku54, 2024-04-29:
> You have "Cloud from scan" checked in MapCloud plugin, to see the camera cloud, uncheck that.

I want to prove in this way that there is no problem with rtabmap navigation. When I try to open an rviz2 again and cancel Cloud from scan, I still cannot display the 3D map? Is it a problem with the Raspberry Pi?

matlabbe, 2024-04-29:
I am not aware of your full config, but there could be a possibility that rtabmap node is not subscribed to image data. You can also use rtabmap-databaseViewer, to verify that you have image and scan data in the database.

matlabbe, 2024-05-12:
With [MapData](http://docs.ros.org/en/noetic/api/rtabmap_msgs/html/msg/MapData.html) msg, if `nodes` has image field `left` set in their [SensorData](http://docs.ros.org/en/noetic/api/rtabmap_msgs/html/msg/Node.html), then yes images are there:
http://docs.ros.org/en/noetic/api/rtabmap_msgs/html/msg/SensorData.html

miku54, 2024-10-22:
Solved, reference:https://github.com/introlab/rtabmap_ros/issues/1223