introlab/rtabmap_ros #1155: Rtabmap - random rotation and jump when mapping.

Krzo99, 2024-05-17:
I came across an issue, that I can't seem to find a way to resolve. 
Hoping there would be someone here that had something similar happen to them.

I'm running an autonomous exploration algorithm on a mobile robot. I'm using ROS2 Humble.
Sometimes, the map appears to reset the robot to (0, 0) (video down below), and rotate it in the middle of mapping, which then means that my algorithm doesn't work as expected. The robot position, relative to the map stays the same. 
I'm using a clean database (just turn on mapping and go, no other maps are saved in the db). 

The full map (and the robot's position in it) stays the same, it's just its location and rotation in /map tf that changes. I don't have data, but I'm guessing /odom -> /map changes. I can't see /odom -> /base_link changing. Or could it? IMU data looks normal while the change happens (When red line (x coordinate) jumps.) There also isn't any "odom_correction_yaw" and "map2odom_yaw" messages recieved on the /info topic around the time or the jump.
[image]

My rtabmap launch paremeters are:
```
        'frame_id': PathJoinSubstitution([LaunchConfiguration("robot_ns"), 'base_footprint']),
        'map_frame_id': 'map',
        'odom_frame_id': PathJoinSubstitution([LaunchConfiguration("robot_ns"), 'odom']),
        'subscribe_depth': True,
        'subscribe_odom_info': False,
        'approx_sync': True,
        'approx_sync_max_interval': 0.02,
        'wait_imu_to_init':True,
        'use_sim_time': LaunchConfiguration("is_sim"),
        "map_always_update": False,                                  # Keep sending the full map on cloud_map topic.
        "database_path": LaunchConfiguration("database_path"),       # Each robot has its own database.

        'Rtabmap/DetectionRate': '1.0',              # Detection rate (Hz). RTAB-Map will filter input images to satisfy this rate.
        'Rtabmap/ImagesAlreadyRectified': 'false',   # By default RTAB-Map assumes that received images are rectified. If they are not, they can be rectified by RTAB-Map if this parameter is false.
        'Rtabmap/LoopThr': '0.11',                   # Loop closing threshold.
        'Odom/Strategy': '0',                        # [0=Frame-to-Map (F2M) 1=Frame-to-Frame (F2F) 2=Fovis 3=viso2 4=DVO-SLAM 5=ORB_SLAM2 6=OKVIS 7=LOAM 8=MSCKF_VIO 9=VINS-Fusion 10=OpenVINS 11=FLOAM 12=Open3D]
        'Odom/FilteringStrategy': '1' ,              # [0=No filtering 1=Kalman filtering 2=Particle filtering.]
        'Optimizer/Strategy': '2',                   # Graph optimization strategy: 0=TORO, 1=g2o, 2=GTSAM and 3=Ceres. Only 1 or 2 can be used.
        'Optimizer/Robust': 'true',                  # Robust graph optimization using Vertigo (only work for g2o and GTSAM optimization strategies). Not compatible with 'RGBD/OptimizeMaxError' if enabled.
        "Optimizer/GravitySigma": "0.3",
        "RGBD/SavedLocalizationIgnored": 'false',    # Ignore saved localization if the loop closure is detected. This will force RTAB-Map to localize again in the current map.
        "Mem/IncrementalMemory": "false",            # true = mapping mode
        "Mem/ReduceGraph": "true",                   # Shuld make the database smaller.
        "Mem/SaveDepth16Format": "true",             # Save depth images in 16 bit format to save size.
        "Mem/UseOdomGravity": "true",                # Use odometry instead of IMU orientation to add gravity links to new nodes created.

        "Reg/Strategy": "0",                         # We use only visual odometry.
```

The video of the incident is bellow. We can clearly see that the robot jumps back to another location (I think it's (0, 0)). The map also rotates 90°.

https://github.com/introlab/rtabmap_ros/assets/20487367/b65898af-2329-4599-ab98-192a942ba96b

I'm wondering why this is happening. I don't think it could be a bad Loop closure, as then the robot's location would move on the map. Furthermore, I'll be happy to provide additional info, and really glad, if anyone can point me in the right direction.

Thanks!

matlabbe, 2024-05-20:
Could you show TF frames (with their label) in rviz? I see it is in localization mode, maybe it relocalized?

Krzo99, 2024-05-27:
Hey,
Sorry for a late reply.
I couldn't reproduce the error again. Maybe it got fixed when I updated the ros packages.
I'll reopen this issue if it comes up again.

closed 2024-05-27 by Krzo99