introlab/rtabmap_ros #1157: /map frame is rotating randomly.

sanjay-swain, 2024-05-23:
I am trying a simple example of handheld mapping. But when all the nodes are running I find the entire frame is constantly as shown in the video below. I have no idea what is causing the issue. I have tried checking the imu of the realsense but it is fine.

Error output:
```
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:28.588) OdometryF2M.cpp:576::computeTransform() Trial with no guess succeeded!
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:28.711) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=83) between -1 and 275" (guess=xyz=-0.000934,-0.000062,0.000550 rpy=-0.466382,0.309671,-0.041613)
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:28.712) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.000934,-0.000062,0.000550 rpy=-0.466382,0.309671,-0.041613), trying again without a guess.
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:28.831) OdometryF2M.cpp:576::computeTransform() Trial with no guess succeeded!
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:28.931) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=105) between -1 and 276" (guess=xyz=0.000442,-0.000032,0.000011 rpy=-0.416033,0.260958,0.063517)
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:28.931) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=0.000442,-0.000032,0.000011 rpy=-0.416033,0.260958,0.063517), trying again without a guess.
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:29.053) OdometryF2M.cpp:576::computeTransform() Trial with no guess succeeded!
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:29.155) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=93) between -1 and 277" (guess=xyz=-0.000225,0.000345,-0.000354 rpy=-0.344084,0.274568,-0.050905)
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:29.155) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.000225,0.000345,-0.000354 rpy=-0.344084,0.274568,-0.050905), trying again without a guess.
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:29.271) OdometryF2M.cpp:576::computeTransform() Trial with no guess succeeded!
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:29.373) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=87) between -1 and 278" (guess=xyz=-0.000215,-0.000215,-0.000315 rpy=-0.113920,0.309356,0.165270)
[rgbd_odometry-1] [ WARN] (2024-05-23 16:29:29.373) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.000215,-0.000215,-0.000315 rpy=-0.113920,0.309356,0.165270), trying again without a guess.

```

https://github.com/introlab/rtabmap_ros/assets/59828852/769db746-1476-4148-bb72-90976dbcaab3

Any guidance will be helpful. I have tried multiple settings change from camera side but same thing is happening...

matlabbe, 2024-05-26:
Could be a calibration issue. What are the commands you executed? I could doublecheck on my side with a realsense. The processing time looks high (200-300 ms per frame), what kind of computer are you using? Depending on the realsense you are using, I generally recommend to use the IR cameras (without ir projector) for better pose estimation, like this example https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/realsense_d435i_infra.launch.py

sanjay-swain, 2024-05-26:
I am running it on my laptop. i5 9300H processor... It is a D455. Let me look into those examples.

matlabbe, 2024-05-27:
Can you run without the IMU? I cannot reproduce the issue with my camera and same example. Which rtabmap version are you using?

sanjay-swain, 2024-05-29:
Yes the imu data is faulty. For some reason the realsense node is outputting bad gyro data to /camera/imu topic... But when I check on `realsense-viewer` the gyro and acceleration streams are completely fine... Are you using D455 camera for your tests? 

I found another issue reported here https://github.com/CCNYRoboticsLab/imu_tools/issues/193. It seems like the same issue as I am facing...

I am using ros humble version rtabmap:

```
❯ rtabmap --version
RTAB-Map:               0.21.4
PCL:                    1.12.1
With VTK:                9.1.0
OpenCV:                  4.5.4
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:           false
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With Open3D:             false
With OctoMap:             true
With GridMap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

Thanks for your support!

sanjay-swain, 2024-05-29:
Update: Well I rebuilt `realsense-ros` package with latest commit and it is fixed now. It was an issue with humble version of `realsense-ros` which was giving bad gyro data to topic...

closed 2024-05-29 by sanjay-swain

anath93, 2024-06-07:
@sanjay-swain same issue here in terms of drift, realsense-viewer is fine, which branch did you end up using on librealsense and realsense-ros side ?

sanjay-swain, 2024-06-08:
Hey. I used the current ros 2 branch with the latest commits. My logic behind this issue is that ros humble version of realsense ros is outdated and librealsense library is updated to latest version. Due to this realsense Ros is not working properly.

5dollars0909, 2025-03-02:
I am currently encountering this issue as well. Below is my file to open. Can you help me take a look? This is my video, what should I do? Sorry for the trouble, everyone

$ roslaunch realsense2_camera rs_camera.launch \
align_depth:=true \
unite_imu_method:="linear_interpolation" \
enable_gyro:=true \
enable_accel:=true \
depth_fps:=6 \
color_fps:=6 \
infra_fps:=6

$ rosrun imu_filter_madgwick imu_filter_node \
_use_mag:=false \
_publish_tf:=false \
_world_frame:="enu" \
/imu/data_raw:=/camera/imu \
/imu/data:=/rtabmap/imu

$ roslaunch rtabmap_ros rtabmap.launch \
rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
depth_topic:=/camera/aligned_depth_to_color/image_raw \
rgb_topic:=/camera/color/image_raw \
camera_info_topic:=/camera/color/camera_info \
approx_sync:=false \
wait_imu_to_init:=true \
imu_topic:=/rtabmap/imu \
rviz:=true \
rtabmapviz:=false

[https://drive.google.com/file/d/13-qM1Xg4qY6MumutcsyGn1q4XVIJHGv2/view?usp=drive_link](url)

sanjay-swain, 2025-03-03:
I have solved the issue by using another version of `realsense-ros`. You can try multiple versions. For some reason the official binary have a bug which gives bad IMU data..

5dollars0909, 2025-03-03:
Thanks for your reply ! Can i know your version of realsense-ros?

anath93, 2025-03-03:
@5dollars0909 Hope this will help you which did to me before I switched from Realsense to ZED,

1. Try to calibrate the IMU of realsense first using there below doc following in 4.3.2 and verify in your base_frame and odom_frame if your robot is still rotating. To go further down, you can also try to do unite_imu_method:=copy_method (please check the syntax) and see how your gyro values look like in ros2 topic echo /camera/imu and if you still see noise follow the below which I did to eliminate it by building the sdk
https://www.intelrealsense.com/wp-content/uploads/2019/07/Intel_RealSense_Depth_D435i_IMU_Calibration.pdf

2. git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
git checkout development
commit 541f542c85514034a7341923540df28809c3f10e (HEAD -> development, origin/development)

2. apply the below code
$ git diff src/proc/motion-transform.cpp
diff --git a/src/proc/motion-transform.cpp b/src/proc/motion-transform.cpp
index dd27c6591..5b080fff7 100644
--- a/src/proc/motion-transform.cpp
+++ b/src/proc/motion-transform.cpp
@@ -121,6 +121,8 @@ namespace librealsense
return ret;
}
+#define GYRO_NOISE_LIMIT 0.05
+
void motion_transform::correct_motion_helper(float3* xyz, rs2_stream stream_type) const
{
// The IMU sensor orientation shall be aligned with depth sensor's coordinate system
@@ -139,6 +141,17 @@ namespace librealsense
*xyz = _gyro_sensitivity * (*xyz) - _gyro_bias;
}
}
+
 +   if (stream_type == RS2_STREAM_GYRO)
 +   {
  +  // temporary workaround - gyro drift due to noise, adjust GYRO_NOISE_LIMIT as appropriate
  +  if (xyz->length() < GYRO_NOISE_LIMIT)
  +  {
  +  xyz->x = 0.0;
  +  xyz->y = 0.0;
  +  xyz->z = 0.0;
  +  }
  +  }
}
void motion_transform::correct_motion(rs2::frame* f) const
{
 
3. patch the kernel for v4l2 backend
./scripts/patch-realsense-ubuntu-L4T.sh
 
4. compile as usual without the RSUSB option and install or with RSUSB, your choice,

5. ros wrapper
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros
git checkout ros2-development
 
commit 4e94c66fc2a2d4b565fc4da88cd0e769dda754f8 

6. Rebuild the realsense-ros against this new SDK changes using colcon and run

5dollars0909, 2025-03-04:
THANK YOU SO MUCH !!! Thanks for your help !