introlab/rtabmap_ros #1168: Creating RTABMAP from Ground Truth Poses

lakshmikanth-nageswar, 2024-06-04:
Can we perform mapping against known ground truth poses? I am particularly looking for mapping in RGBD mode with known ground truth poses. Kindly describe the procedure if it's possible.

Your help would be greatly appreciated !

matlabbe, 2024-06-10:
Yes it could be possible. You can feed the ground truth pose as odometry topic with very small covariance, or use `odom_frame_id` to use TF instead. Another way is to provide `/rtabmap/global_pose` topic instead along your normal odometry, with `Optimizer/PriorsIgnored=False`.

You can disable loop closure detection by setting `Kp/MaxFeatures=-1`.

closed 2024-07-17 by lakshmikanth-nageswar

wojtess, 2025-07-05:
>You can feed the ground truth pose as odometry topic with very small covariance

@matlabbe 
 how to set topic name in witch it is published?

I have ground truth on this topic:
$ ros2 topic info /wamv/sensors/position/ground_truth_odometry -v
Type: nav_msgs/msg/Odometry

Publisher count: 1

Node name: ros_gz_bridge
Node namespace: /wamv
Topic type: nav_msgs/msg/Odometry
Endpoint type: PUBLISHER
GID: 01.0f.56.e7.b6.0f.60.54.00.00.00.00.00.00.1e.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: rqt_gui_py_node_172323
Node namespace: /
Topic type: nav_msgs/msg/Odometry
Endpoint type: SUBSCRIPTION
GID: 01.0f.56.e7.23.a1.fc.f9.00.00.00.00.00.00.11.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

How to use it in rtabmap?

ArghyaChatterjee, 2025-07-23:
Hello, 

Thanks for this. I was trying to generate some pseudo ground truth map from ground truth pose and depth. Here is my bag file condition:
```
Topic information: Topic: /tf | Type: tf2_msgs/msg/TFMessage | Count: 109537 | Serialization Format: cdr
                   Topic: /imu/clean/imu | Type: sensor_msgs/msg/Imu | Count: 101213 | Serialization Format: cdr
                   Topic: /odom | Type: nav_msgs/msg/Odometry | Count: 101213 | Serialization Format: cdr
                   Topic: /imu/noisy/imu | Type: sensor_msgs/msg/Imu | Count: 101212 | Serialization Format: cdr
                   Topic: /front_lidar/scan | Type: sensor_msgs/msg/LaserScan | Count: 16962 | Serialization Format: cdr
                   Topic: /rear_lidar/scan | Type: sensor_msgs/msg/LaserScan | Count: 16962 | Serialization Format: cdr
                   Topic: /left_cam/rgb/image_raw | Type: sensor_msgs/msg/Image | Count: 8307 | Serialization Format: cdr
                   Topic: /left_cam/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 8307 | Serialization Format: cdr
                   Topic: /left_cam/mono/image_raw | Type: sensor_msgs/msg/Image | Count: 8307 | Serialization Format: cdr
                   Topic: /right_cam/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 8307 | Serialization Format: cdr
                   Topic: /right_cam/rgb/image_raw | Type: sensor_msgs/msg/Image | Count: 8307 | Serialization Format: cdr
                   Topic: /right_cam/mono/image_raw | Type: sensor_msgs/msg/Image | Count: 8307 | Serialization Format: cdr
                   Topic: /seg_cam/converted/image_raw | Type: sensor_msgs/msg/Image | Count: 8307 | Serialization Format: cdr
                   Topic: /seg_cam/rgb/image_raw | Type: sensor_msgs/msg/Image | Count: 8307 | Serialization Format: cdr
                   Topic: /seg_cam/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 8307 | Serialization Format: cdr
                   Topic: /depth_cam/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 8307 | Serialization Format: cdr
                   Topic: /depth_cam/mono/image_raw | Type: sensor_msgs/msg/Image | Count: 8307 | Serialization Format: cdr
                   Topic: /imu/noisy/biases/gyro | Type: geometry_msgs/msg/Vector3Stamped | Count: 101150 | Serialization Format: cdr
                   Topic: /imu/noisy/biases/accel | Type: geometry_msgs/msg/Vector3Stamped | Count: 101150 | Serialization Format: cdr
                   Topic: /clock | Type: rosgraph_msgs/msg/Clock | Count: 10151 | Serialization Format: cdr
                   Topic: /tf_static | Type: tf2_msgs/msg/TFMessage | Count: 1 | Serialization Format: cdr
```

Now, I tried with multiple commands but at some point or the other, it fails to create a complete map. It does great initially but it fails after sometime. Here is the command line that I am using (Installed the RTABMap Binary Package):
```
ros2 launch rtabmap_launch rtabmap.launch.py \
  rtabmap_args:="--delete_db_on_start" \
  rgb_topic:=/left_cam/rgb/image_raw \
  depth_topic:=/depth_cam/mono/image_raw \
  camera_info_topic:=/left_cam/camera_info \
  odom_topic:=/odom \
  frame_id:=base_link_gt \
  approx_sync:=false \
  rviz:=true
```

I have also tried with `approx_sync:=true` but didn't work either. Since I have the ground truth odometry, I thought it wouldn't be an issue. Here is the error that I see keeps rolling:
```
[rgbd_odometry-1] [ WARN] (2025-07-22 22:24:52.597) OdometryF2M.cpp:622::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=75) between -1 and 122"
[rgbd_odometry-1] [INFO] [1753241092.598219686] [rtabmap.rgbd_odometry]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.055386s delay=1753241018.793334s
[rviz2-4] [INFO] [1753241092.906513451] [rviz2]: Message Filter dropping message: frame 'left_cam' at time 67.305 for reason 'discarding message because the queue is full'
[rgbd_odometry-1] [ WARN] (2025-07-22 22:24:52.952) OdometryF2M.cpp:622::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=78) between -1 and 123"
[rgbd_odometry-1] [INFO] [1753241092.953877950] [rtabmap.rgbd_odometry]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.061034s delay=1753241018.798992s
[rviz2-4] [INFO] [1753241093.354913416] [rviz2]: Message Filter dropping message: frame 'left_cam' at time 68.755 for reason 'discarding message because the queue is full'
[rgbd_odometry-1] [ WARN] (2025-07-22 22:24:53.412) OdometryF2M.cpp:622::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=65) between -1 and 124"
[rgbd_odometry-1] [INFO] [1753241093.413642811] [rtabmap.rgbd_odometry]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.066229s delay=1753241018.808755s
[rviz2-4] [INFO] [1753241093.418440119] [rviz2]: Message Filter dropping message: frame 'left_cam' at time 69.255 for reason 'discarding message because the queue is full'
[rgbd_odometry-1] [WARN] [1753241093.579762078] [rtabmap.rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rgbd_odometry-1] rgbd_odometry subscribed to (exact sync, topic_queue_size=10, sync_queue_size=10):
[rgbd_odometry-1]    /left_cam/rgb/image_raw,
[... cut ...]
[rgbd_odometry-1] [INFO] [1753241097.750916928] [rtabmap.rgbd_odometry]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.054373s delay=1753241018.796034s
```
Primafacie it looks like it's happening because of some inlier which probably is because of the loop closure module (Please correct me if I am wrong). Is it necessary to do loop closure if you have a ground truth ? What would be an ideal launch file for this one ?? All the transforms are published as `/tf` and `/tf_static` and the transform starts from `world` to `base_link_gt` where the ground truth odom `/odom` is being published to those frames.

Thank you.

matlabbe, 2025-08-03:
You are currently recomputing odometry with `rgbd_odometry` node, which appears to get lost at some point, thus no more odometry and no more update on rtabmap node.

If you want to use your own odometry, do not start `rgbd_odometry` node, just start rtabmap node and remap the `odom` topic for yours:
```
ros2 launch rtabmap_launch rtabmap.launch.py \
  rtabmap_args:="--delete_db_on_start" \
  rgb_topic:=/left_cam/rgb/image_raw \
  depth_topic:=/depth_cam/mono/image_raw \
  camera_info_topic:=/left_cam/camera_info \
  odom_topic:=/odom \
  frame_id:=base_link_gt \
  approx_sync:=false \
  rviz:=true \
  visual_odometry:=false
```

ArghyaChatterjee, 2025-08-05:
@matlabbe , thank you.