introlab/rtabmap_ros #1170: Inlier Visualization

SaiPuneeth666, 2024-06-05:
Hi,
I am using Superpoint with Superglue for StereoOdometry and rtabmap_slam nodes along with F2M strategy.
I would like to view the inliers. I found in the OdomInfo topic, that the inliers , matches and word_keys and word_values are published. 
I am using the timestamp of odominfo to find the corresponding **'LEFT' image.** 
Does the keypoints in Word_values corresponds to the keypoints in the current left images or the previous frames? I have this doubt as there is a slight misalignment of inlier keypoints and image. 
[image]

matlabbe, 2024-06-10:
They look indeed off in that picture. The odom_info topic is published with same stamp than the input image. You could make a simple python script to subscribe with an exact synchronizer to both image and odom_info, so that you can be sure they are coming from same image. 

You can doublecheck in the database with `rtabmap-databaseViewer` to see if the keypoints are correctly aligned.

SaiPuneeth666, 2024-06-10:
The above image is read using python script by matching the exact timestamp of odom_info and left_image topic. The keypoints are plotted by inlier's ids and values in Word_values. 
The keypoints in databaseViewer are aligned in the image. But the look off with the above manual reading. 
[image]

This image is from python script. 
[image]

This is from Databaseviewer at the exact timestamp

matlabbe, 2024-06-10:
Is parameter `Rtabmap/ImagesAlreadyRectified` used (set to false)? It seems that the image in the database doesn't look exactly like the top one, maybe there is some rectification going on. It seems the database is not using the same features. If you connect rtabmap node (`subscribe_rgbd=true`) to rgbd_image published by the odometry node with `Mem/UseOdomFeatures`, you should be able to see the same features used on rtabmap side, which could make it easier to compare.

SaiPuneeth666, 2024-06-12:
Yes. The images are different. Rtabmap is using rectified images and i am plotting the inliers on raw images. I tried plotting on rectified images and the keypoints seems to be aligned. 
Thanks for point out the issue.

closed 2024-06-12 by SaiPuneeth666