introlab/rtabmap_ros #1175: D435+ROS Noetic+RTABMap does not start

TDVL, 2024-06-16:
Dear Support Team,
I have installed the D435 RealSense ROS Wrapper and it works well (using .udo apt-get install ros-noetic-realsense2-camera).
Also, I have installed RTABMap using sudo apt-get install ros-noetic-rtabmap-ros and it starts. however I cannot see the mapping procedure and I have warnings.
aci@laci-Inspiron-3881:~/catkin_ws$ roslaunch rtabmap_launch rtabmap.launch \ rtabmap_args:="--delete_db_on_start" \depth_topic:=/camera/aligned_depth_to_color/image_raw \ rgb_topic:=/camera/color/image_raw \ camera_info_topic:=/camera/color/camera_inf \ approx_sync:=false
... logging to /home/laci/.ros/log/b26f8ca6-2c1a-11ef-81f6-87be2e758d6d/roslaunch-laci-Inspiron-3881-34997.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:38563/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmap_viz/

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/approx_sync_max_interval: 0.0
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/loc_thr: 0.0
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: True
[... 266 lines cut ...]
/rtabmap/octomap_grid
/rtabmap/octomap_ground
/rtabmap/octomap_obstacles
/rtabmap/octomap_occupied_space
/rtabmap/odom
/rtabmap/odom_info
/rtabmap/odom_info_lite
/rtabmap/odom_last_frame
/rtabmap/odom_local_map
/rtabmap/odom_local_scan_map
/rtabmap/odom_rgbd_image
/rtabmap/odom_sensor_data/compressed
/rtabmap/odom_sensor_data/features
/rtabmap/odom_sensor_data/raw
/rtabmap/proj_map
/rtabmap/republish_node_data
/rtabmap/scan_map
/tag_detections
/tf
/tf_static
/user_data_async

Please, what could be the problem? How can I start the mapping and save the result?
Thank you in advance.
Kind regards

matlabbe, 2024-06-17:
It seems the remapping didn't work:
```
/camera/rgb/image_rect_color \
/camera/aligned_depth_to_color/image_raw \
/camera/rgb/camera_info
```
should be:
```
/camera/color/image_raw \
/camera/aligned_depth_to_color/image_raw \
/camera/color/camera_info
```
Can you try removing the "slashes" in the command like this:
`roslaunch rtabmap_launch rtabmap.launch rtabmap_args:="--delete_db_on_start"  depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info approx_sync:=false`

TDVL, 2024-06-17:
Thank you. Now the mapping has been started. However, I still have an error. Also, where are stored these maps, since I noticed that the drive free space is rapidly decreazing? How can I delete the data from memory and how can I reduce the memory usage? In this way the hdd will be full in a few minutes!!!

[ WARN] (2024-06-17 07:22:17.440) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=11) between -1 and 2155"
[ INFO] [1718601737.441288268]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.052256s
[ WARN] (2024-06-17 07:22:17.499) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=15) between -1 and 2156"
[ INFO] [1718601737.499849708]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053314s
[ WARN] (2024-06-17 07:22:17.559) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=16) between -1 and 2157"
[ INFO] [1718601737.560560431]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.058925s
[ WARN] (2024-06-17 07:22:17.617) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=14) between -1 and 2158"
[ INFO] [1718601737.618138764]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053185s
[ WARN] [1718601737.648171641]: Could not get transform from odom to camera_link after 0.200000 seconds (for stamp=1718601736.926017)! Error="Lookup would require extrapolation 125.820122957s into the future.  Requested time 1718601736.926017046 but the latest data is at time 1718601611.105894089, when looking up transform from frame [camera_link] to frame [odom]. canTransform returned after 0.200221 timeout was 0.2.".
[ WARN] (2024-06-17 07:22:17.669) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=13) between -1 and 2159"
[ INFO] [1718601737.669939704]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.050450s
[ERROR] (2024-06-17 07:22:17.672) Rtabmap.cpp:1348::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 3908 is ignored!
[ INFO] [1718601737.672956782]: rtabmap (4): Rate=1.00s, Limit=0.000s, Conversion=0.0026s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] (2024-06-17 07:22:17.726) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 2160"
[ INFO] [1718601737.726448570]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053264s
[ WARN] (2024-06-17 07:22:17.781) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 2161"
[ INFO] [1718601737.781381990]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053204s
[ WARN] (2024-06-17 07:22:17.831) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=10) between -1 and 2162"
[ INFO] [1718601737.831914929]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.049199s
[ WARN] [1718601737.852298874]: Could not get transform from odom to camera_link after 0.200000 seconds (for stamp=1718601736.992727)! Error="Lookup would require extrapolation 125.886832475s into the future.  Requested time 1718601736.992726564 but the latest data is at time 1718601611.105894089, when looking up transform from frame [camera_link] to frame [odom]. canTransform returned after 0.201891 timeout was 0.2.".
[ WARN] [1718601737.857434445]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap_viz subscribed to (exact sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info

[image]
[ WARN] (2024-06-17 07:22:17.440) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=11) between -1 and 2155"
[ INFO] [1718601737.441288268]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.052256s
[ WARN] (2024-06-17 07:22:17.499) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=15) between -1 and 2156"
[ INFO] [1718601737.499849708]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053314s
[ WARN] (2024-06-17 07:22:17.559) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=16) between -1 and 2157"
[ INFO] [1718601737.560560431]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.058925s
[ WARN] (2024-06-17 07:22:17.617) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=14) between -1 and 2158"
[ INFO] [1718601737.618138764]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053185s
[ WARN] [1718601737.648171641]: Could not get transform from odom to camera_link after 0.200000 seconds (for stamp=1718601736.926017)! Error="Lookup would require extrapolation 125.820122957s into the future.  Requested time 1718601736.926017046 but the latest data is at time 1718601611.105894 [... cut ...]
[ WARN] [1718601737.857434445]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap_viz subscribed to (exact sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info

TDVL, 2024-06-17:
Is it possible, that some log files are filling after the rtabmap is stopped?
Now, the hdd consumption is continuous.
How can I delete these data, log files? How can I stop this process?
Thank you in advance

matlabbe, 2024-06-23:
> In this way the hdd will be full in a few minutes!!!

That should not be possible, unless your HDD is a couple of hundred of MB. Do you mean the RAM?

After killing the rtabmap nodes, there should be nothing logged from rtabmap remaining. Could it be from the camera node? When running, rtabmap would log to terminal by default (unless you redirect the log output to a file). The only thing it writes by default is the database, which is written to `~/.ros/rtabmap.db` by default. For your HDD issue, you may check if the database is increasing in size as fast as you are observing. 

To clear space taken from rtabmap, manually delete the database under `~/.ros/rtabmap.db`. If you use `rtabmap_args:="--delete_db_on_start"` with `rtabmap.launch`, the previous database will be always erased so you don't need to manually erase it.

Based on the log above, rtabmap's odometry could not track features for some reason (see [Lost Odometry section](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens)), and failed outputting a pose. Rtabmap then produce an error that odometry is null, discarding the data. The resulting database is mostly empty (only 2 MB).

Note that with D435 camera, I prefer using IR cameras instead of RGB for better and more robust feature tracking. See D435 examples [here](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping) using only IR cameras (either in stereo mode or IR-Depth mode).

TDVL, 2024-06-24:
Thank you Mathieu,
After the re-installation everything works well. I hope it will be in future too. I don't know what was the problem, but the HDD was full in a few minutes.
I have tried the IR cameras too. The result looks OK so far.
I will test it further.
Another question. If the D435 is mounted on collaborative robot. What is the recommended robot moving speed in a horizontal motion for a good mapping procedure? The max. cobot speed of 250mm/s is OK? The DetectionRate parameter is set to 30Hz. Also, maybe the camera will be mounted on a small car that will be moving horizontally on tracks.

matlabbe, 2024-06-28:
Don't set `Rtabmap/DetectionRate` to 30 (default is 1 Hz!). That rate is for map updates, not visual odometry. Unless your robot is super fast and you don't want holes in the map, generally 1 or 2 Hz is enough for the map updates. In contrast, you would want visual odometry to run as fast as possible, which would be doing by default. 

For the max speed, it depends how fast visual odometry can run. Note that if your robot arm base fixed, you should not need visual odometry at all, and just use arm kinematics to know where the camera is at any time. If the robot arm is moving on a vehicle, you may avoid very fast rotations, unless visual odometry can run fast (depending on your computation power). However, even with the most powerful computer, the speed may be limited by the maximum speed you can go without motion blur in the images.

The maximum speed would be dependent on how easy is to track features over multiple frames. For example, the [kitti dataset](https://www.cvlibs.net/datasets/kitti/) features a sequence where a car is driving over 70 Km/h with stereo images taken at 10 Hz and there is no problem for rtabmap's visual odometry. In contrast, in [euroc dataset](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets#the_euroc_mav_dataset) there is a sequence with very aggressive motion (drone flying indoor) that makes rtabmap lose tracking even if images are taken at 20 Hz (by lack of shared features between consecutive frames, and it is kinda blurry on fast rotation).

TDVL, 2024-06-28:
OK, thank you for the suggestions. I will use the default 1Hz DetectionRate.
The robot will moving on a car on tracks, and it will mapping the parking lot with cars.
Now, I am using 5 FPS with D435, should I use the default 30 FPS for D435?

These are my launch parameters for D435 and RTABMp:
roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 depth_fps:=5 color_width:=1280 color_height:=720 color_fps:=5 clip_distance:=6 align_depth:=true filters:=pointcloud initial_reset:=true

roslaunch rtabmap_launch rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info approx_sync:=false/localization:=true/ DetectionRate:=1

Maybe you can suggest other launch settings? 
Also I have used the clip_distance:=2 to limit the depth range estimation in depth map since the depth map and the poindcloud quality decrease with higher clip distance.
Also, you have suggested the infra cameras. Are they useful for parking lot mapping (with parked cars)?

matlabbe, 2024-06-30:
You could use smaller resolution images with higher framerate. Don't use `clip_distance` at the camera level because it can affect badly the visual odometry, you can clip the distance in the UI during visualization (preferences->3D Rendering->Max depth).

I do recommend usage of IR cameras with D435 because the RGB camera has smaller field-of-view and a rolling shutter. Disable ir-emitter and subscribe to Left IR + Depth. 

Based on http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping:
```bash
roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true

rosrun rqt_reconfigure rqt_reconfigure
# select Camera->stereo_module, set emitter_enabled to OFF

roslaunch rtabmap_launch rtabmap.launch \
   rtabmap_args:="--delete_db_on_start" \
   rgb_topic:=/camera/infra1/image_rect_raw \
   depth_topic:=/camera/depth/image_rect_raw \
   camera_info_topic:=/camera/infra1/camera_info \
   approx_sync:=false
```

TDVL, 2024-07-01:
Thank you, OK, I will try it.
Also, the "rosrun rqt_reconfigure rqt_reconfigure", # select Camera->stereo_module, set emitter_enabled to OFF option is OK, but how can I directly disable the IR emitter in Terminal command line, with roslaunch command?

Further, the smaller resolution is OK, but this can reduce the field of view and the pointcloud quality, am I right? What could be the suggested resolution?

I have used the clip_distance to remove the background noise and data, since only the first 1-1.5m meters of the mapped cars are significant to detect the free and occupied zones of the parking lot, since the D435 is a low-cost camera and with the higher distance the pointcloud quality is decreasing.

Also, the D435 preset is the Deafult, since I have noticed that this preset gives me the best map under various illumination conditions. I think this is OK.
Or, is there an option to avoid this quality decreasement?
Thank you in advance.

matlabbe, 2024-07-01:
I think I've shown the rqt_reconfigure approach because that parameter was not exposed outside of `rs_camera.launch`. You could modify `rs_camera.launch` to add that parameter as argument, or another way would be to set the parameter with `rosparam set ...` before launching realsense.

If it is to reduce noise in the occupancy grid created, use `Grid/RangeMax` to 1-1.5 meters. Clipping the raw input will make visual odometry not working well.

I cannot really comment on other D435 presets as I didn't tested the myself. That would be a question for realsense directly.

It is okay only if you already created a map of the area and wanted to do only localization in it. If you want to create the map, remove this.

TDVL, 2024-07-01:
Thank you for the suggestions.
The Grid/RangeMax can be set in Terminal command line?
Hence, the localization option can be excluded? The map will be used later for bounding box determination of cars and free spaces using a clustering algorithm from PCL library. Thus, the localization should be removed? I thought that it can help during the mapping process.
Also, I have found the Grid/MaxObstacleHeight and Grid/NoiseFiltering options. Can I set them in Terminal for fine tuning? Further, can you suggest some parameter settings in general for better mapping?

matlabbe, 2024-07-02:
"localization" means only localizing in a pre-built map. The map is not updated. When localization is false, the robot does SLAM. 

```
roslaunch rtabmap_launch rtabmap.launch \
   rtabmap_args:="--delete_db_on_start --Grid/RangeMax 1" \
   rgb_topic:=/camera/infra1/image_rect_raw \
   depth_topic:=/camera/depth/image_rect_raw \
   camera_info_topic:=/camera/infra1/camera_info \
   approx_sync:=false
```

You can add the other RTAB-Map's parameters the same way. For help for better mapping, I'll need to see that environment look like, what sensors can actually see, the noise, ... If you generate a database, you can share it, it is the easiest way to look at your config/data.

TDVL, 2024-07-02:
OK, thank you.
Here is one of the maps in *.pcd. As you can see, the first 2-2,5m of the pointcloud is OK and very visible (and these parts of the car can be easily extracted from the background) , however with the increasing distance the pointcloud quality decreases. This is expected, since the pointcloud quality with the depth distance is decreasing when RealSense cameras are used. The mapping was done from the back (blue car) and from the front (red car) side of cars as you can see, about 80cm far from the cars.
The mapping was performed outdoors. Naturally, in indoor environment with the a controlled illumination the pointcloud should be better, however the mapping was done in late afternoon, when the sunrays did not affect the mapping, With the sunrays, shadow, etc. the pointcloud quality could be only worse. 
[test8.zip](https://github.com/user-attachments/files/16064766/test8.zip)

matlabbe, 2024-07-02:
How did you generate the point cloud? If you use `rtabmap-export --max_range 1.5 my_map.db`, it will generate the point cloud only up to 1.5 meter range.

TDVL, 2024-07-02:
I have used the rtabmap launch command, the export was done using the export option in RTABMap viewer.
I have used clip_distance:=6 in this example with D435.

matlabbe, 2024-07-02:
In RTABMap viewer, you can set a max depth to export. I would still not use `clip_distance:=6`.

[image]