introlab/rtabmap_ros #1177: a small question to mapping

Boris-Cui, 2024-06-19:
Hello, 
thanks for your patience in answering our questions. I have a simple question about mapping. i use realsense d435i and ros2.  and i use stereo topics(infra1 and infra2) and imu topic to complete the odometry part, and use rgb and depth topic to complete the mapping part. i want to know if the combination has a potential impact to the created map. and what should the odometry display look like in GUI  interface in the combination?   the attached file is my launch file: if it has something wrong in my launch file, Please point out my mistakes. thanks!
[1.txt](https://github.com/user-attachments/files/15905762/1.txt)

matlabbe, 2024-06-23:
At first view it looks good. Make sure the D435i camera's IR emitter is OFF or the odometry won't work correctly. With `approx_sync` to false, I think it should be okay with D435 camera because all image topics are stamped with exactly the same stamp. It should not then be a problem to receive odom topic from stereo_odometry while rtabmap and rtabmap_viz nodes subscribed to RGB-D data.

For the impacts, loop closure detection will be done using RGB-D data, so loop closures may not be as accurate and may suffer from motion blur (that could affect the number of loop closures detected when moving the camera). On rtabmap_viz, you would see the RGB-D point cloud instead of stereo data used by odometry. If you want to see/debug visual odometry info (with `subscribe_odom_info` to `true` for rtabmap_viz node), set stereo input topics for rtabmap_viz so that `odom_info` topic matches with the stereo data to be shown. The map point cloud will still be created with RGB-D published by rtabmap node.

Boris-Cui, 2024-06-24:
thanks for your reply! I have learned a lot. I noticed that you mentioned the issue of timestamps. The reason I use different topics for the odometry and mapping parts is that I want to perform some semantic segmentation on the RGB and depth images. If the rgb and depth image  undergoes semantic segmentation, there should be some delays in mapping, but the odometry part will not undergo any processing. Do you think this is feasible? Could there be any misalignment between the local map and the pose?
Best Regards,
Boris

matlabbe, 2024-06-28:
Make sure the RGB and depth images going through semantic segmentation still keep the same stamps, then for rtabmap you may need to increase the `queue_size` so that odometry topic (published through stereo data without latency) can be synchronized with RGB-D with same stamp (but having a delay). There should not have misalignment then.

Boris-Cui, 2024-07-08:
Thank you for your reply. I would like to ask another question: Does RTAB-Map currently support generating 3D occupancy maps? If so, are there any relevant examples?

Best regards,
Boris

matlabbe, 2024-07-21:
Yes.

Set `Grid/3D=true`.

To filter dynamic objects, set `Grid/RayTracing=true`.

Related discussions: 
* http://official-rtab-map-forum.206.s1.nabble.com/How-does-rtabmap-deal-with-dynamic-objects-tp4735p8150.html
* http://official-rtab-map-forum.206.s1.nabble.com/How-to-avoid-detecting-obstacles-passing-in-front-of-a-running-robot-td10040.html#a10045

cheers,
Mathieu

Boris-Cui, 2024-07-21:
thank you very much. i don't have question anymore.
Thank you again!

Best regards
Boris

closed 2024-07-21 by matlabbe