introlab/rtabmap_ros #1178: rtabmap not subscribe to odom

Bouchrikator, 2024-06-22:
so i used the same code a kinect here [http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot](url) Kinect + Odometry i tried also Kinect + Odometry + Fake 2D laser from Kinect but it didint work too here is my launch file :
`<launch>
<arg name="approx_sync" value="false"/> 
<arg name="publish_tf_odom" value="false"/> 
<arg name="subscribe_odom_info" value="false"/> 
  <group ns="rtabmap">
 
    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_sync/rgbd_sync" output="screen">
      <remap from="rgb/image"       to="/camera/rgb/image_rect_color"/>
      <remap from="depth/image"     to="/camera/depth_registered/image_raw"/>
      <remap from="rgb/camera_info" to="/camera/rgb/camera_info"/>
      <remap from="rgbd_image"      to="rgbd_image"/> 
      
      
      <param name="approx_sync"       value="true"/> 
    </node>

    <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="--delete_db_on_start">
          <param name="frame_id" type="string" value="base_link"/>

          <param name="subscribe_depth" type="bool" value="false"/>
          <param name="subscribe_rgbd" type="bool" value="true"/>

          <remap from="odom" to="/odom"/>

          <remap from="rgbd_image" to="rgbd_image"/>

          <param name="queue_size" type="int" value="10"/>

          
          <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
          <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
    </node>
  </group>
</launch>`
[image]
[image]
here is also the rqt_graph
[image]
i'm suck i have a deadline

Bouchrikator, 2024-06-23:
UPDATE: i tried this when and used that launch file i linked the camera_link with the base_link and here what happend:

[image]

[image]

i used this launch file:

`<launch>
  <group ns="rtabmap">

    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_sync/rgbd_sync" output="screen">
      <remap from="rgb/image"        to="/camera/rgb/image_rect_color"/>
      <remap from="depth/image"      to="/camera/depth_registered/image_raw"/>
      <remap from="rgb/camera_info"  to="/camera/rgb/camera_info"/>
      <remap from="rgbd_image"       to="rgbd_image"/> 
      
      
      <param name="approx_sync"       value="true"/> 
    </node>

    
    <node pkg="rtabmap_odom" type="rgbd_odometry" name="rgbd_odometry" output="screen">
      <param name="subscribe_rgbd" type="bool"   value="true"/>
      <param name="frame_id"       type="string" value="base_link"/>
      <remap from="rgbd_image" to="rgbd_image"/>
    </node>

    <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="--delete_db_on_start">
          <param name="frame_id"        type="string" value="base_link"/>
          <param name="subscribe_depth" type="bool"   value="false"/>
          <param name="subscribe_rgbd"  type="bool"   value="true"/>

          <remap from="odom"       to="odom"/>
          <remap from="rgbd_image" to="rgbd_image"/>

          <param name="queue_size"  type="int"  value="10"/>
          <param name="approx_sync" type="bool" value="false"/>

          
          <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
          <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
    </node>
    <node pkg="rtabmap_viz" type="rtabmap_viz" name="rtabmap_viz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
        <param name="subscribe_depth"             type="bool" value="true"/>
        <param name="subscribe_scan"              type="bool" value="true"/>
        <param name="frame_id"                    type="string" value="base_link"/>

      <remap from="rgb/image"        to="/camera/rgb/image_rect_color"/>
      <remap from="depth/image"      to="/camera/depth_registered/image_raw"/>
      <remap from="rgb/camera_info"  to="/camera/rgb/camera_info"/>
        
    </node>
  </group>
</launch>`

matlabbe, 2024-06-24:
You may have two odometry nodes (rgbd_odometry and your wheel odometry) publishing on same TF. Add this to rgbd_odometry node (assuming your wheel odometry is publishing TF odom->base_link) to avoid TF issues:
```xml
<param name="odom_frame_id"       type="string" value="vo"/>
<param name="guess_frame_id"       type="string" value="odom"/>
```
You may also try without rgbd_odometry too, and just remap your `/odom` topic in rtabmap node.