introlab/rtabmap_ros #1181: ROS2: cannot change image_transport plugin

matlabbe, 2024-07-05:
Under ros1, we could do:
```
rosrun rtabmap_odom rgbd_odometry _rgb/image_transport:=compressed _depth/image_transport:=compressedDepth
...
/rgbd_odometry subscribed to (approx sync):
   /rgb/image/compressed \
   /depth/image/compressedDepth \
   /rgb/camera_info
...
```
and the node will subscribe to input topics with corresponding `image_transport` suffix.

Similar commands under ROS2 don't work:
```
ros2 run rtabmap_odom rgbd_odometry --ros-args -p rgb/image_transport:=compressed -p depth/image_transport:=compressedDepth
...
rgbd_odometry subscribed to (approx sync):
   /rgb/image,
   /depth/image,
   /rgb/camera_info
...
```
```
ros2 run rtabmap_odom rgbd_odometry --ros-args -p image_transport:=compressed
...
rgbd_odometry subscribed to (approx sync):
   /rgb/image,
   /depth/image,
   /rgb/camera_info
...
```

Current workaround is to use `image_transport`'s `republish` node:
```
ros2 run image_transport republish compressed raw --ros-args -r in/compressed:=/rgb/image/compressed -r out:=/rgb/image
ros2 run image_transport republish compressedDepth raw --ros-args -r in/compressedDepth:=/depth/image/compressed -r out:=/depth/image
ros2 run rtabmap_odom rgbd_odometry 
...
rgbd_odometry subscribed to (approx sync):
   /rgb/image,
   /depth/image,
   /rgb/camera_info
...
```

matlabbe, 2025-03-16:
With https://github.com/introlab/rtabmap_ros/pull/1282, we can now do:
```
ros2 run rtabmap_odom rgbd_odometry --ros-args \
   -p rgb_transport:=compressed \
   -p depth_transport:=compressedDepth \
   -r rgb/image:=camera/rgb/image_rect \
   -r depth/image:=camera/depth/image_raw \
   -r rgb/camera_info:=camera/rgb/camera_info

....
rgbd_odometry subscribed to (approx sync, topic_queue_size=10, sync_queue_size=5):
   /camera/rgb/image_rect/compressed,
   /camera/depth/image_raw/compressedDepth,
   /camera/rgb/camera_info
```

Alessio-Parmeggiani, 2025-08-08:
Hello, can you confirm that this feature is not available in the rtabmap node?