introlab/rtabmap_ros #1182: Did not receive data since 5 seconds!

whaudrms, 2024-07-08:
Hello, I am having trouble running rtabmap
I'm usuing oak-d-pro and livox avia and I'm trying to use 'subscribe_scan_cloud' parameter but the following error appears.

```

SUMMARY
========

PARAMETERS
 * /cmdline_file_path: livox_test.lvx
 * /cmdline_str: 100000000000000
 * /common/imu_topic: /livox/imu
 * /common/lid_topic: /livox/lidar
 * /common/time_offset_lidar_to_imu: 0.0
 * /common/time_sync_en: False
 * /cube_side_length: 1000.0
 * /data_src: 0
 * /enable_imu_bag: True
 * /enable_lidar_bag: True
 * /feature_extract_enable: False
 * /filter_size_map: 0.5
 * /filter_size_surf: 0.5
 * /frame_id: livox_frame
 * /livox_repub_online_node/livox_points_message_name: /livox/points
 * /livox_repub_online_node/livox_raw_message_name: /livox/lidar
 * /mapping/acc_cov: 0.1
 * /mapping/b_acc_cov: 0.0001
 * /mapping/b_gyr_cov: 0.0001
 * /mapping/det_range: 450.0
 * /mapping/extrinsic_R: [1, 0, 0, 0, 1, 0...
 * /mapping/extrinsic_T: [0.04165, 0.02326...
 * /mapping/extrinsic_est_en: False
 * /mapping/fov_degree: 90
 * /mapping/gyr_cov: 0.1
 * /max_iteration: 3
 * /multi_topic: 0
 * /oak/camera_i_base_frame: oak-d_frame
 * /oak/camera_i_cam_pitch: 0.0
 * /oak/camera_i_cam_pos_x: 0.0
 * /oak/camera_i_cam_pos_y: 0.0
 * /oak/camera_i_cam_pos_z: 0.0
 * /oak/camera_i_cam_roll: 0.0
 * /oak/camera_i_cam_yaw: 0.0
 * /oak/camera_i_imu_from_descr: False
 * /oak/camera_i_nn_type: none
 * /oak/camera_i_parent_frame: oak-d-base-frame
 * /oak/camera_i_publish_tf_from_calibration: False
 * /oak/robot_description: <?xml version="1....
 * /oak/stereo_i_subpixel: True
 * /output_data_type: 0
 * /pcd_save/interval: -1
 * /pcd_save/pcd_save_en: True
 * /point_filter_num: 3
 * /preprocess/blind: 4
 * /preprocess/lidar_type: 1
 * /preprocess/scan_line: 6
 * /publish/dense_publish_en: True
 * /publish/path_en: False
 * /publish/scan_bodyframe_pub_en: True
 * /publish/scan_publish_en: True
 * /publish_freq: 10.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /rtabmap/Grid/RangeMax: 0
 * /rtabmap/Icp/CorrespondenceRatio: 0.2
 * /rtabmap/Icp/RangeMax: 0
 * /rtabmap/Icp/VoxelSize: 0.05
 * /rtabmap/RGBD/ProximityBySpace: true
 * /rtabmap/Reg/Force3DoF: true
 * /rtabmap/Reg/Strategy: 1
[... 206 lines cut ...]
        <param name="subscribe_rgbd"  type="bool" value="false"/>
        <param name="subscribe_scan_cloud"  type="bool" value="true"/>

        <remap from="scan_cloud" to="/livox/points"/>
        <remap from="rgb/image" to="oak/rgb/image_raw"/>
        <remap from="rgb/camera_info" to="oak/rgb/camera_info"/>
        <remap from="odom" to="/Odometry"/>
        <param name="subscribe_odom_info" type="bool" value="true"/>

        <!-- RTAB-Map's parameters -->
        <param name="Reg/Strategy"       type="string" value="1"/>    <!-- 0=Visual, 1=ICP, 2=Visual+ICP -->
        <param name="Reg/Force3DoF"      type="string" value="true"/>
        <param name="RGBD/ProximityBySpace"    type="string" value="true"/>
        <param name="Icp/CorrespondenceRatio"  type="string" value="0.2"/>
        <param name="Icp/VoxelSize"            type="string" value="0.05"/>
        <param name="Icp/RangeMax"             type="string" value="0"/>
        <param name="Grid/RangeMax"            type="string" value="0"/>
    </node>

</launch>

```

Any help on this is greatly appreciated.
thank you :)

matlabbe, 2024-07-21:
Not sure why rtabmap switched from rgbd_image to rgb image input between these two logs:
```
[ WARN] [1720443486.890467063]: /rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap subscribed to (approx sync):
   /Odometry \
   /rgbd/image \
   /rgb/camera_info \
   /livox/points
[ INFO] [1720443486.944870637]: Camera ready!
[ WARN] [1720443491.890569975]: /rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap subscribed to (approx sync):
   /Odometry \
   /oak/rgb/image_raw \
   /rgb/camera_info \
   /livox/points
```

The camera info is not remapped correctly too. I tried rtabmap part of your launch and this is what should be expected:
```
/rtabmap subscribed to (approx sync):
   /Odometry \
   /oak/rgb/image_raw \
   /oak/rgb/camera_info \
   /livox/points \
   /odom_info
[ INFO] [1721581756.110587353]: rtabmap 0.21.5 started...
[ WARN] [1721581760.046839697]: /rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap subscribed to (approx sync):
   /Odometry \
   /oak/rgb/image_raw \
   /oak/rgb/camera_info \
   /livox/points \
   /odom_info
```

You would also need to set `subscribe_odom_info` to false instead of true, as the topic doesn't exist in your case. Do:
```
    <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="--delete_db_on_start">
        <param name="frame_id" type="string" value="oak"/>

        <param name="subscribe_rgb"   type="bool" value="true"/>
        <param name="subscribe_depth" type="bool" value="false"/>
        <param name="subscribe_rgbd"  type="bool" value="false"/>
        <param name="subscribe_scan_cloud"  type="bool" value="true"/>

        <remap from="scan_cloud" to="/livox/points"/>
        <remap from="rgb/image" to="oak/rgb/image_raw"/>
        <remap from="rgb/camera_info" to="oak/rgb/camera_info"/>
        <remap from="odom" to="/Odometry"/>
        <param name="subscribe_odom_info" type="bool" value="false"/>

        <!-- RTAB-Map's parameters -->
        <param name="Reg/Strategy"       type="string" value="1"/>    <!-- 0=Visual, 1=ICP, 2=Visual+ICP -->
        <param name="Reg/Force3DoF"      type="string" value="true"/>
        <param name="RGBD/ProximityBySpace"    type="string" value="true"/>
        <param name="Icp/CorrespondenceRatio"  type="string" value="0.2"/>
        <param name="Icp/VoxelSize"            type="string" value="0.05"/>
        <param name="Icp/RangeMax"             type="string" value="0"/>
        <param name="Grid/RangeMax"            type="string" value="0"/>
    </node>
```

matlabbe, 2024-07-21:
To get loop closure detection working, I suggest to use depth image of the oak as it is available. To do so, you will need to use RGBDImage interface. Following this [example](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_Odometry_.2B-_2D_laser), that would look like this:
```xml
    <node pkg="rtabmap_sync" type="rgbd_sync" name="rgbd_sync" output="screen">
      <remap from="rgb/image"         to="oak/rgb/image_raw"/>
      <remap from="depth/image"       to="oak/depth/image"/>
      <remap from="rgb/camera_info"   to="oak/rgb/camera_info"/>
    </node>

    <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="--delete_db_on_start">
        <param name="frame_id" type="string" value="oak"/>

        <param name="subscribe_rgb"   type="bool" value="false"/>
        <param name="subscribe_depth" type="bool" value="false"/>
        <param name="subscribe_rgbd"  type="bool" value="true"/>
        <param name="subscribe_scan_cloud"  type="bool" value="true"/>

        <remap from="scan_cloud" to="/livox/points"/>
        <remap from="odom" to="/Odometry"/>
        <param name="subscribe_odom_info" type="bool" value="false"/>

        <!-- RTAB-Map's parameters -->
        <param name="Reg/Strategy"       type="string" value="1"/>    <!-- 0=Visual, 1=ICP, 2=Visual+ICP -->
        <param name="Reg/Force3DoF"      type="string" value="true"/>
        <param name="RGBD/ProximityBySpace"    type="string" value="true"/>
        <param name="Icp/CorrespondenceRatio"  type="string" value="0.2"/>
        <param name="Icp/VoxelSize"            type="string" value="0.05"/>
        <param name="Icp/RangeMax"             type="string" value="0"/>
        <param name="Grid/RangeMax"            type="string" value="0"/>
    </node>
```
You may have to adjust the remap of the oak depth image, I just guessed the name.