introlab/rtabmap_ros #1185: RTABMAP shows black screen in 3D map and the visualizer frames are very laggy.

Shivam7Sharma, 2024-07-18:
Hi,

I am using a custom stereo setup and driver for two OAK FFC OV9282 M12 cameras with an OAK FFC 3P. I have calibrated the camera using pinhole functions of OpenCV, and I have received reprojection error of 0.17.  The problem I am facing is that I get black screen in the 3D map of RTABMAP visualizer with the following messages in the terminal:

```
[rtabmap-2] [WARN] [1721257512.432158353] [rtabmap.rtabmap]: rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 1 respectively).
[rtabmap-2] rtabmap subscribed to (approx sync):
[rtabmap-2]    /rtabmap/odom \
[rtabmap-2]    /synced/left/image_rect \
[rtabmap-2]    /synced/right/image_rect \
[rtabmap-2]    /left/camera_info \
[rtabmap-2]    /right/camera_info \
[rtabmap-2]    /rtabmap/odom_info
[stereo_odometry-1] [INFO] [1721257512.570976086] [rtabmap.stereo_odometry]: Odom: quality=380, std dev=0.001542m|0.008712rad, update time=0.271491s
[stereo_odometry-1] [WARN] [1721257512.954000463] [rtabmap.stereo_odometry]: The time difference between left and right frames is high (diff=0.063504s, left=1721257512.375110s, right=1721257512.438614s). If your left and right cameras are hardware synchronized, use approx_sync:=false. Otherwise, you may want to set approx_sync_max_interval lower than 0.01s to reject spurious bad synchronizations.
[stereo_odometry-1] [INFO] [1721257513.272862162] [rtabmap.stereo_odometry]: Odom: quality=351, std dev=0.001901m|0.008712rad, update time=0.316917s

```

My launch command is :

```
ros2 launch rtabmap_launch rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start --Grid/RayTracing true --Vis/EstimateOdom true --Grid/3D false --subscribe_odom_info false --queue_size 30 --approx_sync_max_interval 0.02" \
    stereo:=true \
    frame_id:=base_link \
    wait_imu_to_init:=false \
    left_image_topic:=/synced/left/image_rect \
    right_image_topic:=/synced/right/image_rect \
    left_camera_info_topic:=/left/camera_info \
    right_camera_info_topic:=/right/camera_info \
    approx_sync:=true
```

Why is RTABMAP not mapping with the stereo setup I have? I will upload a link to the ROS bag in a few hours so that someone can help me debug.

matlabbe, 2024-07-21:
Testing visual odometry against a white wall is not the best. In this case VO is loosing tracking and get lost, causing a red screen: https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens

[image]

The major problem is that it seems the left and right images are not synchronized or stereo calibration is poor. 

Here is an example where there is a vertical disparity between left and right images:
[image]

Here is an example where during motion between two consecutive stereo frames the horizontal disparity varies (meaning the left and right cameras are not synchronized):

[image]

Are the two OAK FFC OV9282 M12 cameras trigger a picture with same hardware signal?

Shivam7Sharma, 2024-07-22:
> Testing visual odometry against a white wall is not the best. In this case VO is loosing tracking and get lost, causing a red screen: [...]

Thank you for your reply!

I saw that you were getting some points, but I was not getting any points. I am trying to synchronize the left and right frames using software. Is frame dropping also an issue with Rtabmap? I think my calibration is good because the disparity and depth was good. Once I synchronize left and right I will give an update on performance.

Shivam7Sharma, 2024-07-24:
> Testing visual odometry against a white wall is not the best. In this case VO is loosing tracking and get lost, causing a red screen: [...]

Hi,

I added synchronization logic to my Ros 2 driver and edited the QoS settings since that was causing some synchronization issues. I have attached the terminal output and a new robag link with this reply. Can you please help me understand why you were getting points on the first rosbag I sent and I was getting a black screen? Also, I have set the maximum time difference to 0.040 seconds, but still, RTABMAP complained about the 0.031 second time difference. Is this time difference not good enough?
I still get a black screen. The terminal output and link to ROSbag are attached below.

My command to run RTABMAP was :

`ros2 launch rtabmap_launch rtabmap.launch.py     rtabmap_args:="--delete_db_on_start --Vis/EstimateOdom true --subscribe_odom_info false --queue_size 30"     stereo:=true     frame_id:=base_link     wait_imu_to_init:=false     left_image_topic:=/left/image_rect     right_image_topic:=/right/image_rect     left_camera_info_topic:=/left/camera_info     right_camera_info_topic:=/right/camera_info approx_sync_max_interval:=0.04     approx_sync:=true`

[rtabmap_debug7.txt](https://github.com/user-attachments/files/16365699/rtabmap_debug7.txt)

[Link to google drive ROS Bag](https://drive.google.com/drive/folders/1QjydyoSl35IVpzN43iwhRllp2IkZDAU2?usp=sharing)

matlabbe, 2024-07-27:
It seems to work:
```
[stereo_odometry-1] [INFO] [1722114647.571042491] [rtabmap.stereo_odometry]: Odom: quality=326, std dev=0.001560m|0.000173rad, update time=0.034837s
[stereo_odometry-1] [INFO] [1722114647.612182747] [rtabmap.stereo_odometry]: Odom: quality=293, std dev=0.001731m|0.008712rad, update time=0.040051s
[stereo_odometry-1] [INFO] [1722114647.649321451] [rtabmap.stereo_odometry]: Odom: quality=316, std dev=0.001202m|0.000173rad, update time=0.034762s
[stereo_odometry-1] [INFO] [1722114647.684650695] [rtabmap.stereo_odometry]: Odom: quality=322, std dev=0.001380m|0.000173rad, update time=0.034368s
[stereo_odometry-1] [INFO] [1722114647.723724790] [rtabmap.stereo_odometry]: Odom: quality=334, std dev=0.001277m|0.008712rad, update time=0.037868s
[stereo_odometry-1] [INFO] [1722114647.760536512] [rtabmap.stereo_odometry]: Odom: quality=303, std dev=0.001547m|0.008712rad, update time=0.035568s
[stereo_odometry-1] [INFO] [1722114647.798730269] [rtabmap.stereo_odometry]: Odom: quality=324, std dev=0.001457m|0.008712rad, update time=0.036987s
[rtabmap-2] [INFO] [1722114647.820579185] [rtabmap.rtabmap]: rtabmap (6): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0574s, Maps update=0.0001s pub=0.0011s (local map=1, WM=1)
[stereo_odometry-1] [INFO] [1722114647.838038385] [rtabmap.stereo_odometry]: Odom: quality=303, std dev=0.001343m|0.000173rad, update time=0.038168s
[stereo_odometry-1] [INFO] [1722114647.874417770] [rtabmap.stereo_odometry]: Odom: quality=321, std dev=0.001341m|0.000173rad, update time=0.035408s
[stereo_odometry-1] [INFO] [1722114647.911916739] [rtabmap.stereo_odometry]: Odom: quality=309, std dev=0.001532m|0.000173rad, update time=0.034522s
[stereo_odometry-1] [INFO] [1722114647.951691461] [rtabmap.stereo_odometry]: Odom: quality=316, std dev=0.001110m|0.000173rad, update time=0.035869s
[stereo_odometry-1] [INFO] [1722114647.986655126] [rtabmap.stereo_odometry]: Odom: quality=317, std dev=0.001538m|0.010360rad, update time=0.034051s
[stereo_odometry-1] [INFO] [1722114648.024644190] [rtabmap.stereo_odometry]: Odom: quality=320, std dev=0.001351m|0.008712rad, update time=0.034484s
[stereo_odometry-1] [INFO] [1722114648.061335622] [rtabmap.stereo_odometry]: Odom: quality=308, std dev=0.001658m|0.011465rad, update time=0.035638s
[stereo_odometry-1] [INFO] [1722114648.097434390] [rtabmap.stereo_odometry]: Odom: quality=314, std dev=0.001364m|0.000173rad, update time=0.035141s
[stereo_odometry-1] [INFO] [1722114648.133750464] [rtabmap.stereo_odometry]: Odom: quality=333, std dev=0.001238m|0.000173rad, update time=0.035355s
[stereo_odometry-1] [INFO] [1722114648.175154550] [rtabmap.stereo_odometry]: Odom: quality=313, std dev=0.001399m|0.000173rad, update time=0.040162s
[stereo_odometry-1] [INFO] [1722114648.215593757] [rtabmap.stereo_odometry]: Odom: quality=306, std dev=0.001802m|0.000173rad, update time=0.034935s
```
[image]

Without moving, the stereo calibration looks fine. If you didn't change anything from the previous bag, then the time sync is causing a problem when moving. You cannot use rtabmap in stereo mode if the left and right camera shutters are not triggered at the same time (same hardware signal). Normally `approx_sync` should be false with stereo data, but we accept it when the camera driver cannot stamp with exact same time both images as long as they were still taken at the same time by hardware (e.g., depthai driver). Assuming the conditions above are true, normally the delay between left and right should not be more than 1 ms. This is what we do with depthai example [here](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping): `approx_sync_max_interval:=0.001`.

For the "Did not receive data since 5 seconds!" warning, is it when replaying the bag? You should launch `rtabmap.launch.py` with `use_sim_time:=true` and play the bag with `--clock` option. It is sure if you have sync issue and rtabmap node didn't publish for 5 sec, `rtabmap_viz` will then complain that it didn't receive data for past 5 sec.

cheers,
Mathieu

Shivam7Sharma, 2024-07-28:
> It seems to work:
> 
> ```
> [stereo_odometry-1] [INFO] [1722114647.571042491] [rtabmap.stereo_odometry]: Odom: quality=326, std dev=0.001560m|0.000173rad, update time=0.034837s
> [stereo_odometry-1] [INFO] [1722114647.612182747] [rtabmap.stereo_odometry]: Odom: quality=293, std dev=0.001731m|0.008712rad, update time=0.040051s
> [stereo_odometry-1] [INFO] [1722114647.649321451] [rtabmap.stereo_odometry]: Odom: quality=316, std dev=0.001202m|0.000173rad, update time=0.034762s
> [stereo_odometry-1] [INFO] [1722114647.684650695] [rtabmap.stereo_odometry]: Odom: quality=322, std dev=0.001380m|0.000173rad, update time=0.034368s
> [stereo_odometry-1] [INFO] [1722114647.723724790] [rtabmap.stereo_odometry]: Odom: quality=334, std dev=0.001277m|0.008712rad, update time=0.037868s
> [stereo_odometry-1] [INFO] [1722114647.760536512] [rtabmap.stereo_odometry]: Odom: quality=303, std dev=0.001547m|0.008712rad, update time=0.035568s
> [stereo_odometry-1] [INFO] [1722114647.798730269] [rtabmap.stereo_odometry]: Odom: quality=324, std dev=0.001457m|0.008712rad, update time=0.036987s
> [rtabmap-2] [INFO] [1722114647.820579185] [rtabmap.rtabmap]: rtabmap (6): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0574s, Maps update=0.0001s pub=0.0011s (local map=1, WM=1)
> [stereo_odometry-1] [INFO] [1722114647.838038385] [rtabmap.stereo_odometry]: Odom: quality=303, std dev=0.001343m|0.000173rad, update time=0.038168s
> [stereo_odometry-1] [INFO] [1722114647.874417770] [rtabmap.stereo_odometry]: Odom: quality=321, std dev=0.001341m|0.000173rad, update time=0.035408s
> [stereo_odometry-1] [INFO] [1722114647.911916739] [rtabmap.stereo_odometry]: Odom: quality=309, std dev=0.001532m|0.000173rad, update time=0.034522s
> [stereo_odometry-1] [INFO] [1722114647.951691461] [rtabmap.stereo_odometry]: Odom: quality=316, std dev=0.001110m|0.000173rad, update time=0.035869s
> [stereo_odometry-1] [INFO] [1722114647.986655126] [rtabmap.stereo_odometry]: Odom: quality=317, std dev=0.001538m|0.010360rad, update time=0.034051s
> [stereo_odometry-1] [INFO] [1722114648.024644190] [rtabmap.stereo_odometry]: Odom: quality=320, std dev=0.001351m|0.008712rad, update time=0.034484s
> [stereo_odometry-1] [INFO] [1722114648.061335622] [rtabmap.stereo_odometry]: Odom: quality=308, std dev=0.001658m|0.011465rad, update time=0.035638s
> [stereo_odometry-1] [INFO] [1722114648.097434390] [rtabmap.stereo_odometry]: Odom: quality=314, std dev=0.001364m|0.000173rad, update time=0.035141s
> [stereo_odometry-1] [INFO] [1722114648.133750464] [rtabmap.stereo_odometry]: Odom: quality=333, std dev=0.001238m|0.000173rad, update time=0.035355s
> [stereo_odometry-1] [INFO] [1722114648.175154550] [rtabmap.stereo_odometry]: Odom: quality=313, std dev=0.001399m|0.000173rad, update time=0.040162s
> [stereo_odometry-1] [INFO] [1722114648.215593757] [rtabmap.stereo_odometry]: Odom: quality=306, std dev=0.001802m|0.000173rad, update time=0.034935s
> ```
> 
> ![Screenshot from 2024-07-27 14-08-55](https://private-user-images.githubusercontent.com/2319645/352777455-19227368-1830-4afd-934b-ff732ab4051f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIxNzc2NTMsIm5iZiI6MTcyMjE3NzM1MywicGF0aCI6Ii8yMzE5NjQ1LzM1Mjc3NzQ1NS0xOTIyNzM2OC0xODMwLTRhZmQtOTM0Yi1mZjczMmFiNDA1MWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjhUMTQzNTUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDUyZjFjY2RhNGFkYzgwNTkzYjYyNmQ1YTYwMDEwMTBmZDkzOTFkNjMyZDM5ZmI0NDExNTg5YjAyOTI1NjA0MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Hwx4PdE5RU_9a10AYBa2qB4ovdI0-E-Axg0irHTQdgM)
> 
> Without moving, the stereo calibration looks fine. If you didn't change anything from the previous bag, then the time sync is causing a problem when moving. You cannot use rtabmap in stereo mode if the left and right camera shutters are not triggered at the same time (same hardware signal). Normally `approx_sync` should be false with stereo data, but we accept it when the camera driver cannot stamp with exact same time both images as long as they were still taken at the same time by hardware (e.g., depthai driver). Assuming the conditions above are true, normally the delay between left and right should not be more than 1 ms. This is what we do with depthai example [here](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping): `approx_sync_max_interval:=0.001`.
> 
> For the "Did not receive data since 5 seconds!" warning, is it when replaying the bag? You should launch `rtabmap.launch.py` with `use_sim_time:=true` and play the bag with `--clock` option. It is sure if you have sync issue and rtabmap node didn't publish for 5 sec, `rtabmap_viz` will then complain that it didn't receive data for past 5 sec.
> 
> cheers, Mathieu

Thank you for your reply and for supporting this software. Getting this work done is part of my full-time job, so I really appreciate your help.

I have tested it in real time with the custom stereo setup I have. I am using my own ROS 2 driver using the depthai library for the mono oak cameras. So, it is a steep learning curve on how to synchronize the left and right images with software. I will check the rosbags with the flags you suggested and edit my reply. I am using Jetson Orin Nano; does that affect the behaviour of RTABMAP? In real time also, I get a black screen in the 3D visualization in RTABMAP, and in RVIZ, only the local map and last frame point clouds are being published. I am also changing some things in my ROS 2 driver to get better sync. I will update on Monday. 

To answer your question, yes, I got the 5-second warning when I ran the first rosbag with RTABMAP.