introlab/rtabmap_ros #1191: PluginlibFactory: The plugin for class 'rtabmap_ros/MapGraph' failed to load.  Error: According to the loaded plugin descriptions the class rtabmap_ros/MapGraph with base class type rviz::Display does not exist.

leslie-fig, 2024-08-02:
I ran into this issue after downloading a navigation package and following the Readme file to execute the mapping command。By the way,it didn't work though I had install rtabmap-rviz packages.

matlabbe, 2024-08-02:
Humm, can you do a `sudo apt update && sudo apt upgrade` ? Also if you built rtabmap_ros from source, clean your catkin or ros2 ws and rebuild everything. There are maybe unsync libraries.

leslie-fig, 2024-08-03:
> Humm, can you do a `sudo apt update && sudo apt upgrade` ? Also if you built rtabmap_ros from source, clean your catkin or ros2 ws and rebuild everything. There are maybe unsync libraries.
Ok,I will try the method you offered me later。

leslie-fig, 2024-08-03:
> Humm, can you do a `sudo apt update && sudo apt upgrade` ? Also if you built rtabmap_ros from source, clean your catkin or ros2 ws and rebuild everything. There are maybe unsync libraries.

> Humm, can you do a `sudo apt update && sudo apt upgrade` ? Also if you built rtabmap_ros from source, clean your catkin or ros2 ws and rebuild everything. There are maybe unsync libraries.

There is another issue I met that was "Failed to create the global_planner/GlobalPlanner planner, are you sure it is properly registered and that the containing library is built? Exception: According to the loaded plugin descriptions the class global_planner/GlobalPlanner with base class type nav_core::BaseGlobalPlanner does not exist. Declared types are  carrot_planner/CarrotPlanner evolutionary_planner/EvolutionaryPlanner graph_planner/GraphPlanner lazy_planner/LazyPlanner navfn/NavfnROS sample_planner/SamplePlanner"。And some of the content above like "evolutionary_planner/EvolutionaryPlanner graph_planner/GraphPlanner lazy_planner/LazyPlanner" is in my other project when I was launching a navigation file。

matlabbe, 2024-08-03:
Doublecheck if library containing nav_core::BaseGlobalPlanner is installed.

leslie-fig, 2024-08-03:
Can u tell me how to check this problem that whether I'v installed the plugin or not?I'm not good at it。😭😭😭

leslie-fig, 2024-08-04:
> Doublecheck if library containing nav_core::BaseGlobalPlanner is installed.
[image]
The picture I show you that's my navigation launch file.And I have installed global-planner using "ros-noetic-global-planner" 。I found the solution that was using the command above to install nav_core::BaseGlobalPlanner.But the bug still remains，And I think why I can't solve this problem because of the error in the later paragraph.Weird
[image]

matlabbe, 2024-08-12:
Not sure if `ros-noetic-global-planner` should have worked. You could try this planner instead (which seems to be installed in your error msg):
```
<param name="base_global_planner" value="navfn/NavfnROS"/>
```