introlab/rtabmap_ros #1192: Weird cone-shaped point cloud using mono camera with Depth-Anything-V2 to get the RGB-D image

Muktiadi, 2024-08-03:
Hi, I am trying to use Depth-Anything-V2 with mono camera to emulate RGB-D image. I have managed to create the nodes and produce a neat depth image out of it. But I got problem with rtabmap reconstructing the point cloud. It seems make an erratic cone-shaped 3D from the middle of the frame, and expanding deeper to the edges.

[image]
[image]

I wonder if anyone can help me point out what did I miss.

**Launcher file**
```
<launch>
    <!-- Single camera setup -->
    <node name="camera" pkg="usb_cam" type="usb_cam_node" output="screen">
        <param name="video_device" value="/dev/video0" />
        <param name="image_width" value="640" />
        <param name="image_height" value="480" />
        <param name="pixel_format" value="mjpeg" />
        <param name="camera_frame_id" value="camera_optical_frame" />
        <param name="io_method" value="mmap"/>
        <param name="framerate" value="1" />
        <param name="camera_info_url" value="file://$(find my_stereo_setup)/ost.yaml"/>
    </node>

    <!-- Image rectification -->
    <node pkg="image_proc" type="image_proc" name="image_proc">
        <remap from="image_raw" to="/camera/image_raw"/>
        <remap from="camera_info" to="/camera/camera_info"/>
    </node>

    <!-- Depth-Anything-V2 node -->
    <node name="depth_anything_node" pkg="my_stereo_setup" type="depth_anything_node.py" output="screen">
        <param name="model_path" value="$(env HOME)/Projects/Depth-Anything-V2/checkpoints/depth_anything_v2_vits.pth"/>
        <param name="encoder" value="vits"/>
        <param name="calibration_file" value="$(find my_stereo_setup)/ost.yaml"/>
    </node>

    <!-- RGBD Sync node -->
    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_sync/rgbd_sync" output="screen">
        <remap from="rgb/image" to="/camera/rgb/image_rect_color"/>
        <remap from="depth/image" to="/camera/depth_registered/image_raw"/>
        <remap from="rgb/camera_info" to="/camera/camera_info"/>
        <remap from="rgbd_image" to="/rgbd_image"/>
        
        <param name="approx_sync" value="true"/>
        <param name="approx_sync_max_interval" value="0.1"/>
        <param name="queue_size" value="5"/>
    </node>

    <node pkg="tf" type="static_transform_publisher" name="odom_to_camera" 
          args="0 0 0 0 0 0 odom camera_optical_frame 10" />

	<!-- RTAB-Map node -->
<include file="$(find rtabmap_launch)/launch/rtabmap.launch">
    <arg name="rgbd_topic" value="/rgbd_image"/>
    <arg name="camera_info_topic" value="/camera/camera_info"/>
    <arg name="frame_id" value="camera_optical_frame"/>
    <arg name="odom_frame_id" value="odom"/>
    <arg name="approx_sync" value="true"/>
    <arg name="queue_size" value="5"/>
    <arg name="visual_odometry" value="true"/>
    <arg name="wait_for_transform" value="5.0"/>
    <arg name="odom_guess_frame_id" value="odom"/>
    <arg name="rgbd_depth_scale" value="0.1"/>
    <arg name="rviz" value="true"/>
    <arg name="rtabmap_args" value="
        --delete_db_on_start
        --Vis/FeatureType 1
        --Vis/MinInliers 10
        --Vis/MaxFeatures 1000
        --RGBD/ProximityBySpace true
        --RGBD/AngularUpdate 0.05
[... 134 lines cut ...]
        current_time = rospy.Time.now()

        # Publish RGB and depth images
        rgb_msg = self.bridge.cv2_to_imgmsg(cv_image, "bgr8")
        rgb_msg.header.stamp = current_time
        rgb_msg.header.frame_id = "camera_optical_frame"
        self.rgb_pub.publish(rgb_msg)

        depth_msg = self.bridge.cv2_to_imgmsg(depth_normalized, "16UC1")
        depth_msg.header.stamp = current_time
        depth_msg.header.frame_id = "camera_optical_frame"
        self.depth_pub.publish(depth_msg)

        # Publish camera info
        self.camera_info.header.stamp = current_time
        self.camera_info.header.frame_id = "camera_optical_frame"
        self.camera_info_pub.publish(self.camera_info)

        rospy.loginfo(f"Published RGB and depth images. RGB shape: {cv_image.shape}, Depth shape: {depth_normalized.shape}")

if __name__ == '__main__':
    rospy.init_node('depth_anything_node', anonymous=True)
    node = DepthAnythingNode()
    rospy.spin()
```

Muktiadi, 2024-08-04:
I just tried to use ORB_SLAM3 RGB-D mode, and it reproduce the similar cone shape. Must be a bug on my implementation code.

[image]

Muktiadi, 2024-08-04:
Preliminary guess: Should the depth information be in meters? I've used the full range of 16UC1 (0, 65535). Now I've realized there's a [metric model for Depth-Anything-V2](https://huggingface.co/depth-anything/Depth-Anything-V2-Metric-VKITTI-Large) aswell. I'm going to try to refactor my code with it.

matlabbe, 2024-08-12:
If you use 16UC1, it should be in mm. If you use 32FC1, it should be in meters. How is the depth scale estimated? If scale is not fixed between the images while the camera is moving, that will make visual odometry fails.