introlab/rtabmap_ros #1194: OctoMap fails to fully detect moving objects

csvicbot-7, 2024-08-08:
Hi @matlabbe 

I'm encountering an issue with OctoMap when using it to detect obstacles. Initially, when I launch RTAB-Map, OctoMap detects all obstacles correctly. However, when someone walks around the LiDAR sensor, OctoMap fails to detect the person’s entire body.

[image]

https://github.com/user-attachments/assets/9e70dc4a-75e4-4d00-9fab-479631def5b5

In the attached files, you can see the contrast between the two topics: `/rtabmap/local_grid_obstacle` (red points) and `/rtabmap/octomap_obstacles` (cubes). OctoMap detects only a small part of the person's body, while the /rtabmap/local_grid_obstacle topic captures the person's body more accurately.

To illustrate this issue further, I’ve attached the following files in the [link](https://drive.google.com/drive/folders/1xMq_NwA-hvHC_-uR4ev7es9zKQr59yuI?usp=sharing)
- A database
- A rosbag 
- A launch file

Additionally, I am unable to visualize the `/rtabmap/octomap_full` topic in RViz. I have already installed the [octomap_rviz_plugins](https://github.com/OctoMap/octomap_rviz_plugins) package, but I receive the error message "Failed to create octree structure."
[image]

Could you please provide some advice on how to resolve these issues?

Thanks.

matlabbe, 2024-08-16:
You should use ColorOccupancyGrid display type:
[image]

Looking at the octomap, we see more cubes added:
[image]

However, if the obstacle cloud doesn't get updated similarly, it means the occupancy probability inside those cubes are below `GridGlobal/OccupancyThr` (default 0.5). Here showing cell probability:
[image]

Not sure why in your video there are no points added around the head where it seems keeping more bright green cubes. A rosbag with the ouster point cloud could be useful to test with.

Note also that OctoMap is not meant to track dynamic obstacles as its refresh rate is low (1 Hz by default). For fast dynamic objects, consider using a [local voxel grid](http://wiki.ros.org/costmap_2d/hydro/obstacles#VoxelCostmapPlugin).

csvicbot-7, 2024-10-02:
Hi @matlabbe 

Thank you for your response.

I was testing octomap in two modes: Grid/RayTracing parameter set to true and false.

`Grid/RayTracing = true`

In the first part of the video you can see a person walking in front of the sensor. The whole body of the person is identified as an obstacle, but as he moves, the head part is not refreshed. Why does this happen?

However, after a few seconds if I enable and disable the topic (`/rtabmap/octomap_full`) in rviz, the head voxels disappear. But if I have a node that subscribes to the map (`/rtabmap/octomap_full`) these voxels never disappear, unless the node stops subscribing to the map. This can be seen at min 00:51 of the video. 
The node that records the rosbag subscribes to the map topic. So while the recording is running the voxels are not refreshed; but when the recording stops the voxels disappear.

https://github.com/user-attachments/assets/aa957024-9d1c-4b78-961d-48aef7c33915

`Grid/RayTracing = false`

In this case the person's body is detected as an obstacle but the voxels are not refreshed. Why does this happen?

As in the previous case, after a few seconds the voxels corresponding to the person's body are refreshed, only if there are no nodes subscribed to the map topic (min 00:39).

https://github.com/user-attachments/assets/477e95ab-b5e2-4296-bca4-2f7802ef4d69

In both cases, why if there is a node subscribing to the map topic (`/rtabmap/octomap_full`) some voxels are not refreshed?

I’ve attached the following files in the [link](https://drive.google.com/drive/folders/1xMq_NwA-hvHC_-uR4ev7es9zKQr59yuI?usp=sharing)

- Two rosbags (one for each case), each rosbag has: /tf transformations, Ouster point cloud /os_cloud_node/point, /rtabmap/octomap_full map.
- Database
- Launch file

Thank you in advance for your reply.

matlabbe, 2024-10-05:
Thanks for the video and the bag. The reason why the octomap is cleared after unsubscribing to octomap topic is that rtabmap will flush the whole octomap kept in RAM and reconstruct it from scratch **with nodes in the graph**. As you are using `RGBD/LinearUpdate=0.05` and `RGBD/AngularUpdate=0.05`, the graph doesn't increase in size because the lidar is not moving. So the resulting re-generated octomap contains only the first node added at the beggining (where nobody was in front of the sensor). If you set `RGBD/LinearUpdate=0`, you swill see that the octomap is re-created exactly like before unsubscribing to it.

And for why the octomap is updated even if no nodes are added to map, it is because of parameter `map_always_update` is true.

The main issue however is why the head is not removed from the octomap? It is because `Grid/MaxObstacleHeight` is set to 2 meters, so all lidar points over that threshold are not used for ray tracing. 
[image]

Setting to 0 would make the head cleared.

cheers,
Mathieu

closed 2025-03-25 by csvicbot-7