introlab/rtabmap_ros #1195: Map changing in map topic

brayanpa, 2024-08-12:
Hello,

I'm currently using rtabmap in combination with nav2, and I'm encountering an issue where rtabmap modifies the map during the loading of the database in localization mode. Specifically, I'm using the map.pgm file generated by the map_server for nav2. Initially, the map loaded from the image and the map published by rtabmap align perfectly, allowing for consistent navigation and localization.

However, after loading, the map published by rtabmap changes and no longer aligns with the map saved as an image. Once this happens, the maps never coincide again. This issue does not occur with all maps, only with some of them.

Do you have any idea what might be causing this deviation? If there's any additional information I can provide to help resolve this issue, please let me know. Thank you for your assistance!

I've also attached some images illustrating the misalignment between the maps. This not happened in all the maps just in some of them. 

[image]
[image]
[image]

matlabbe, 2024-08-15:
If the database doesn't have an optimized map saved, it will re-optimize and re-assemble on runtime, which may create some variations. You can make sure an optimized is saved by opening the database in rtabmap-databaseViewer, and check in File menu if you can visualize an optimized map (if not, you can generate with corresponding menu option). Also when the map was exported to PGM? if it was during mapping, if rtabmap continues mapping afterwards it may extend the map to that origin doesn't match with the one previously saved.

Ideally after mapping with rtabmap, make sure in rtabmap-databaseViewer there is an optimized map, then re-launch in localization mode and save the map as PGM/metadata. If the database is always opened in localization mode, it should not change over time.

brayanpa, 2025-02-27:
You were absolutely right! We are now generating the PGM after ensuring that the optimized map exists, so the map never changes.

Thanks a lot for your help!

closed 2025-02-27 by brayanpa